UniversalRobots/Universal_Robots_ROS_Driver #4: Can't accept new action goals

Abbyls, 2019-10-14:
# Versions
 - ROS Driver version: Kinetic+ubuntu16.04
 - URCaps Software version(s): 5.2 for UR5e

# Summary
I am trying to control the robot arm with moveit planning. Before I start, I have changed the controller.yaml under the directory: $HOME/ur_ros_driver/src/universal_robot/ur5_e_moveit_config/config/controllers.yaml to:
`controller_list:
  - name: scaled_pos_traj_controller
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint`
What I did is just as the name of "scaled_pos_traj_controller".
Then, 
Here are the commands I run:
# terminal1:
`roslaunch ur_robot_driver ur5e_bringup.launch use_tool_communication:=true tool_voltage:=24 tool_parity:=0 tool_baud_rate:=115200 tool_stop_bits:=1 tool_rx_idle_chars:=1.5 tool_tx_idle_chars:=3.5 robot_ip:=localhost tool_device_name:=/tmp/ttyUR
` 
output in terminal 1:
`SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 500
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /scaled_pos_traj_controller/action_monitor_rate: 10
 * /scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
[... 333 lines cut ...]
[ INFO] [1571057846.842176006]: Ready to take commands for planning group manipulator.
[ INFO] [1571057846.842297713]: Looking around: no
[ INFO] [1571057846.842375344]: Replanning: no
[ WARN] [1571057846.849456909]: Interactive marker 'EE:goal_ee_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

Then, I planned a motion in Rviz, it succeeds, but when I press the execute button, 
**errors happpened:**
**New Output in terminal 1:**
> [ERROR] [1571057857.641259323]: **_Can't accept new action goals. Controller is not running_**.
**New outputs in terminal 2**:
> [ INFO] [1571057853.198470389]: manipulator[RRTConnectkConfigDefault]: Created 5 states (2 start + 3 goal)
[ INFO] [1571057853.198749599]: manipulator[RRTConnectkConfigDefault]: Created 5 states (2 start + 3 goal)
[ INFO] [1571057853.198825319]: ParallelPlan::solve(): Solution found by one or more threads in 0.001333 seconds
[ INFO] [1571057853.204060450]: SimpleSetup: Path simplification took 0.005165 seconds and changed from 3 to 2 states
[ INFO] [1571057853.211439154]: Planning adapters have added states at index positions: [ 0 ]
[ INFO] [1571057857.638956931]: Received new trajectory execution service request...
[ WARN] [1571057857.641742353]: **Controller scaled_pos_traj_controller failed with error code INVALID_GOAL**
[ WARN] [1571057857.641909116]: **Controller handle scaled_pos_traj_controller reports status FAILED**
[ INFO] [1571057857.641984913]: Completed trajectory execution with status FAILED ...
[ INFO] [1571057857.642147358]: Execution completed: FAILED

It seems the scaled_pos_traj_controller can not communicate with the moveit. The moveit package that I was running is from the Build guidance:
`git clone -b calibration_devel https://github.com/fmauch/universal_robot.git`.

Is there someone who has any ideas for this problem?

fmauch, 2019-10-14:
You just closed this issue again. Is it resolved?

I guess you missed starting the program on the robot's teach pendant according to the output?

Abbyls, 2019-10-14:
@fmauch No, I just did it by mistake, I want to change the format to be easy understanding, but clicked the close button<<<.  Can you reopen it again? I can not find a way to open it. And I could you explain to me more details reason that this problem is related to the teach pendant? Thanks!

Abbyls, 2019-10-14:
@fmauch No, I just did it by mistake, I want to change the format to be easy understanding, but clicked the close button<<<. 
And could you explain to me more details that this problem is related to the teach pendant? Thanks!

Abbyls, 2019-10-14:
When I use Ctrl+C to stop the  ur5e_bringup.launch in Terminal 1, it shows:
`[ERROR] [1571064332.286059757]: **Can't accept new action goals. Controller is not running**.
^C[controller_stopper-6] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
Interrupt signal (2) received.
[INFO] [1571064337.869564]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1571064337.870365]: Stopping all controllers...
[ERROR] [1571064337.873004254]: **Could not stop controller 'scaled_pos_traj_controller' since it is not running**
[INFO] [1571064337.873174]: Unloading all loaded controllers...
[INFO] [1571064337.873963]: Trying to unload force_torque_sensor_controller
terminate called without an active exception
[WARN] [1571064337.967741]: **Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer**
`
It shows that the scaled_pos_traj_controller is not loaded well, do you have any ideas according to this error?

fmauch, 2019-10-15:
A suggested above it seems as if the program was never started on the robot. Did you install the URCap, create a program with the `External Control` program node and start that program after launching the driver?

You should see those lines in the shell running the driver after starting the program on the teach pendant:
```
[ INFO] [1571124040.693851608]: Robot requested program
[ INFO] [1571124040.693924407]: Sent program to robot
[ INFO] [1571124040.772090597]: Robot ready to receive control commands.
```

Abbyls, 2019-10-15:
Yes, you are right, I did not run the `External Control` program after the driver starting. 
And the main problem is the network configuration. We are using the University's network, there is a firewall between the remote pc and robot arm, So I should change the Host IP address in the `External Control` to localhost and add the port into config file. 

After these things are done, it works well. 
Thanks!

closed 2019-10-15 by Abbyls

cardboardcode, 2019-10-17:
Hi @Abbyls, facing the same issue on my end as well. 
Tried the solution above to simply start the URCap program after starting the driver. Still doesn't work.

May I clarify which `config file` you `added the port to`?

fmauch, 2019-10-17:
@cardboardcode I think @Abbyls case is a bit special due to the firewall.

For the driver to work, you should
 1. Install and setup the `External Control` URCap (**With the ROS-PC's IP address) and create a Program with the URCap's program node as explained here: [CB3](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/velocity_interface/ur_robot_driver/doc/install_urcap_cb3.md), [e-Series](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/velocity_interface/ur_robot_driver/doc/install_urcap_e_series.md)
 2. Start the driver handing the IP **of the robot** as explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/velocity_interface#quick-start)
 3. Load and start the previously generated program on the TP. It will fail if no connection with the ROS-PC can be established with a popup saying `The connection to the remote PC could not be established`. When this is happening, either the ROS node is not running or the IP setup in the URCap's installation is wrong. If no error shows, the shell running the ROS node should show the three lines as explained above:

```
[ INFO] [1571124040.693851608]: Robot requested program
[ INFO] [1571124040.693924407]: Sent program to robot
[ INFO] [1571124040.772090597]: Robot ready to receive control commands.
```

If you followed these steps and do not see those three lines, please open up a new issue describing exactly, what steps you performed.

cardboardcode, 2019-10-17:
Hi @fmauch, thank you for the quick reponse.

The issue has been resolved on my end. 

It turned out that I did not Load the `External Control` URCap program properly before running it under the `Run` tab of the UR3e Teach Pendant. 

Followed your instructions above. Works perfectly. Thank you.

Csp0920, 2019-11-20:
HI! @fmauch  Sorry to disturb you.
I follow the steps to do UR3e, but there are some problems that I can't solve.
And I have installed URCaps.

1.When I run the code "roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=192.168.56.1"
The results are as follow:
[ INFO] [1574214426.534807566]: Initializing dashboard client
[ INFO] [1574214426.539844598]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1574214426.541180419]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[INFO] [1574214426.844632]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1574214426.846603]: Controller Spawner: Waiting for service controller_manager/load_controller
[WARN] [1574214172.516681]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1574214172.517799]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[ros_control_controller_spawner-4] process has finished cleanly
log file: /home/csp/.ros/log/a7bc892a-0b37-11ea-b92e-1cb72c9ca797/ros_control_controller_spawner-4*.log
[ros_control_stopped_spawner-5] process has finished cleanly
log file: /home/csp/.ros/log/a7bc892a-0b37-11ea-b92e-1cb72c9ca797/ros_control_stopped_spawner-5*.log
[ERROR] [1574209530.114572643]: Connection setup failed for 192.168.56.1:29999
[ INFO] [1574209530.145880846]: Initializing urdriver
[ERROR] [1574209661.186558331]: Connection setup failed for 192.168.56.1:30002
[ INFO] [1574209661.186732848]: Checking if calibration data matches connected robot.
[ERROR] [1574209792.258277611]: Connection setup failed for 192.168.56.1:30001
[FATAL] [1574209792.258600472]: Failed to connect to robot. Please check if the robot is booted and connected.
[ERROR] [1574209792.258695138]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 23831, exit code 1, cmd /home/csp/catkin_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/csp/.ros/log/f3db34e8-0b2b-11ea-b92e-1cb72c9ca797/ur_hardware_interface-3.log].
log file: /home/csp/.ros/log/f3db34e8-0b2b-11ea-b92e-1cb72c9ca797/ur_hardware_interface-3*.log
Initiating shutdown!

There are two Host IP of UR robot, one is in the network setting, which is 192.168.1.10. The other one is in URCaps, which is 192.168.56.1. So which one should connect when I set the PC cable connection? Should I make the two Host IP same?

When I pressed the run of the teach pendant , it showed "the connection to the remote PC could not be established".
Hoping for your reply. Thanks.

Csp0920, 2019-11-20:
HI! @fmauch  @cardboardcode .  Sorry to disturb you.
I have a problem can't solve. Could you help me?

When I run "roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=192.168.56.1", the results are as follow:

[ INFO] [1574241722.736017884]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1574241722.738892201]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ INFO] [1574241722.742789839]: Initializing dashboard client
[ INFO] [1574241722.745620115]: Connected: Universal Robots Dashboard Server

[ INFO] [1574241722.792444165]: Initializing urdriver
[ INFO] [1574241722.792761629]: Checking if calibration data matches connected robot.
[ WARN] [1574241722.792998737]: No realtime capabilities found. Consider using a realtime system for better performance
[ERROR] [1574241723.016394620]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [TODO Link to documentation] for details.
[INFO] [1574241723.019236]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1574241723.021915]: Controller Spawner: Waiting for service controller_manager/load_controller
[ WARN] [1574241723.820101389]: No realtime capabilities found. Consider using a realtime system for better performance
[ INFO] [1574241724.117637351]: Setting up RTDE communication with frequency 500.000000
[ WARN] [1574241725.133279716]: No realtime capabilities found. Consider using a realtime system for better performance
[ INFO] [1574241725.133437315]: Loaded ur_robot_driver hardware_interface
[ INFO] [1574241725.172698763]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1574241725.172744848]: Service available.
[ INFO] [1574241725.172785232]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1574241725.173775416]: Service available.
[ INFO] [1574241725.317197688]: Robot's safety mode is now NORMAL
[ INFO] [1574241725.318116377]: Robot mode is now RUNNING
[INFO] [1574241725.447552]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1574241725.449081]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1574241725.450459]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1574241725.451439]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1574241725.452602]: Loading controller: joint_state_controller
[INFO] [1574241725.453695]: Loading controller: pos_traj_controller
[INFO] [1574241725.462671]: Loading controller: scaled_pos_traj_controller
[INFO] [1574241725.504371]: Controller Spawner: Loaded controllers: pos_traj_controller
[INFO] [1574241725.552416]: Loading controller: speed_scaling_state_controller
[INFO] [1574241725.558484]: Loading controller: force_torque_sensor_controller
[INFO] [1574241725.564435]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1574241725.568487]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
^C[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [1574241828.228064]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1574241828.228706]: Stopping all controllers...
[INFO] [1574241828.228852]: Shutting down spawner. Stopping and unloading controllers...
Interrupt signal (2) received.
[INFO] [1574241828.229510]: Stopping all controllers...
[ERROR] [1574241828.232296058]: Could not stop controller 'pos_traj_controller' since it is not running
[INFO] [1574241828.232577]: Unloading all loaded controllers...
[INFO] [1574241828.232985]: Trying to unload pos_traj_controller
[ERROR] [1574241828.236400130]: Could not stop controller 'scaled_pos_traj_controller' since it is not running
[INFO] [1574241828.236779]: Unloading all loaded controllers...
[INFO] [1574241828.237132]: Trying to unload force_torque_sensor_controller
terminate called without an active exception
[WARN] [1574241828.425136]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[WARN] [1574241828.425533]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

And when I press run bottom in Teach Pendant, it showed "The connection to the remote PC could not be established". 
I really don't know how to solve these problems.

Hoping for your reponse. Thank you very much.

Csp0920, 2019-11-20:
@gavanderhoorn  Could I put my problems here? And do you have any suggestion?
Thank you very much.

gavanderhoorn, 2019-11-20:
I would probably have posted a new issue, as it's unclear right now whether your problem is related to what was reported by the OP in this one.

You may still want to post this as a new issue. The maintainers here can then decide whether it's a duplicate of a stand-alone problem.

Csp0920, 2019-11-20:
@fmauch  Yes, I have followed the steps as suggested in #4(comment).

Csp0920, 2019-11-20:
@fmauch  I open an issue that describes my operation steps and settings.

rijalmadhav12, 2022-03-31:
**Output in terminal 1** command **roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=157.182.212.168**

[ INFO] [1648752007.906606352]: Robot requested program
[ INFO] [1648752007.906654709]: Sent program to robot
[ INFO] [1648752008.035448733]: Robot connected to reverse interface. Ready to receive control commands.

**Output in terminal 2** command **roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true**
INFO] [1648752113.584409468]: Execution request received
[ WARN] [1648752113.592252256]: Controller scaled_pos_joint_traj_controller failed with error INVALID_GOAL: 
[ WARN] [1648752113.592340594]: Controller handle scaled_pos_joint_traj_controller reports status FAILED
[ INFO] [1648752113.592365572]: Completed trajectory execution with status FAILED ...
[ INFO] [1648752113.592424714]: Execution completed: FAILED

**Output in terminal 3** command  **roslaunch ur_robot_driver example_rviz.launch**
 INFO] [1648752073.648837615]: Loading robot model 'ur5_robot'...
[ WARN] [1648752073.666914085]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1648752073.709676447]: Starting planning scene monitor
[ INFO] [1648752073.711262853]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1648752073.720208224]: No active joints or end effectors found for group 'endeffector'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1648752073.720740370]: No active joints or end effectors found for group 'endeffector'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1648752073.723755380]: Constructing new MoveGroup connection for group 'endeffector' in namespace ''
[ INFO] [1648752074.812047769]: Ready to take commands for planning group endeffector.
[ INFO] [1648752074.812117879]: Looking around: no
[ INFO] [1648752074.812152580]: Replanning: no
[ INFO] [1648752098.962033712]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1648752098.977448419]: Ready to take commands for planning group manipulator.
[ INFO] [1648752098.977476058]: Looking around: no
[ INFO] [1648752098.977485525]: Replanning: no
[ INFO] [1648752113.613834200]: ABORTED: Solution found but controller failed during execution

Robot type ur5 cb3
ros melodic ,ununtu 18
External control urcap is also running in the robot..

Why am i still geting the error ; **Can't accept new action goals. Controller is not running.**

fmauch, 2022-04-01:
@rijalmadhav12 please do not hijack old closed issues. The short answer: Do you use my fork of universal_robot on the calibration_devel? I assume not, as on that branch the launchfile does not have a `limited` parameter anymore.