UniversalRobots/Universal_Robots_ROS_Driver #5: Tool Communication understanding

Abbyls, 2019-10-14:
# Summary
I am confused about the use of the tool communication. 
In your features list you mentioned that:
use tool communication on e-series: yes
use the driver without a teach pendant necessary: planned
Do you mean that if I want to use the RS485 cable for control, like in the following image, I still need to use the teach pendant? Can I use your driver to open the tool communication and use the qb device ros driver [()](http://wiki.ros.org/Robots/qbhand) to send commands control the hand remotely without program on the teach pendant?  
[image]

Currently, I have tried your driver to open the tool communication:
1. installed the externalcontrol-1.0.urcap and the rs485-1.0.urcap on the teach pendant (5.2 version for ur5e)
2. set the IP address of the externalcontrol urcap the same as my UR5e arm
3. run the driver:
`roslaunch ur_robot_driver ur5e_bringup.launch use_tool_communication:=true tool_voltage:=24 tool_parity:=0 tool_baud_rate:=115200 tool_stop_bits:=1 tool_rx_idle_chars:=1.5 tool_tx_idle_chars:=3.5 robot_ip:=localhost tool_device_name:=/tmp/ttyUR`
The driver seems to work.
Then I run the qb hand driver:
`roslaunch qb_device_driver communication_handler.launch`
But it outputs that the driver can not connect to the device:
`[ WARN] [1569862191.354844984]: [CommunicationHandler] is waiting for devices...
[ INFO] [1569862192.355195259]: [CommunicationHandler] has found [0] devices connected:`

fmauch, 2019-10-15:
> 2. set the IP address of the externalcontrol urcap the same as my UR5e arm

I am confused. In the "Installation" of `External Control` you should configure the IP **of the ROS-Pc**. Then, when launching the driver, the parameter `robot_ip` should contain the IP address **of your robot**

To your actual question: From my knowledge, the qb driver has some "interesting" method of trying to aquire each tty device it can find. For that it searches at different locations where it expects them to be. `/tmp/ttyUR` is probably none of these locations. For it to work, you'll probably have to create your local `ttyUR` inside `/dev` which however might require some privilege modifications which is why this is not the default.

Abbyls, 2019-10-15:
Thanks, according to your info, I checked the directory that the qb device searching, found it only searches /dev directory, then I need to use `sudo su` to run the driver for writing the file into /dev directory. 

However, the detected device_id is not settled, it is a random number, do you have ideas about how to set the detected device_id to be the same at a different time?

fmauch, 2019-10-16:
You can set the device name as launch file parameter, then it should always be that.

However, I think you are referring to the device_id as detected by the qb driver. I hope, you understand, that we cannot give support for the qb_hand driver here.

Abbyls, 2019-10-16:
Yes, we have set the device name as a parameter, and want to detect the device_id or keep the device_id as the same at different running round. 

Thanks, I understand,  such questions should be asked under the QB hand driver page.

closed 2019-10-16 by Abbyls