UniversalRobots/Universal_Robots_ROS_Driver #6: Failed to read from stream, Pipeline producer overflowed!

Briancbn, 2019-10-16:
# Summary
After installing the externalcontrol-1.0.urcap and use my remote pc ip address as the remote host address, I launch the ur3e_bringup.launch file but I keep getting the error **Pipeline producer overflowed! <RTDE Data Pipeline>**, Communication error, what other setting steps did I miss?

# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s): PolyScope 5.4
 - Affected Robot Hardware Version(s): ur3e
 - Robot Serial Number:
 - UR+ product(s) installed: ur3e
 - URCaps Software version(s): externalcontrol-1.0.urcap

# Issue details
[image]

## Use Case and Setup
Just testing the ur3e driver usability currently

## Steps to Reproduce
See summary

## Expected Behavior
I didn't expect there to be any error when running the ur3e_bringup program

Abbyls, 2019-10-22:
This error also happened to me, but not every time. And it did not affect my Moveit controlling.

fmauch, 2019-10-23:
Thanks for this clarification. I'll prepare a fix that should prevent this from happening.

Briancbn, 2019-10-26:
We managed to temporarily resolved the problem by switching to another PC. It happens all the time on the original PC I tested the code with, but the new PC seems to be ok for now. (I am not sure this helps)

fmauch, 2019-10-28:
I still think, that this is a bug. Which is why I asked whether the errors eventually stop and the driver is running normally. I'll provide a fix, anyway.

fmauch, 2019-10-28:
@Briancbn, @Abbyls could you please try to reproduce the issue using #18 applied?

fmauch, 2019-11-07:
Should be fixed in #18. Closing this. If it appears to be not fixed, please reopen.

closed 2019-11-07 by fmauch

ljden, 2021-05-10:
I'm getting this error while running the current stable (master) branch. Would you like me to open a new issue or reopen this one?

fmauch, 2021-05-15:
> I'm getting this error while running the current stable (master) branch. Would you like me to open a new issue?

Please open a new one describing your specific problem. If we realize that this is really the same problem as here, we can merge them later ( define the new issue a duplicate).

ljden, 2021-05-15:
> > I'm getting this error while running the current stable (master) branch. Would you like me to open a new issue?
> 
> Please open a new one describing your specific problem. If we realize that this is really the same problem as here, we can merge them later ( define the new issue a duplicate).

See #367. Probably a different source of error