UniversalRobots/Universal_Robots_ROS_Driver #7: TypeError: sequence item 2: expected string, int found

cardboardcode, 2019-10-17:
# Summary

Following the instructions need to setup tool communication [link](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/setup_tool_communication.md), I encountered the following issue.

[image]

[Textual Interpretation]
```
File "/home/user/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/scripts/tool_communication", line 31, in main
    cmd.append(":".join(["tcp", robot_ip, tcp_port]))
TypeError: sequence item 2: expected string, int found
```
# Versions
 - ROS Driver version: PolyScope 5.4
 - Affected Robot Software Version(s): ur3e
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:ur3e
 - URCaps Software version(s): externalcontrol-1.0.urcap 

# Issue details
To provide more context to my issue, I encountered this problem as I am trying to integrate a Robotiq 2 Fingered 85mm gripper as the UR3e end effector. A USB to RS-485 wire connects the gripper to the UR3e control box digital I/O as well as USB port. 

My goal is to communicate via TCP and send scripts to open and close the gripper. However, as it is required to run the `ExtCon.urp` URcap program under `Local`, I am unable to achieve the above.

fmauch, 2019-10-17:
Could you try this branch, please?

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/fix_tool_comm

Does that fix the issue?

cardboardcode, 2019-10-17:
Hi @fmauch, thank you. 

The issue has been resolved. I tried something similar by explicit-casting `tcp_port` directly to `string`. 

It works now.

closed 2019-10-17 by cardboardcode

fmauch, 2019-10-17:
But the string casting was necessary, right? Then I'll go ahead and merge this.