UniversalRobots/Universal_Robots_ROS_Driver #11: Fail to build the package in catkin workspace

XYHong18, 2019-10-21:
Hi,
I followed your instruction to git clone this repository and universal_robot, and install all the dependencies. But I failed to build the package using command "catkin_make". 

I am using Ubuntu 16.04, ROS kinetic, UR5e of version 5.1.0.

Below is my error:

Base path: /home/xinyi/catkin_ws
Source space: /home/xinyi/catkin_ws/src
Build space: /home/xinyi/catkin_ws/build
Devel space: /home/xinyi/catkin_ws/devel
Install space: /home/xinyi/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/xinyi/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/xinyi/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetPayload
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetIO
[  0%] Built target _ur_msgs_generate_messages_check_deps_MasterboardDataMsg
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetSpeedSliderFraction
[  0%] Built target _ur_msgs_generate_messages_check_deps_IOStates
[  0%] Built target _ur_msgs_generate_messages_check_deps_Digital
[  0%] Built target _ur_msgs_generate_messages_check_deps_ToolDataMsg
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _ur_msgs_generate_messages_check_deps_RobotStateRTMsg
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target controller_manager_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target controller_manager_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target controller_manager_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target controller_manager_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target controller_manager_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target ur_driver_gencfg
[  0%] Built target _catkin_empty_exported_target
[  0%] Built target _ur_msgs_generate_messages_check_deps_Analog
[  0%] Built target _ur_msgs_generate_messages_check_deps_RobotModeDataMsg
[  2%] Built target ur_hardware_interface
[  2%] Built target sensor_msgs_generate_messages_nodejs
[  2%] Built target sensor_msgs_generate_messages_eus
[  2%] Built target sensor_msgs_generate_messages_py
[  5%] Built target kinect2_registration
[  5%] Built target sensor_msgs_generate_messages_lisp
[  5%] Built target sensor_msgs_generate_messages_cpp
[  5%] Built target geometry_msgs_generate_messages_py
[  5%] Built target geometry_msgs_generate_messages_cpp
[  5%] Built target geometry_msgs_generate_messages_lisp
[  5%] Built target geometry_msgs_generate_messages_eus
[  5%] Built target geometry_msgs_generate_messages_nodejs
[  5%] Built target tf_generate_messages_lisp
[  5%] Built target tf2_msgs_generate_messages_eus
[  5%] Built target actionlib_msgs_generate_messages_py
[  5%] Built target tf2_msgs_generate_messages_lisp
[  5%] Built target tf2_msgs_generate_messages_py
[  5%] Built target actionlib_generate_messages_cpp
[  5%] Built target actionlib_msgs_generate_messages_nodejs
[  5%] Built target tf_generate_messages_nodejs
[... 372 lines cut ...]
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:104: recipe for target '/home/xinyi/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_RobotModeDataMsg.py' failed
make[2]: *** [/home/xinyi/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_RobotModeDataMsg.py] Error 1
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:94: recipe for target '/home/xinyi/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_IOStates.py' failed
make[2]: *** [/home/xinyi/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_IOStates.py] Error 1
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:99: recipe for target '/home/xinyi/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_RobotStateRTMsg.py' failed
make[2]: *** [/home/xinyi/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_RobotStateRTMsg.py] Error 1
CMakeFiles/Makefile2:1047: recipe for target 'universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/all' failed
make[1]: *** [universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 61%] Linking CXX shared library /home/xinyi/catkin_ws/devel/lib/libur_controllers.so
[ 61%] Built target ur_controllers
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

I doubt if the issue is caused by python version.
And below is what I get if I use command "echo $PYTHONPATH":

/opt/ros/kinetic/lib/python2.7/dist-packages:/home/xinyi/catkin_ws/devel/lib/python3/dist-packages:/usr/lib/python2.7/dist-packages

Thanks for your help!
Best.

fmauch, 2019-10-21:
There seems to be some misconfiguration in your python setup, as you are using python 2.7 (which is normal) and python3 inside your workspace (Which doesn't play along well)

Try cleaning your workspace (deleting build, devel and install), open a new shell and make sure nothing python3 related is getting sourced. Then, try to rebuild.

gavanderhoorn, 2019-10-21:
I can confirm this.

@XYHong18: see one of the many Q&As on ROS Answers about this: [site:answers.ros.org no module named error](https://www.google.nl/search?q=no+module+named+error++site%3Aanswers.ros.org).

XYHong18, 2019-10-22:
@fmauch @gavanderhoorn Thanks for your reply! Problem solved!

closed 2019-10-22 by XYHong18

Karthik-Ranga, 2020-02-11:
I know this is a closed issue, but is it possible to run the driver under python 3? I need to do this so I can integrate UR with another system. I get a failed catkin_make. is there a workaround? I know it is the problem with ROS.
I built the whole ROS melodic under python 3 and I get the error "ModuleNotFoundError" during catkin_make.

gavanderhoorn, 2020-02-12:
There is almost zero use of Python in these packages (only a single script IIRC).

A build error from `catkin_make` also does not seem like something we can do anything about here.

I'd recommend posting this to a more suitable forum.