UniversalRobots/Universal_Robots_ROS_Driver #13: Probe for sending script code on eSeries

fmauch, 2019-10-21:
When an e-Series robot runs in local mode, script code sent through the primary interface is ignored. However, the user doesn't get notified of this. Unfortunately, we cannot read whether the robot is currently in remote_control_mode or in local_mode.

The idea is to write a small probe routine that tries to send a string through a socket via script code and see whether we receive this string on the remote side.

fmauch, 2019-11-27:
One important question is when to actually probe this. As this can be changed at any arbitrary moment while the driver is running, only probing at driver startup will not be sufficient.

Performing the check every time we want to send script code also seems wrong as this produces a quite significant delay each time a user wants to send script code.

A third option would be to perform this check cyclically. While I also don't like this method very much it seems to be the best solution currently coming to my mind.

gavanderhoorn, 2019-11-27:
It would be perfect if local/remote mode would be something that can be retrieved either over RTDE or in any other way from the controller.

Then we could keep track of it locally in `ur_robot_driver` and act accordingly.

No need to poll or check.

fmauch, 2019-11-27:
> Perhaps something to discuss with UR.

Forgot to mention: Already did that.

Unfortunately, there seems to be no interface (neither of URScript, primary interface, RTDE, dashbaord) to provide that information.

gavanderhoorn, 2019-11-27:
No, not now.

It could be added.

That would make sense anyway: it's a rather basic thing to be able to export local/remote configuration from a robot controller. For instance to a PLC, to make sure the system is in a consistent state before starting a cell.

fmauch, 2020-01-10:
Update: Since software version 5.6 there is an interface inside the dashboard server that provides this information (see also #76). However, this would still require some kind of active polling as discussed above.

gavanderhoorn, 2020-01-10:
Hm. I thought I'd understood from Anders that this information was actually exported over RTDE (as well)?

Would be really unfortunate if that is not the case.

urfeex, 2026-01-19:
Since ROS (1) is end of life this will not be worked on anymore.

See the workaround mentioned in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/13#issuecomment-572914625

closed 2026-01-19 by urfeex