UniversalRobots/Universal_Robots_ROS_Driver #15: Connection to robot dropped, waiting for new connection.

Abbyls, 2019-10-22:
# Summary

Sometimes I got info after starting the driver:
`[ INFO] [1571753843.897362575]: Robot requested program
[ INFO] [1571753843.897400758]: Sent program to robot
[ INFO] [1571753854.029805703]: Robot ready to receive control commands.
[ INFO] [1571753854.057101264]: **Connection to robot dropped, waiting for new connection**.`

I did not enable the realtime function on my computer. 

I am not sure if this will affect my controlling performance. 
Actually my qb hand controlling not runs well through this tool communication. It seems there is a big delay for my qb hand control. 

# Versions
 - ROS Driver version: Kinetic
 - Robot Serial Number:UR5e + 5.2
 - URCaps Software version(s): ExternalControl, qb hand, RS485

Abbyls, 2019-10-22:
I got an issue when I run:
`rostopic echo /ur_hardware_interface/io_states`

The output in the driver terminal is:
'''
[ INFO] [1571760445.028097672]: Robot requested program
[ INFO] [1571760445.028197749]: Sent program to robot
[ INFO] [1571760455.157596158]: Robot ready to receive control commands.
[ INFO] [1571760455.348997308]: Connection to robot dropped, waiting for new connection.
[ERROR] [1571760473.108828244]: Client [/rostopic_6355_1571760472865] wants topic /ur_hardware_interface/io_states to have datatype/md5sum [ur_msgs/IOStates/0a5c7b73e3189e9a2caf8583d1bae2e2], but our version has [ur_msgs/IOStates/3033784e7041da89491b97cc4c1105b5]. Dropping connection.
[ERROR] [1571760474.117565527]: Client [/rostopic_6355_1571760472865] wants topic /ur_hardware_interface/io_states to have datatype/md5sum [ur_msgs/IOStates/0a5c7b73e3189e9a2caf8583d1bae2e2], but our version has [ur_msgs/IOStates/3033784e7041da89491b97cc4c1105b5]. Dropping connection.
[ERROR] [1571760475.121444803]: Client [/rostopic_6355_1571760472865] wants topic /ur_hardware_interface/io_states to have datatype/md5sum [ur_msgs/IOStates/0a5c7b73e3189e9a2caf8583d1bae2e2], but our version has [ur_msgs/IOStates/3033784e7041da89491b97cc4c1105b5]. Dropping connection.
^C[controller_stopper-7] killing on exit
'''
Is there someone know the solution? Do I really need to install the realtime kernel in my PC?

fmauch, 2019-10-23:
@gavanderhoorn is right, this usually means that message versions differ. This could either result from a `rostopic echo ...` called from a shell that didn't source the workspace but only the system-wide ROS-installation which contains the released version of `ur_msgs`, or from calling `rostopic echo ...` from another PC which also only contains the released messages.

fmauch, 2019-10-23:
To comment on the original question:

The message `**Connection to robot dropped, waiting for new connection**.` means that the Polyscope program running the `External Control` node got interrupted, which can have multiple reasons:
 * You pressed *pause* on the TP
 * You pressed *stop* on the TP
 * You moved around the robot using the TP
 * You started another program e.g. through the primary interface (not likely, as the robot would have to be in remote-control mode for this)
 * There is a fault or EM-stop happening on the robot
 * Probably a couple more reasons I don't think of now...
Anyway, you can simply press *play* on the TP again (In case, you didn't load a different program in the meantime) and you should see the `Robot ready to receive control commands.` output again and the robot will be up and running again.

Maybe we can improve the documentation and output on that...

Abbyls, 2019-10-23:
@fmauch 
I found I can not repress the start button on the TP to start the External Control URcap.
The program tries to start, then after less than 1 second, it stops again. The output from the terminal is:
`[ INFO] [1571835693.690114333]: Robot requested program
[ INFO] [1571835693.690202238]: Sent program to robot
[ INFO] [1571835694.436166279]: Robot ready to receive control commands.
[ INFO] [1571835694.491010487]: Connection to robot dropped, waiting for new connection.
[ INFO] [1571835697.964610909]: Robot requested program
[ INFO] [1571835697.964642722]: Sent program to robot
[ INFO] [1571835698.714006277]: Robot ready to receive control commands.
[ INFO] [1571835698.813041225]: Connection to robot dropped, waiting for new connection.
[ INFO] [1571835700.138341864]: Robot requested program
[ INFO] [1571835700.138386929]: Sent program to robot
[ INFO] [1571835700.899810198]: Robot ready to receive control commands.
[ INFO] [1571835701.026888034]: Connection to robot dropped, waiting for new connection.
[ INFO] [1571835704.337226085]: Robot requested program
[ INFO] [1571835704.337287842]: Sent program to robot
[ INFO] [1571835705.083735909]: Robot ready to receive control commands.
[ INFO] [1571835705.246928995]: Connection to robot dropped, waiting for new connection.
[ INFO] [1571835709.070370748]: Robot requested program
[ INFO] [1571835709.070473914]: Sent program to robot
[ INFO] [1571835709.819887993]: Robot ready to receive control commands.
[ INFO] [1571835709.910761412]: Connection to robot dropped, waiting for new connection.
.....`
And I need to restart the TP to get the program start again, but later after one time of success, the problem happened again...

Abbyls, 2019-10-23:
@fmauch @gavanderhoorn 
Comment on the second issue:
You are right, I run the 
> rostopic echo ...
in another UR driver workspace...
After I sourced this package, the rostopic echo works well.

fmauch, 2019-10-23:
> @fmauch
> I found I can not repress the start button on the TP to start the External Control URcap.
> The program tries to start, then after less than 1 second, it stops again. The output from the terminal is:
> 
> > [ INFO] [1571835693.690114333]: Robot requested program
> > [ INFO] [1571835693.690202238]: Sent program to robot
> > [ INFO] [1571835694.436166279]: Robot ready to receive control commands.
> > [ INFO] [1571835694.491010487]: Connection to robot dropped, waiting for new connection.
> > [ INFO] [1571835697.964610909]: Robot requested program
> > [ INFO] [1571835697.964642722]: Sent program to robot
> > [ INFO] [1571835698.714006277]: Robot ready to receive control commands.
> > [ INFO] [1571835698.813041225]: Connection to robot dropped, waiting for new connection.
> > [ INFO] [1571835700.138341864]: Robot requested program
> > [ INFO] [1571835700.138386929]: Sent program to robot
> > [ INFO] [1571835700.899810198]: Robot ready to receive control commands.
> > [ INFO] [1571835701.026888034]: Connection to robot dropped, waiting for new connection.
> > ....
> 
> And I need to restart the TP to get the program start again, but later after one time of success, the problem happened again...

Then, let's debug this any further. Could you please
- Save the robot program from the teach pendant as a name, lets call it ''\<robotProgramName>''
- Save the installation from the teach pendant, the default name is ''default.installation''

and attach the following files from the robot to this issue:
 - The log: ''/root/log_history.txt''
 - The robot program: ''/programs/\<robotProgramName>.urp''
 - The generated script: ''/programs/\<robotProgramName>.script''
 - The installation: ''/programs/default.installation''

You can copy the files using scp with user: root, pw: easybot

Abbyls, 2019-10-24:
Hi,

The files have been attached. 
I want to say, this problem always happens after I got a fault (on the TP, says "too high current on tool power pin") from running the QB soft-hand ROS control driver. Then, I can not start the external control program again (always says "Connection to robot dropped, waiting for new connection") unless I restart the arm. 

[files.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/3767334/files.zip)

Thanks!

fmauch, 2019-10-25:
> I want to say, this problem always happens after I got a fault (on the TP, says "too high current on tool power pin") from running the QB soft-hand ROS control driver. Then, I can not start the [...]

That's a rather interesting bit of information! What kind of message is the error? A safety popup? A violation?

I am a bit worried about the `Not able to open socket reverse_socket to host: 192.168.1.2 at port: 50001` lines in the log. That would mean, that the driver would not be correctly running on the ROS machine. Please check that port `50001` and port `50002` are open, when the error occurs:

```
$ netstat -tlpn | grep ur_robot
tcp        0      0 0.0.0.0:50001           0.0.0.0:*               LISTEN      7983/ur_robot_drive
tcp        0      0 0.0.0.0:50002           0.0.0.0:*               LISTEN      7983/ur_robot_drive
tcp        0      0 0.0.0.0:39097           0.0.0.0:*               LISTEN      7983/ur_robot_drive
tcp        0      0 0.0.0.0:55649           0.0.0.0:*               LISTEN      7983/ur_robot_drive
```

Abbyls, 2019-10-25:
That is a safety popup:
[image]
[image]

These two images generated at different times, one is saying "too high current on tool power pin", the other says "Too high current on tool connector supply pins". 

I can not get the ports info by running:
> netstat -tlpn | grep ur_robot
So I have tested the port: 50001 and port: 50002 by:
> netstat -tlpn | grep 50001
> netstat -tlpn | grep 50002
before and after the error happened. The output is:
`(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
tcp        0      0 0.0.0.0:50001           0.0.0.0:*               LISTEN      -               
`
`(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
tcp        0      0 0.0.0.0:50002           0.0.0.0:*               LISTEN      -   `

gavanderhoorn, 2019-10-25:
@Abbyls: for the `-p` option to `netstat` your user needs to have the capabilities enabled *or* you need to run `netstat` with `sudo`.

fmauch, 2019-10-25:
> @Abbyls: for the `-p` option to `netstat` your user needs to have the capabilities enabled _or_ you need to run `netstat` with `sudo`.

For me it was working without sudo, then I have setup my machine differently...

@Abbyls From your photos, I expect, you did go through the initialization screen and recovery procedure and not simply dismiss the dialog? When this occurs, the robot is switched off and has to be powered on again, before being able to start the program again. OK, I just checked, one cannot start the program without the robot being operational.

However, in the program you sent, there is no program node for the qb hardware, while in the photos there is a second node called "qb: 50%" Could you please attach the complete program (*.script) that is actually causing the problem, so I can check that for any interferences? Best, have the ROS side running when writing the program, then it will contain the full script code.

Abbyls, 2019-10-28:
Hi,

I have tried to "go the initialization" or "not now" options when this safety error happens. Both of them can not help me start the external control program. 

The first image that contains the "qb:50%" is from my previous experiment, which is not related to the files that I attached before. So I think I do not need to attach these files again, it will be the same.

fmauch, 2019-11-07:
@Abbyls I can't quite get behind what could be the problem there. Could you please try it with the latest master version, as we now also publish the robot's status.

Abbyls, 2019-11-14:
Ok, I will try it later.

fmauch, 2020-05-12:
Any news on this?

Abbyls, 2020-05-21:
> Any news on this?

Hi, we are using the original cable connection currently. So no more testing has been taken on this issue.

fmauch, 2020-09-25:
Closing due to inactivity. Please reopen if new information is gathered.

closed 2020-09-25 by fmauch