UniversalRobots/Universal_Robots_ROS_Driver #17: Improve documentation and output on dropping robot control

fmauch, 2019-10-23:
This driver will recognize, when the command-interpreting part of the driver running as Polyscope program stops due to some reason and notify the user with a message `**Connection to robot dropped, waiting for new connection**`

With that, one can simply press *play* on the TP again (or call `rosservice call /ur_hardware_interface/resend_program` in headless mode).

However, this is not really documented and we could also add a hint to the actual ROS output.

Abbyls, 2019-10-23:
Hi,

Thanks for your updated comment. However, I can not restart the external control in the TP side.
Even when I try to run the rosservice:
> rosservice call /ur_hardware_interface/resend_program
It shows an error that:
> ERROR: Service [/ur_hardware_interface/resend_program] is not available.

The driver is running on another terminal:
> roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=ur5 headless_mode:=true
And the TP is under remote control mode (the error is the same when I change it to local control).

fmauch, 2019-10-23:
The service you tried to call is only available in headless mode, which is not yet fully supported.

But let's keep discussing this issue in #15, please

PR #44 (2019-11-27): Adding a troubleshooting section with previously raised questions.
This adds a troubleshooting section to the main README file.

This should resolve #17 and #28.
closed 2019-11-27 by fmauch