UniversalRobots/Universal_Robots_ROS_Driver #23: actionlib client not working

DebasmitaGhose, 2019-11-05:
# Summary

I am trying to create an actionlib client to send joint positions to the actionlib server. 

# Impact

The client is stuck at client.waiting_for_server()

# Issue details
Here is the code snippet I am running:
```
def ur_move_client():
	client = actionlib.SimpleActionClient('foo', FollowJointTrajectoryAction)
	print("0")
	pprint(client)
	client.wait_for_server()
	print("1")

	goal = control_msgs.msg.FollowJointTrajectoryActionGoal()
	print(goal)
	goal.goal.trajectory.joint_names = ['base','shoulder', 'elbow', 'wrist1', 'wrist2', 'wrist3']
	goal.goal.trajectory.points  = [-178.73, -67.50, -105.47, -191.12, -80.19, -91.92]
	client.send_goal(goal)	
	client.wait_for_result()
	print("2")

if __name__ == '__main__':
	rospy.init_node('ur_move')
	rospy.sleep(2.0)

	ur_move_client()
```

## Project status at point of discovered
The robot moves when controlled from the GUI.
Here is the [rqt_plot](https://pasteboard.co/IFkOIut.png)
The client node is ```ur_move```

fmauch, 2019-11-06:
Please let me ask the obvious question first: did you remap  'foo' to the actual action server when starting the node?

fmauch, 2019-11-07:
I've just taken a second look at your code and there seem to be a lot of errors in the python snippet. You could look at the [sample script of the old python-based driver](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_driver/test_move.py)

You can use this also with our driver by starting it in the correct namespace, e.g.

**CAUTION: This will move the robot**
```
ROS_NAMESPACE=/scaled_pos_traj_controller rosrun ur_driver test_move.py
```

I'm closing this as I don't see an issue with the driver. If you have difficulties setting up a working ROS node to command the robot, please refer to https://answers.ros.org

closed 2019-11-07 by fmauch