UniversalRobots/Universal_Robots_ROS_Driver #29: FollowJointTrajectory action with control loop on UR controller

v4hn, 2019-11-07:
Thank you so much for this work, it seems to literally clean every single interface that might be relevant and forward a lot of things that had to be ignored before. :bouquet:

Looking into deploying the new driver, we see the need for a controller that forwards `FollowJointTrajectory` goals/trajectories to the robot controller in terms of a quintic profile specified by waypoints. I am honestly surprised this is not the default mode of operation of URs officially backed ROS controller, as the `ros_control` approach is of course nicer if properly setup, but still requires fiddling with realtime parameters and load management to run stable.

The [low_bandwidth_follower](https://github.com/ros-industrial/ur_modern_driver/blob/kinetic-devel/launch/ur_common.launch#L18) of the `ur_modern_driver` roughly provides this functionality, but fails to accurately arrive at the destined goal location.

Are there plans to incorporate such a mode in this driver, too?
Will the `ur_modern_driver` remain the best way to do this for the foreseeable future?
Is this already readily supported in some UR-backed way and we are just not aware of it?

fmauch, 2019-11-07:
Thank you for your feedback.

We have been talking about forwarding whole trajectories to the robot for execution internally. Such an interface is not part of the current development phase, that's all I can say right now...

v4hn, 2019-11-07:
Thanks for the quick response, although with bad news.

I guess, until such a feature is implemented, this also means the `ur_modern_driver` is not fully deprecated by this package :pensive:

gavanderhoorn, 2019-11-07:
I don't believe anyone claimed that @v4hn ?

gavanderhoorn, 2019-11-07:
> After all [the feature list](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/features.md) clearly implies it's superior (or planned to be) in every relevant aspect.

except apparently forwarding full trajectories .. (and a few other things)

stefanscherzinger, 2021-09-01:
This feature will be available with [`pass_through` mechanisms](https://github.com/UniversalRobots/Universal_Robots_ROS_passthrough_controllers#rationale) of Cartesian and joint-based trajectories with #408

closed 2021-09-01 by stefanscherzinger

v4hn, 2021-09-01:
I saw this effort recently and have to thank you a lot for it!
It was way overdue to have these features available in a generic implementation and you did a great job conceptualizing interfaces and implementing them!

Now I just have to get around to actually use them in a future project. :-)

stefanscherzinger, 2021-09-02:
Thank you for this feedback @v4hn !