UniversalRobots/Universal_Robots_ROS_Driver #32: Publish industrial_msgs/RobotStatus messages

gavanderhoorn, 2019-11-07:
tl;dr: would there be interest in adding [industrial_robot_status_controller](http://wiki.ros.org/industrial_robot_status_controller) compatibility to `ur_robot_driver`?

(I've ignored the template, as it doesn't seem to cater to discussions about potential contributions / feature/enhancement requests)

---

`ur_modern_driver` publishes [industrial_msgs/RobotStatus](http://docs.ros.org/api/industrial_msgs/html/msg/RobotStatus.html) messages which give a (admittedly limited) view on robot/controller state.

`ur_robot_driver` does not do this, which complicates migrating from `ur_modern_driver`.

Instead of adding support for this directly to `ur_robot_driver`, an intermediary in the form of the [industrial_robot_status_controller](http://wiki.ros.org/industrial_robot_status_controller) could be used. `ur_robot_driver` would have to expose an `IndustrialRobotStatusHandle` and populate some fields in that, after which the controller would be responsible for conversion to and publication of the actual `RobotStatus` message.

As without the controller nothing happens, users would have an opt-in and the default behaviour of the `hardware_interface` would not change.

fmauch, 2019-11-07:
Yes, this seems very reasonable to me :-)

gavanderhoorn, 2019-11-07:
I'll see if I can submit a PR for this in the next few days.

axelschroth, 2020-01-22:
Any plans/progress on this?

gavanderhoorn, 2020-01-22:
It's on my TODO list, but I've not worked on it yet.

axelschroth, 2020-03-18:
Since we are using robot status messages and need this to be implemented, do you have a rough estimate when you'll be able to submit a PR? Can I support you in any way?

gavanderhoorn, 2020-03-18:
If you need this, then please don't wait on me, as I will not have time to work on this in the foreseeable future.

IIRC, @simonschmeisser has already implemented this once in `ur_modern_driver`. The idea is the same, but now you'd use the intermediate `IndustrialRobotStatusInterface` to push all the info into. The `IndustrialRobotStatusController` will do the rest.

The readme of `industrial_robot_status_controller` should contain enough info on this ([here](https://github.com/gavanderhoorn/industrial_robot_status_controller#usageintegration)).

axelschroth, 2020-03-18:
Alright, thank you! I'll have a look into this issue shortly.

axelschroth, 2020-03-24:
Does anyone have an idea, why no further state messages are published by industrial_robot_status_controller as soon as I stop the robot program (with ExternalControl Cap, see PR)?

gavanderhoorn, 2020-04-21:
Closing as #141 got merged.

Thanks again @axelschroth.

closed 2020-04-21 by gavanderhoorn