UniversalRobots/Universal_Robots_ROS_Driver #33: URScript via topic cancels robot program on controller

tragolf, 2019-11-07:
I have a question or rather a functionallity question. Currently we control the robot partly by script commands via the ROS topic and partly by a program on the robot. The program gets canceled if there are commands send via the topic. We solve this issue by restarting the program on the controller by setting of an IO. The problem is that the loading takes a few seconds which is not good for our time critical process.
Is there a way to not cancel the running program on the robot controller when sending commands via the topic?
It is currently not possible to do the whole execution via script commands and without the program on the controller.

fmauch, 2019-11-07:
You basically have two options (depending on the type of scripts you're sending only one)

 1. You can restart the program using the new dashboard interface after sending your script via ´rosservice call /ur_hardware_interface/dashboard/play` This will work in every case, but still interrupt the main program on script commands.
 2. You can send your commands wrapped into a [secondary program](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/secondary-program-17257/):
```
sec programName():
  script_command_you_want_to_send()
end
```
However, only some script commands qualify for this (See link above: _A secondary program may not take any physical time to execute_)

tragolf, 2019-11-07:
1. Is this the same as we already did? We set a digital_in which automatically starts the program (you can define this in the io settings). The program is huge so loading takes ca. 4s (as if we press play on the touch panel). The ´rosservice call /ur_hardware_interface/dashboard/play` is basically the same?
2. The script commands include move commands so this sadly doesn't work.
Thank you for your help

fmauch, 2019-11-07:
> 1. Is this the same as we already did? We set a digital_in which automatically starts the program (you can define this in the io settings). The program is huge so loading takes ca. 4s (as if we [...]
Yes, just without the IO

> 2. The script commands include move commands so this sadly doesn't work.
>    Thank you for your help
Yes, but that's a restriction (or design) of the controller. There can*t be two primary programs running at the same time. Are you ok with closing this issue for now?

tragolf, 2019-11-07:
> Are you ok with closing this issue for now?
Yes. Thank you!

closed 2019-11-07 by tragolf