UniversalRobots/Universal_Robots_ROS_Driver #39: Calibration correction: process has died

atenpas, 2019-11-15:
# Summary
The `calibration_correction` program does not work, and results in a `process has died exit code -11` message.

# Versions
 - ROS Driver version: master
 - Affected Robot Software Version(s): 3.11.0
 - Affected Robot Hardware Version(s): ur5 cb31
 - Robot Serial Number: 
 - UR+ product(s) installed:
 - URCaps Software version(s): External Control V 1.0.1

# Impact
The robot's calibration cannot be corrected.

# Issue details
The complete output from the program is:
```
[DEBUG]: Starting up producer
[WARNING]: No realtime capabilities found. Consider using a realtime system for better performance
[WARNING]: Failed to read from stream, reconnecting in 1 seconds...
[DEBUG]: Invalid robot package type recieved: 24
[DEBUG]: Invalid robot package type recieved: 23
[DEBUG]: Invalid robot package type recieved: 25
[DEBUG]: Invalid robot package type recieved: 5
[INFO]: checksum: [3606817617 2359058659 797309772 1842629547 2361290454 2337035062 ]
dh_theta: [-2.13924526400084e-05 0.707863340837074 -0.697806691494563 -0.00939010396913045 -0.000103960175745391 2.84715683846765e-05 ]
dh_a: [0.000135505406191284 -0.323000093546539 -0.39222225336313 8.95993028540395e-05 6.7159905438739e-05 0 ]
dh_d: [0.0892432254030618 305.938911206983 -305.696785008317 -0.1315987344071 0.0948065362922913 0.0824990251147532 ]
dh_alpha: [1.57067557276495 0.000903167434024947 0.0155918816117158 1.569814771042 -1.56965778256486 0 ]
calibration_status: 2

[DEBUG]: Destructing pipeline
[DEBUG]: Stopping pipeline! <Pipeline>
[DEBUG]: Pipeline consumer ended! <Pipeline>
[DEBUG]: Pipeline producer ended! <Pipeline>
[ INFO] [1573830518.763447395]: Writing calibration data to "/home/ur5/.ros/test.yaml"
[calibration_correction-1] process has died [pid 17366, exit code -11
```

## Steps to Reproduce
With *external control* loaded as a program in Polyscope, execute `$ roslaunch ur_calibration calibration_correction.launch robot_ip:=10.75.15.199 target_filename:="${HOME}/test.yaml"`.

fmauch, 2019-11-18:
I just checked and it doesn't seem to be a version regression. Works with a 3.11 URSim on my machine.

Could you please try using the branch https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/file_error_output? That should print a textual representation of the error code.

fmauch, 2019-12-19:
@atenpas Does the error still exist?

w-kyle, 2020-03-02:
I am also experiencing this problem both with a UR5 and UR5e, with polyscope versions 3.11 and 5.3 respectively, with the master branch on Ubuntu 16.04, ROS Kinetic. Was a solution found?

fmauch, 2020-03-05:
@w-kyle Could you please post the complete output of the run?

w-kyle, 2020-03-06:
Here's the output. It's essentially identical to atenpas's output.

roslaunch ur_calibration calibration_correction.launch robot_ip:=10.75.15.250 target_filename:="test_caibration.yaml"
... logging to /home/colin/.ros/log/1f243fde-5fe4-11ea-aa52-4ccc6a230519/roslaunch-ur5-for-president-12802.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ur5-for-president:46167/

SUMMARY
========

PARAMETERS
 * /calibration_correction/output_filename: /home/colin/caibr...
 * /calibration_correction/robot_ip: 10.75.15.250
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    calibration_correction (ur_calibration/calibration_correction)

auto-starting new master
process[master]: started with pid [12817]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1f243fde-5fe4-11ea-aa52-4ccc6a230519
process[rosout-1]: started with pid [12830]
started core service [/rosout]
process[calibration_correction-2]: started with pid [12834]
[DEBUG]: Starting up producer
[WARNING]: No realtime capabilities found. Consider using a realtime system for better performance
[WARNING]: Failed to read from stream, reconnecting in 1 seconds...
[DEBUG]: Invalid robot package type recieved: 24
[DEBUG]: Invalid robot package type recieved: 23
[DEBUG]: Invalid robot package type recieved: 5
[INFO]: checksum: [126914495 1585184434 2599448413 2692272767 3573782575 3886035546 ]
dh_theta: [-2.95309332065357e-05 0.667740210622319 -0.640475987028862 -6.31048499045852 -7.47414180193452e-06 1.05903328671499e-05 ]
dh_a: [0.000338212593485813 -0.333588830339598 -0.392088727172344 -4.43600177715567e-05 4.78015086783138e-05 0 ]
dh_d: [0.0891650821286223 304.963188136377 -300.557489744559 -4.29567752378192 0.0949247819222485 0.0824867440901803 ]
dh_alpha: [1.57038772780874 0.00086264335794213 0.00242777755182474 1.57118219430416 -1.5704429295828 0 ]
calibration_status: 1

[DEBUG]: Destructing pipeline
[DEBUG]: Stopping pipeline! <Pipeline>
[DEBUG]: Pipeline consumer ended! <Pipeline>
[DEBUG]: Pipeline producer ended! <Pipeline>
[ INFO] [1583524398.959700522]: Writing calibration data to "/home/colin/caibration.yaml"
[calibration_correction-2] process has died [pid 12834, exit code -11, cmd /home/colin/ur_driver_ws/devel/lib/ur_calibration/calibration_correction __name:=calibration_correction __log:=/home/colin/.ros/log/1f243fde-5fe4-11ea-aa52-4ccc6a230519/calibration_correction-2.log].
log file: /home/colin/.ros/log/1f243fde-5fe4-11ea-aa52-4ccc6a230519/calibration_correction-2*.log

fmauch, 2020-03-27:
I think, I need some debugging help from your side:

Could you please replace  [these lines](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_calibration/src/calibration_correction.cpp#L121-L125) with the following code:

```c++
    std::ofstream file(output_filename_);
    ROS_INFO_STREAM("Created ofstream");
    if (file.is_open())
    {
      ROS_INFO_STREAM("File is open.");
      ROS_INFO_STREAM("Writing data:\n" << *calibration_data_);
      file << *calibration_data_;
    }
```

I would be curious which line is actually failing...

fmauch, 2020-11-16:
Closing due to inactivity. Feel free to comment or reopen if anything new comes up.

closed 2020-11-16 by fmauch

atenpas, 2020-12-15:
Reopening. I still get about the same error for the latest [commit](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/783b1d1f5774220058d0a2aa3dd93185641e8c7e), but with a different calibration status (2 instead of 1).

```
$ roslaunch ur_calibration calibration_correction.launch robot_ip:=10.75.15.198 target_filename:="${HOME}/Documents/my_robot_calibration.yaml" -v
... logging to /home/ur5/.ros/log/505ab716-3efc-11eb-be77-1c1b0dec0dfd/roslaunch-ur5-earth-9494.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

... loading XML file [/opt/ros/kinetic/etc/ros/roscore.xml]
... executing command param [rosversion roslaunch]
Added parameter [/rosversion]
... executing command param [rosversion -d]
Added parameter [/rosdistro]
Added core node of type [rosout/rosout] in namespace [/]
... loading XML file [/home/ur5/ur_driver_ws/src/Universal_Robots_ROS_Driver/ur_calibration/launch/calibration_correction.launch]
Added parameter [/calibration_correction/robot_ip]
Added parameter [/calibration_correction/output_filename]
Added node of type [ur_calibration/calibration_correction] in namespace [/]
started roslaunch server http://ur5-earth:34215/

SUMMARY
========

PARAMETERS
 * /calibration_correction/output_filename: /home/ur5/Documen...
 * /calibration_correction/robot_ip: 10.75.15.198
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    calibration_correction (ur_calibration/calibration_correction)

auto-starting new master
process[master]: started with pid [9504]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 505ab716-3efc-11eb-be77-1c1b0dec0dfd
WARNING: Package name "Seq6DofManipRobot" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosout-1]: started with pid [9519]
started core service [/rosout]
process[calibration_correction-2]: started with pid [9522]
[DEBUG]: Starting up producer
[WARNING]: No realtime capabilities found. Consider using a realtime system for better performance
[WARNING]: Failed to read from stream, reconnecting in 1 seconds...
[DEBUG]: Invalid robot package type recieved: 24
[DEBUG]: Invalid robot package type recieved: 23
[DEBUG]: Invalid robot package type recieved: 5
[INFO]: checksum: [3606817617 2359058659 797309772 1842629547 2361290454 2337035062 ]
dh_theta: [-2.13924526400084e-05 0.707863340837074 -0.697806691494563 -0.00939010396913045 -0.000103960175745391 2.84715683846765e-05 ]
dh_a: [0.000135505406191284 -0.323000093546539 -0.39222225336313 8.95993028540395e-05 6.7159905438739e-05 0 ]
dh_d: [0.0892432254030618 305.938911206983 -305.696785008317 -0.1315987344071 0.0948065362922913 0.0824990251147532 ]
dh_alpha: [1.57067557276495 0.000903167434024947 0.0155918816117158 1.569814771042 -1.56965778256486 0 ]
calibration_status: 2

[DEBUG]: Destructing pipeline
[DEBUG]: Stopping pipeline! <Pipeline>
[DEBUG]: Pipeline consumer ended! <Pipeline>
[DEBUG]: Pipeline producer ended! <Pipeline>
[ INFO] [1608053898.538215255]: Writing calibration data to "/home/ur5/Documents/my_robot_calibration.yaml"
[ INFO] [1608053898.538374185]: Writing calibration data to /home/ur5/Documents/my_robot_calibration.yaml
[calibration_correction-2] process has died [pid 9522, exit code -11, cmd /home/ur5/ur_driver_ws/devel/lib/ur_calibration/calibration_correction __name:=calibration_correction __log:=/home/ur5/.ros/log/505ab716-3efc-11eb-be77-1c1b0dec0dfd/calibration_correction-2.log].
log file: /home/ur5/.ros/log/505ab716-3efc-11eb-be77-1c1b0dec0dfd/calibration_correction-2*.log
/home/ur5/Documents/my_robot_calibration.yaml^C[rosout-1] killing on exit
```

fmauch, 2020-12-31:
@atenpas I would be curious why there are two `Writing calibration data to` lines....

Anyway, this narrows it down to the actual streaming of yaml data. Could you please test the version from [this branch](https://github.com/fmauch/Universal_Robots_ROS_Driver/tree/debug_calibration)? That might give some insights on why this actually fails.