UniversalRobots/Universal_Robots_ROS_Driver #40: Remote URScript execution best practice

khssnv, 2019-11-18:
Hello. I need to execute URScript functions remotely generated *on-the-fly*. I've tried raw socket interface and this driver `/ur_hardware_interface/script_command` topic. Raw socket interface provides response when function execution ends and works good. In this driver ROS API I have not find how to get the same for functions sent to the `script_command` topic.

I used a little trick with `set` and `reset` io pin in the beginning and in the end of function. Like this:
```python
def my_func():
    set_digital_out(5, True)
    # function body
    set_digital_out(5, False)
end
```
This way I can track function execution in my higher-level code looking at `io` pin state by `/ur_hardware_interface/io_states` topic.

I see #16 provides ROS API for dashboard, but as I understand, it works only with scripts in controller memory. This means for *on-the-fly* remotely generated URScript functions, I have to load new one by ssh / scp in filesystem and don't forget to delete them sometimes.

Is there a better practice or which one recommended? Thank you.

fmauch, 2019-11-18:
> Raw socket interface provides response when function execution ends.

I have to admit that I am surprised by that. Never having used the raw socket interface that much this might just me lacking knowledge, but from my understanding I send script code to either the primary or secondary interface and it gets executed. What I get sent back is only data being sent through the interface anyway (as specified here: https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/)

Where would I get such feedback about finished code execution?

khssnv, 2019-11-19:
Excuse me, I misinformed you. For raw socket connection I used a module written by my colleague.
It adds `socket_send_byte()` before the end of each URScript I send, so I can wait for a byte from robot to send next URScript. Something like:

Replace
```python
end
```
with
```python
open_result = socket_open(SERVER_ADDRESS, SERVER_PORT, "socket_0")\n
if (open_result == True):\n
    socket_send_byte(79, "socket_0")\n
    socket_close(socket_name="socket_0")\n
end\n
```

Anyway, what I need is a way to know when robot complete execution of URScript sent to it. If there is no method or plan for anything (like action server recieves URScript function to execute) but methods described in opening message, please close this issue.

fmauch, 2019-11-20:
I see. That's of course a different story. You can do the same thing with using the driver's script capabilities. The only difference would be to use `rostopic pub` for sending the script instead to an own socket. I'll go ahead and close this, then.

closed 2019-11-20 by fmauch