UniversalRobots/Universal_Robots_ROS_Driver #43: Can't connect UR3e with PC

Csp0920, 2019-11-20:
# Summary
I am trying to connect PC with UR3e. Following the tutorials, I can't realize the connection.
There are two problems:
1.Running "roslaunch ur3_e_bringup.launch" error.
2.When run URCaps in the Tech Pendant, it showed "The connection to the remote PC could not be established".

# Versions
 - ROS Driver version:Kinetic+ubuntu16.04
 - Affected Robot Software Version(s): 5.3 for UR3e

# Issue details
I downloaded the ROS-Industrial Universal Robot meta-package and Universal_Robots_ROS_Driver. Then I followed the steps and installed URCaps, but I skipped real-time kernel.

After doing catkin_make and source...., I run "roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=192.168.56.1", the results are as follow:

[ INFO] [1574241722.736017884]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1574241722.738892201]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ INFO] [1574241722.742789839]: Initializing dashboard client
[ INFO] [1574241722.745620115]: Connected: Universal Robots Dashboard Server

[ INFO] [1574241722.792444165]: Initializing urdriver
[ INFO] [1574241722.792761629]: Checking if calibration data matches connected robot.
[ WARN] [1574241722.792998737]: No realtime capabilities found. Consider using a realtime system for better performance
[ERROR] [1574241723.016394620]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [TODO Link to documentation] for details.
[INFO] [1574241723.019236]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1574241723.021915]: Controller Spawner: Waiting for service controller_manager/load_controller
[ WARN] [1574241723.820101389]: No realtime capabilities found. Consider using a realtime system for better performance
[ INFO] [1574241724.117637351]: Setting up RTDE communication with frequency 500.000000
[ WARN] [1574241725.133279716]: No realtime capabilities found. Consider using a realtime system for better performance
[ INFO] [1574241725.133437315]: Loaded ur_robot_driver hardware_interface
[ INFO] [1574241725.172698763]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1574241725.172744848]: Service available.
[ INFO] [1574241725.172785232]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1574241725.173775416]: Service available.
[ INFO] [1574241725.317197688]: Robot's safety mode is now NORMAL
[ INFO] [1574241725.318116377]: Robot mode is now RUNNING
[INFO] [1574241725.447552]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1574241725.449081]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1574241725.450459]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1574241725.451439]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1574241725.452602]: Loading controller: joint_state_controller
[INFO] [1574241725.453695]: Loading controller: pos_traj_controller
[INFO] [1574241725.462671]: Loading controller: scaled_pos_traj_controller
[INFO] [1574241725.504371]: Controller Spawner: Loaded controllers: pos_traj_controller
[INFO] [1574241725.552416]: Loading controller: speed_scaling_state_controller
[INFO] [1574241725.558484]: Loading controller: force_torque_sensor_controller
[INFO] [1574241725.564435]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1574241725.568487]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
^C[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [1574241828.228064]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1574241828.228706]: Stopping all controllers...
[INFO] [1574241828.228852]: Shutting down spawner. Stopping and unloading controllers...
Interrupt signal (2) received.
[INFO] [1574241828.229510]: Stopping all controllers...
[ERROR] [1574241828.232296058]: Could not stop controller 'pos_traj_controller' since it is not running
[INFO] [1574241828.232577]: Unloading all loaded controllers...
[INFO] [1574241828.232985]: Trying to unload pos_traj_controller
[ERROR] [1574241828.236400130]: Could not stop controller 'scaled_pos_traj_controller' since it is not running
[INFO] [1574241828.236779]: Unloading all loaded controllers...
[INFO] [1574241828.237132]: Trying to unload force_torque_sensor_controller
terminate called without an active exception
[WARN] [1574241828.425136]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[WARN] [1574241828.425533]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

And I tried to press the run bottom in the Tech Pendant, it showed "The connection to the remote PC could not be established".

I didn't change any configuration in the package. If there were something that needed to modify?

The External Control Host IP:192.168.56.1
UR3e IP: 192.168.56.1
My computer IP: 192.168.56.2
I don't know if my settings are correct, so I showed them.

Hoping for your response. Sorry for disturbing you.  And thank you very much.

gavanderhoorn, 2019-11-20:
> Running "roslaunch ur3_e_bringup.launch" error.

The console log you include shows you pressing <kbd>ctrl</kbd>+<kbd>c</kbd> right after the driver reports:

```
[ INFO] [1574245557.636817392]: Robot mode is now RUNNING
```

up until that point there doesn't appear to be any problem.

It would seem to make sense for the External Control on the robot controller to complain it can't connect to the PC if you've just shut everything down.

Csp0920, 2019-11-20:
@gavanderhoorn  I‘m sorry that I didn't get your points.

gavanderhoorn, 2019-11-20:
The log you show, has this:

```
[ INFO] [1574245557.636512869]: Robot's safety mode is now NORMAL
[ INFO] [1574245557.636817392]: Robot mode is now RUNNING
^C[ur_hardware_interface/ur_robot_state_helper-6] killing on exit
[controller_stopper-5] killing on exit
[ros_control_stopped_spawner-4] killing on exit
[ros_control_controller_spawner-3] killing on exit
[ur_hardware_interface-2] killing on exit
[robot_state_publisher-1] killing on exit
Interrupt signal (2) received.
```

that seems to suggest you pressed <kbd>ctrl</kbd>+<kbd>c</kbd> at some point (note the `^C`).

Why did you press <kbd>ctrl</kbd>+<kbd>c</kbd>?

From the output it would seem the driver was started OK, with no errors. At least on the ROS side.

Csp0920, 2019-11-20:
@gavanderhoorn  The program stays at:
[ INFO] [1574245557.636512869]: Robot's safety mode is now NORMAL
[ INFO] [1574245557.636817392]: Robot mode is now RUNNING

And then I press the run bottom in the Tech Pendant, it showed "The connection to remote PC could not be established".

So I press ctrl+c

gavanderhoorn, 2019-11-20:
> The External Control Host IP:192.168.56.1
UR3e IP: 192.168.56.1
My computer IP: 192.168.56.2

As mentioned in the [e-series setup guide](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md), you should enter the IP of the "external PC which will be running the ROS driver" on the *External Control* setup screen.

In your case that would seem to be `192.168.56.2`, not `192.168.56.1` (as that would be the robot's IP address).

Csp0920, 2019-11-20:
@gavanderhoorn  I see what you mean. So i need to make the ip of PC and External Control same.  I will try it tomorrow.

Csp0920, 2019-11-20:
@gavanderhoorn  Thank you for you patient reponse.

Csp0920, 2019-11-22:
@gavaderhoorn @fmauch 
Thanks for your help. Now I can successfully connect my pc with ur3e.
I change the ip from 192.168.56.x to 192.168.1.x. Then it succeed. 
Sorry that I didn't reply in time, and really thanks for your help.

closed 2019-11-22 by Csp0920

tt2cloud, 2020-06-10:
Hello, brother,I recently encountered a same problem.I am trying to connect PC with UR10e. Following the tutorials, I don't able to the UR10e.I downloaded (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git src/Universal_Robots_ROS_Driver) and (https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot) two packages, when I was connecting the PC IP

termial1:roslaunch ur_robot_driver ur10e_bringup.launch
auto-starting new master
process[master]: started with pid [16046]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f9f4466a-aaf6-11ea-87df-c8d9d21823f2
process[rosout-1]: started with pid [16063]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [16080]
process[ur_hardware_interface-3]: started with pid [16081]
process[ros_control_controller_spawner-4]: started with pid [16082]
[ INFO] [1591778833.244461464]: Main thread: SCHED_FIFO OK
[ INFO] [1591778833.244515999]: Main thread priority is 99
process[ros_control_stopped_spawner-5]: started with pid [16097]
[ INFO] [1591778833.254436740]: Initializing dashboard client
[ INFO] [1591778833.259312031]: Connected: Universal Robots Dashboard Server

process[controller_stopper-6]: started with pid [16133]
process[ur_hardware_interface/ur_robot_state_helper-7]: started with pid [16151]
[ INFO] [1591778833.279400014]: Initializing urdriver
[ INFO] [1591778833.279905846]: Checking if calibration data matches connected robot.
[ INFO] [1591778833.280442042]: Producer thread: SCHED_FIFO OK
[ INFO] [1591778833.280469870]: Thread priority is 99
[ INFO] [1591778833.286314783]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1591778833.287298633]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ERROR] [1591778833.342191180]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [TODO Link to documentation] for details.
[INFO] [1591778833.634791]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1591778833.691968]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1591778834.345545407]: Producer thread: SCHED_FIFO OK
[ INFO] [1591778834.345570416]: Thread priority is 99
[ INFO] [1591778834.389224845]: Setting up RTDE communication with frequency 500.000000
[FATAL] [1591778835.405016530]: Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured
[ERROR] [1591778835.405094844]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 16081, exit code 1, cmd /home/yun/catkin_ws7/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/yun/.ros/log/f9f4466a-aaf6-11ea-87df-c8d9d21823f2/ur_hardware_interface-3.log].
log file: /home/yun/.ros/log/f9f4466a-aaf6-11ea-87df-c8d9d21823f2/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[WARN] [1591778835.978961]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1591778835.978991]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_stopper-6] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
terminal2:roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch limited:=true
SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/endeffector/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/default_planner_config: RRTConnectkConfig...
 * /move_group/manipulator/longest_valid_segment_fraction: 0.01
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
[... 99 lines cut ...]
PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver_attempts: 3
 * /rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver_search_resolution: 0.005
 * /rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver_timeout: 0.005

NODES
  /
    rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018 (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

[... cut ...]
[ INFO] [1591778834.483084141]: OpenGl version: 3 (GLSL 1.3).
[ERROR] [1591778835.330058916]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1591778835.330242282]: Loading robot model 'ur10e'...
[ INFO] [1591778835.330292133]: No root/virtual joint specified in SRDF. Assuming fixed joint
The TP show don't connect to the remote PC establish.
Could you help me ,this question has really been bothering me, thanks advance!@fmauch@Csp0920

gavanderhoorn, 2020-06-10:
@tt2cloud: your issue is different (note the RTDE problem reported by the driver).

I would recommend you post it in a new issue.

@fmauch: I've seen this mentioned in PickNikRobotics/picknik_ur5_moveit_config#9 as well.

tt2cloud, 2020-06-10:
> @tt2cloud: your issue is different (note the RTDE problem reported by the driver).
> 
> I would recommend you post it in a new issue.
> 
> @fmauch: I've seen this mentioned in [PickNikRobotics/picknik_ur5_moveit_config#9](https://github.com/PickNikRobotics/picknik_ur5_moveit_config/pull/9) as well.

brother, thank you reply.

gavanderhoorn, 2020-06-10:
> brother, thank you reply.

Please don't call me that.

tt2cloud, 2020-06-10:
Hi～ brother,can you help me solve the problem ------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;G.A.&amp;nbsp;vd.&amp;nbsp;Hoorn&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年6月10日(星期三) 下午5:28
收件人:&nbsp;&quot;UniversalRobots/Universal_Robots_ROS_Driver&quot;<Universal_Robots_ROS_Driver@noreply.github.com&gt;;
抄送:&nbsp;&quot;tt2cloud&quot;<1091350626@qq.com&gt;;&quot;Mention&quot;<mention@noreply.github.com&gt;;
主题:&nbsp;Re: [UniversalRobots/Universal_Robots_ROS_Driver] Can&#39;t connect UR3e with PC (#43)

tt2cloud, 2020-06-11:
> @tt2cloud: your issue is different (note the RTDE problem reported by the driver).
> 
> I would recommend you post it in a new issue.
> 
> @fmauch: I've seen this mentioned in [PickNikRobotics/picknik_ur5_moveit_config#9](https://github.com/PickNikRobotics/picknik_ur5_moveit_config/pull/9) as well.

Hi,brother,Do I need to modify the code, if it needs to be modified, please tell me how to modify it,because I don’t understand it too much,thank you very much!

fmauch, 2020-06-11:
@tt2cloud  I'd also appreciate if you'd open a new issue and if you would kindly stop, calling anybody "brother" here. Thanks.

tt2cloud, 2020-06-12:
> @tt2cloud I'd also appreciate if you'd open a new issue and if you would kindly stop, calling anybody "brother" here. Thanks.

Hi,I am so glad you can answer my questions, I am a chinese student.
I open a new issue that describes my operation steps and settings[(https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/191)]
So, I hope you can answer this question