UniversalRobots/Universal_Robots_ROS_Driver #45: Connection setup failed for :50002

khssnv, 2019-11-25:
# Summary
I have two robots setup and launching two driver nodes on one PC. Launch fails on attempt to run second driver instance.

From error log message (please find at *Issue details* section) seems like it needs this *TODO* implemented: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/c12e39366920b3e28ac8838200c90d2ff9425df3/ur_robot_driver/src/ur/ur_driver.cpp#L84

# Versions
 - ROS Driver version: commit [0bec493](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/0bec493abd41a68dd3b9c580374a6c681d46b75e)
 - Affected Robot Software Version(s): please ask if important
 - Affected Robot Hardware Version(s): please ask if important
 - Robot Serial Number: please ask if important
 - UR+ product(s) installed: please ask if important
 - URCaps Software version(s): please ask if important

# Impact
Impossible to use more then one driver instance on one PC.

# Issue details
Please find driver log error message below

```console
[ INFO] [1574698359.241920363]: Setting up RTDE communication with frequency 500.000000
[ERROR] [1574698360.273018129]: Connection setup failed for :50002
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not bind to server
================================================================================REQUIRED process [green/robot/ur_hardware_interface-2] has died!
process has died [pid 20560, exit code -6, cmd /home/khassanov/Workspace/remy/after_oven_ros/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/khassanov/.ros/log/9777a6be-0f90-11ea-bd6a-c485089fc0e0/green-robot-ur_hardware_interface-2.log].
log file: /home/khassanov/.ros/log/9777a6be-0f90-11ea-bd6a-c485089fc0e0/green-robot-ur_hardware_interface-2*.log
Initiating shutdown!
================================================================================
```

## Use Case and Setup
Setup: two URSim UR5

## Steps to Reproduce
1. Launch 2 instances of URSim,
2. Launch two instances of this package in different namespaces with IP addresses to connect to URSim instances,
3. Attempting to launch second instance of robot driver will produce you the error.

## Expected Behavior
One can use this driver to control a setup with multiple robots.

## Actual Behavior
Second instance of driver fails with an error on launch.

## Workaround Suggestion
Seems like it works if there is another mahine (possible virtual) to launch second driver instance, but I did not check. Another workaround seems possible is to build a copy of this driver with
```cpp
  uint32_t script_sender_port = 50003;
```

fmauch, 2019-11-25:
Yes, this is a design issue and we postponed that for later. As you correctly found out the source lies in having the script sender on a fixed port.

This is nothing currently on our schedule, but we would be happy to assist you in making a PR bringing this up to a ROS parameter. However, it would also be necessary to define the reverse port (currently hard coded to 50001) using a parameter, as well, as they would also be clashing.

Let me know whether you are interested in implementing this, then I can give further details on where to start.

gavanderhoorn, 2019-11-26:
It may be instructive to see how this is done in `ur_modern_driver`: [here](https://github.com/ros-industrial/ur_modern_driver/blob/b210fd03b970ad20c92e8bf410bf34214ff9b809/src/ros/trajectory_follower.cpp).

Specifically: [here](https://github.com/ros-industrial/ur_modern_driver/blob/b210fd03b970ad20c92e8bf410bf34214ff9b809/src/ros/trajectory_follower.cpp#L108-L109).

ShuSewts, 2019-12-04:
I'd like to try implementing this. Please let me know if this solution makes sense,

If it is just a matter of making `reverse_port` and `script_sender_port` within `ur_driver.cpp` into parameters, is the fix just adding a these as part of the method signature for the constructor for the UrDriver and as arguments in the `ur_common.launch` file? Please let me know if I am moving in the right direction.

fmauch, 2019-12-05:
Yes, exactly. As it is currently implemented, you would have to add the parameters to `UrDriver::UrDriver`, which is called here: 

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/fad8c0b34f291280cb03bde791b71ac7a3ba1349/ur_robot_driver/src/ros/hardware_interface.cpp#L208

Before this call, read them from a ROS parameter. This parameter should not be required and default to the currently hardcoded values. Please also add a short docstring directly above the line reading the parameters. We use these to auto-generate the [ROS-Interface documentation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md) using [catkin_doc](https://github.com/fzi-forschungszentrum-informatik/catkin_doc)

Adding this to all the launch files as arguments would then be the final step.

fmauch, 2019-12-16:
This got merged in #57, so I'm closing this.

closed 2019-12-16 by fmauch

ShuSewts, 2020-01-07:
Hi, I've pulled the changes and set up the respective port and script numbers for both UR5 robots. However, I'm currently having an issue where only one robot is executing all the commands. The ur_robot_driver seems to be able to recognize both robots.

khssnv, 2020-01-07:
Did you set correct IP addresses as well?

ShuSewts, 2020-01-07:
Yes, they should both be from the machine I'm controlling the UR5s from right?

khssnv, 2020-01-07:
Just to clarify what we call controller, it's a contol box connected to robot or a (virtual) machine where you launch `ursim`. For this driver argument `robot_ip` should be an IP address of such a robot controller. For instance, if I have a PC (with ROS and this driver installed) on `192.168.1.10` and two robot controllers with `192.168.1.11` and `192.168.1.12`, I will set `192.168.1.11` and default ports for first driver instance and `192.168.1.12` and some not default ports for second driver instance. Could you share your launch files if there is an issue still?

ShuSewts, 2020-01-07:
Ok, so the robot_ip should be the ip addresses of each robot_ip in the launch file. And the external control on each ur5 panel should be the ip address of the PC yes? Thats what I have.

`<?xml version="1.0"?>
<launch>

 <group ns="frank">
   <arg name="limited" default="false"/>
   <arg name="robot_ip" default="169.254.198.38"/>
   <arg name="kinematics_config" default="$(find ur_robot_driver)/config/frank_calibration.yaml"/>
   <arg name="tf_prefix" default="frank_"/>

  <include file="$(find ur_bringup)/launch/ur5_bringup.launch">
   <arg name="limited" value="$(arg limited)"/>
   <arg name="robot_ip" value="$(arg robot_ip)"/>
   <arg name="kinematics_config" value="$(arg kinematics_config)"/>
   <arg name="reverse_port" value="50003"/>
   <arg name="script_sender_port" value="50004"/>  
   <arg name="tf_prefix" value="frank_"/>
  </include>
 </group>

 <group ns="klaus">
   <arg name="limited" default="false"/>
   <arg name="robot_ip" default="169.254.198.35"/>
   <arg name="kinematics_config" default="$(find ur_robot_driver)/config/klaus_calibration.yaml"/>
   <arg name="tf_prefix" default="klaus_"/>

  <include file="$(find ur_robot_driver)/launch/ur5_bringup.launch"> 
   <arg name="limited" value="$(arg limited)"/>
   <arg name="robot_ip" value="$(arg robot_ip)"/>
   <arg name="kinematics_config" value="$(arg kinematics_config)"/>
   <arg name="reverse_port" value="50001"/>
   <arg name="script_sender_port" value="50002"/>
   <arg name="tf_prefix" value="klaus_"/>
  </include>
 </group>

</launch>
`

gavanderhoorn, 2020-01-07:
Could you please show us what `rqt_graph` looks like for you after you've launched everything?

Just make a screenshot of it and post that here.

ShuSewts, 2020-01-07:
[image]

gavanderhoorn, 2020-01-07:
Please don't hide anything next time, but from this it looks like both `frank/move_group` and `klaus/move_group` publish to both `ur_hardware_interface` instances. That seems strange to me.

Could you please provide a screenshot of `rqt_graph` without hiding anything? It's hard to see what is really going on in this one.

ShuSewts, 2020-01-07:
[image]
 Sorry, here's the full rqt graph

ShuSewts, 2020-01-07:
[image]

gavanderhoorn, 2020-01-07:
It's ok to hide the `Debug` category of topics. But I'd like to see not just *Nodes only*, but everything.

ShuSewts, 2020-01-07:
Would that be Nodes/Topics (all)?

gavanderhoorn, 2020-01-07:
Could you also describe your MoveIt configuration? Specifically the controller configuration? It's typically stored in a file called `controllers.yaml` (or similar).

ShuSewts, 2020-01-07:
```

controller_list:
  - name: "/frank/scaled_pos_traj_controller"
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
  - name: "/klaus/scaled_pos_traj_controller"
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint

```

gavanderhoorn, 2020-01-07:
So you have a single MoveIt configuration package, with a single `controllers.yaml`? And you have both `move_group` instances connect to both instances of the driver?

ShuSewts, 2020-01-07:
Yes

gavanderhoorn, 2020-01-07:
Do you  need both `move_groups` to send goals to both driver instances? Or just to "their" instance?

ShuSewts, 2020-01-07:
I think it should be to their respective instances - I'm trying to just move one at a time at this moment.

gavanderhoorn, 2020-01-07:
Then you should change your `controllers.yaml`.

It's probably easiest to create two and then load the one that is needed in their respective namespaces.

ShuSewts, 2020-01-07:
I've created a klaus_controller.yaml and frank_controller.yaml. Now instead of Frank, Klaus is executing all commands. My rqt graph looks like this.
[image]

ShuSewts, 2020-01-08:
I don't think the issue is with separating the controllers - I've tried making both instances of the ur5_moveit_config point to controllers.yaml that now looks like this:
`controller_list:
  - name: "scaled_pos_traj_controller"
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint`

and now the move_group instances don't publish to both ur_hardware_interface instances. I'm still having an issue where one robot is executing all commands, however. My rqt_graph now looks like this (let me know if I need to move this discussion somewhere else as I'm guessing it's not a Universal_Robots_ROS_Driver issue)
[image]

fmauch, 2020-01-08:
You could check by listening to the driver's `goal` topics

```
/frank/scaled_pos_traj_controller/follow_joint_trajectory/goal
/klaus/scaled_pos_traj_controller/follow_joint_trajectory/goal
```
If all commands show up on klaus, it probably means (according to your rosgraph) that they also all go through `/klaus/move_group` which is no error of this driver. If they show up on their respective goal topics, but are still executed by one machine this means that there is something fishy going on.

ShuSewts, 2020-01-08:
Thanks for the suggestion, 
Calling `rostopic list /frank/scaled_pos_traj_controller/follow_joint_trajectory/goal ` correctly returns  /frank/move_group as the only publisher and `rostopic list /klaus/scaled_pos_traj_controller/follow_joint_trajectory/goal ` correctly returns  /klaus/move_group as the only publisher.

However, echoing both `/frank/scaled_pos_traj_controller/follow_joint_trajectory/goal` and 
`/klaus/scaled_pos_traj_controller/follow_joint_trajectory/goal` and subsequently giving each robot a goal sends two messages through `/klaus/scaled_pos_traj_controller/follow_joint_trajectory/goal` and none through `/frank/scaled_pos_traj_controller/follow_joint_trajectory/goal`

fmauch, 2020-01-08:
This means that all commands sent come from `/klaus/move_group`. So you should search for the error in the MoveIt setup / usage. Are you sure that the commands are sent to different robots in the first place? I would further trace the commands to where they come from (I expect from `/klaus/move_group/goal`) and see who publishes on that and so on...

ShuSewts, 2020-01-08:
> Are you sure that the commands are sent to different robots in the first place?

That was it, along with changing the controller.yaml file. Thanks

gavanderhoorn, 2020-01-08:
Yes, this was where I was going as well.

The connections in the `rqt_graph` screenshot made me suspicious.

@ShuSewts: it would be appreciated if you could now go back to your ROS Answers post and answer your own question.