UniversalRobots/Universal_Robots_ROS_Driver #46: Something wrong with my install dependecies

bodon97, 2019-11-25:
Hi!

I'm using Ubuntu 18.04 LTS with ROS melodic. I'm planning to use this driver with a UR3e robotic arm.
I'm having trouble with the install dependencies.
For this command sudo apt update -qq
 I get the following message:
450 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/bionic/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Some index files failed to download. They have been ignored, or old ones used instead.

And for this line rosdep install --from-path src --ignore-src -y

This error:

executing command [sudo -H apt-get install -y ros-melodic-velocity-controllers]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.18.0-17 linux-headers-4.18.0-17-generic
  linux-image-4.18.0-17-generic linux-modules-4.18.0-17-generic
  linux-modules-extra-4.18.0-17-generic
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-melodic-velocity-controllers
0 upgraded, 1 newly installed, 0 to remove and 450 not upgraded.
Need to get 70,7 kB of archives.
After this operation, 293 kB of additional disk space will be used.
Err:1 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-velocity-controllers amd64 0.15.0-0bionic.20190329.171037
  404  Not Found [IP: 64.50.236.52 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-velocity-controllers/ros-melodic-velocity-controllers_0.15.0-0bionic.20190329.171037_amd64.deb  404  Not Found [IP: 64.50.236.52 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-melodic-velocity-controllers] failed

Thank you for your help!

gavanderhoorn, 2019-11-25:
Please don't cross-post. I've already responded to your comment on your ROS Answers Q&A: [Missing joint trajectory controller](https://answers.ros.org/question/338321).

It's very unlikely your latest problem has anything to do with the packages in this repository.

bodon97, 2019-11-25:
Sorry! You solved my problem thank you!

closed 2019-11-25 by bodon97