UniversalRobots/Universal_Robots_ROS_Driver #47: rqt_joint_trajectory_controller crash

bodon97, 2019-11-26:
# Summary

I created my own launch file following the tutorial. It runs perfectly. Then I open a new terminal source it. When I try to launch the rqt it shows up. I can set the controller manager ns and the controller, but when I hit the power button it crashes with the following error message. I use Ubuntu 18.04 LTS, ROS melodic

cmd: rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller

error: Traceback (most recent call last):
  File "/home/bodon/catkin_ws/src/ros_controllers/rqt_joint_trajectory_controller/src/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 416, in _update_cmd_cb
    pos = self._joint_pos[name]['position']
KeyError: 'position'

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s): PolyScope 5.5.1
 - Affected Robot Hardware Version(s): UR3e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 4.1.4

gavanderhoorn, 2019-11-26:
I've often seen this error when `robot_description` is either empty or in a namespace `rqt_joint_trajectory_controller` has no access to.

bodon97, 2019-11-28:
Here's my launch file: 

<?xml version="1.0"?>
<launch>
  <arg name="debug" default="false" doc="Debug flag that will get passed on to ur_common.launch"/>
  <arg name="robot_ip" default="192.168.0.113" doc="IP address by which the robot can be reached."/>
  <arg name="tf_prefix" default="" doc="tf_prefix used for the robot."/>
  <arg name="controllers" default="joint_state_controller scaled_pos_traj_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>
  <arg name="stopped_controllers" default="pos_traj_controller" doc="Controllers that are initally loaded, but not started."/>
  <arg name="controller_config_file" default="$(find ur_robot_driver)/config/ur3e_controllers.yaml" doc="Config file used for defining the ROS-Control controllers."/>
  <arg name="robot_description_file" default="$(find ur_e_description)/launch/ur3e_upload.launch" doc="Robot description launch file."/>
  <arg name="kinematics_config" default="$(find bme_ur_launch)/etc/ur3e_calibration.yaml" doc="Kinematics config file used for calibration correction. This will be used to verify the robot's calibration is matching the robot_description."/>
  <arg name="use_tool_communication" default="false" doc="On e-Series robots tool communication can be enabled with this argument"/>
  <arg name="tool_voltage" default="0" doc="Tool voltage set at the beginning of the UR program. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_parity" default="0" doc="Parity configuration used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_baud_rate" default="115200" doc="Baud rate used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_stop_bits" default="1" doc="Number of stop bits used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_rx_idle_chars" default="1.5" doc="Number of idle chars in RX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tx_idle_chars" default="3.5" doc="Number of idle chars in TX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_device_name" default="/tmp/ttyUR" doc="Local device name used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tcp_port" default="54321" doc="Port on which the robot controller publishes the tool comm interface. Only used, when `use_tool_communication` is set to true."/>
  <arg name="limited" default="false" doc="Use the description in limited mode (Every axis rotates from -PI to PI)"/>
  <arg name="headless_mode" default="false" doc="Automatically send URScript to robot to execute. On e-Series this does require the robot to be in 'remote-control' mode. With this, the URCap is not needed on the robot."/>

  <include file="$(find ur_robot_driver)/launch/ur_common.launch">
    <arg name="debug" value="$(arg debug)"/>
    <arg name="use_tool_communication" value="$(arg use_tool_communication)"/>
    <arg name="controller_config_file" value="$(arg controller_config_file)"/>
    <arg name="robot_description_file" value="$(arg robot_description_file)"/>
    <arg name="kinematics_config" value="$(arg kinematics_config)"/>
    <arg name="robot_ip" value="$(arg robot_ip)"/>
    <arg name="limited" value="$(arg limited)"/>
    <arg name="tf_prefix" value="$(arg tf_prefix)"/>
    <arg name="controllers" value="$(arg controllers)"/>
    <arg name="stopped_controllers" value="$(arg stopped_controllers)"/>
    <arg name="headless_mode" value="$(arg headless_mode)"/>
    <arg name="tool_voltage" value="$(arg tool_voltage)"/>
    <arg name="tool_parity" value="$(arg tool_parity)"/>
    <arg name="tool_baud_rate" value="$(arg tool_baud_rate)"/>
    <arg name="tool_stop_bits" value="$(arg tool_stop_bits)"/>
    <arg name="tool_rx_idle_chars" value="$(arg tool_rx_idle_chars)"/>
    <arg name="tool_tx_idle_chars" value="$(arg tool_tx_idle_chars)"/>
    <arg name="tool_device_name" value="$(arg tool_device_name)"/>
    <arg name="tool_tcp_port" value="$(arg tool_tcp_port)"/>
  </include>

</launch>

Here's the rosparam list:

/controller_stopper/consistent_controllers
/force_torque_sensor_controller/publish_rate
/force_torque_sensor_controller/type
/hardware_control_loop/loop_hz
/hardware_interface/joints
/joint_state_controller/publish_rate
/joint_state_controller/type
/pos_traj_controller/action_monitor_rate
/pos_traj_controller/constraints/elbow_joint/goal
/pos_traj_controller/constraints/elbow_joint/trajectory
/pos_traj_controller/constraints/goal_time
/pos_traj_controller/constraints/shoulder_lift_joint/goal
/pos_traj_controller/constraints/shoulder_lift_joint/trajectory
/pos_traj_controller/constraints/shoulder_pan_joint/goal
/pos_traj_controller/constraints/shoulder_pan_joint/trajectory
/pos_traj_controller/constraints/stopped_velocity_tolerance
/pos_traj_controller/constraints/wrist_1_joint/goal
/pos_traj_controller/constraints/wrist_1_joint/trajectory
/pos_traj_controller/constraints/wrist_2_joint/goal
/pos_traj_controller/constraints/wrist_2_joint/trajectory
[... 57 lines cut ...]
/ur_hardware_interface/kinematics/wrist_1/yaw
/ur_hardware_interface/kinematics/wrist_1/z
/ur_hardware_interface/kinematics/wrist_2/pitch
/ur_hardware_interface/kinematics/wrist_2/roll
[... cut ...]
/ur_hardware_interface/kinematics/wrist_3/yaw
/ur_hardware_interface/kinematics/wrist_3/z
/ur_hardware_interface/output_recipe_file
/ur_hardware_interface/robot_ip
/ur_hardware_interface/script_file
/ur_hardware_interface/tf_prefix
/ur_hardware_interface/tool_baud_rate
/ur_hardware_interface/tool_parity
/ur_hardware_interface/tool_rx_idle_chars
/ur_hardware_interface/tool_stop_bits
/ur_hardware_interface/tool_tx_idle_chars
/ur_hardware_interface/tool_voltage
/ur_hardware_interface/use_tool_communication

bodon97, 2019-11-28:
Launch file again:
[ur3e.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/3901060/ur3e.zip)

fmauch, 2019-11-28:
I'm sorry, but the provided zip file does not contain any launch files, but some part of Clang documentation as it seems.

bodon97, 2019-11-28:
Sorry wrong file, my bad.
here it is:
[ur3e.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/3901084/ur3e.zip)

gavanderhoorn, 2019-11-28:
Launch file:

```xml
<?xml version="1.0"?>
<launch>
  <arg name="debug" default="false" doc="Debug flag that will get passed on to ur_common.launch"/>
  <arg name="robot_ip" default="192.168.0.113" doc="IP address by which the robot can be reached."/>
  <arg name="tf_prefix" default="" doc="tf_prefix used for the robot."/>
  <arg name="controllers" default="joint_state_controller scaled_pos_traj_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>
  <arg name="stopped_controllers" default="pos_traj_controller" doc="Controllers that are initally loaded, but not started."/>
  <arg name="controller_config_file" default="$(find ur_robot_driver)/config/ur3e_controllers.yaml" doc="Config file used for defining the ROS-Control controllers."/>
  <arg name="robot_description_file" default="$(find ur_e_description)/launch/ur3e_upload.launch" doc="Robot description launch file."/>
  <arg name="kinematics_config" default="$(find bme_ur_launch)/etc/ur3e_calibration.yaml" doc="Kinematics config file used for calibration correction. This will be used to verify the robot's calibration is matching the robot_description."/>
  <arg name="use_tool_communication" default="false" doc="On e-Series robots tool communication can be enabled with this argument"/>
  <arg name="tool_voltage" default="0" doc="Tool voltage set at the beginning of the UR program. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_parity" default="0" doc="Parity configuration used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_baud_rate" default="115200" doc="Baud rate used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_stop_bits" default="1" doc="Number of stop bits used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_rx_idle_chars" default="1.5" doc="Number of idle chars in RX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tx_idle_chars" default="3.5" doc="Number of idle chars in TX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_device_name" default="/tmp/ttyUR" doc="Local device name used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tcp_port" default="54321" doc="Port on which the robot controller publishes the tool comm interface. Only used, when `use_tool_communication` is set to true."/>
  <arg name="limited" default="false" doc="Use the description in limited mode (Every axis rotates from -PI to PI)"/>
  <arg name="headless_mode" default="false" doc="Automatically send URScript to robot to execute. On e-Series this does require the robot to be in 'remote-control' mode. With this, the URCap is not needed on the robot."/>

  <include file="$(find ur_robot_driver)/launch/ur_common.launch">
    <arg name="debug" value="$(arg debug)"/>
    <arg name="use_tool_communication" value="$(arg use_tool_communication)"/>
    <arg name="controller_config_file" value="$(arg controller_config_file)"/>
    <arg name="robot_description_file" value="$(arg robot_description_file)"/>
    <arg name="kinematics_config" value="$(arg kinematics_config)"/>
    <arg name="robot_ip" value="$(arg robot_ip)"/>
    <arg name="limited" value="$(arg limited)"/>
    <arg name="tf_prefix" value="$(arg tf_prefix)"/>
    <arg name="controllers" value="$(arg controllers)"/>
    <arg name="stopped_controllers" value="$(arg stopped_controllers)"/>
    <arg name="headless_mode" value="$(arg headless_mode)"/>
    <arg name="tool_voltage" value="$(arg tool_voltage)"/>
    <arg name="tool_parity" value="$(arg tool_parity)"/>
    <arg name="tool_baud_rate" value="$(arg tool_baud_rate)"/>
    <arg name="tool_stop_bits" value="$(arg tool_stop_bits)"/>
    <arg name="tool_rx_idle_chars" value="$(arg tool_rx_idle_chars)"/>
    <arg name="tool_tx_idle_chars" value="$(arg tool_tx_idle_chars)"/>
    <arg name="tool_device_name" value="$(arg tool_device_name)"/>
    <arg name="tool_tcp_port" value="$(arg tool_tcp_port)"/>
  </include>

</launch>
```

fmauch, 2019-11-28:
Your launchfile seems to be the same as the included one except for the default values for the IP address and the kinematics config. Note that it might be better to actually include the base launch file and providing the arguments only:

```xml
<?xml version="1.0"?>
<launch>
  <arg name="robot_ip" default="192.168.0.113"/>
  <arg name="kinematics_config" default="$(find bme_ur_launch)/etc/ur3e_calibration.yaml"/>

  <include file="$(find ur_robot_driver)/launch/ur3e_bringup.launch">
    <arg name="robot_ip" value="$(arg robot_ip)"/>
    <arg name="kinematics_config" value="$(arg kinematics_config)"/>
  </include>
</launch>
```

This way your local launch files will remain small and if new features are added to the driver's launch files they will be included directly.

But back to topic: There seems nothing to be obviously wrong with your setup. This means, we'll have to do further debugging. There could be something wrong in the calibration, thereby your kinematics config resulting in a faulty robot description. Could you please also provide

 - The robot description as it is on the parameter server
`rosparam get -p /robot_description `
 - The kinematics config file `bme_ur_launch/etc/ur3e_calibration.yaml`

bodon97, 2019-11-28:
I'll change my launch file, thank you for your suggestion. Here are the requested lines

Calibration: 

kinematics:
  shoulder:
    x: 0
    y: 0
    z: 0.1519767680918936
    roll: -0
    pitch: 0
    yaw: -3.294372687926567e-08
  upper_arm:
    x: -0.0001168180396615669
    y: 0
    z: 0
    roll: 1.571360681555981
    pitch: 0
    yaw: -8.772150944853642e-06
  forearm:
    x: -0.2437440609883655
    y: 0
    z: 0
    roll: 0.009100361320476999
    pitch: 0.001028886812629192
    yaw: -3.797928354459098e-06
  wrist_1:
    x: -0.2133213123756582
    y: -0.0004029569170161354
    z: 0.1310209377034569
    roll: 0.003075505744278148
    pitch: 0.0006792353649060244
    yaw: 3.859726467671596e-06
  wrist_2:
    x: 0.0002205030541780027
    y: -0.08535930261550119
    z: -3.999801892927933e-05
    roll: 1.571264910941864
    pitch: 0
    yaw: 5.470283068772529e-05
  wrist_3:
    x: -7.151457348874671e-05
    y: 0.09205381461236188
    z: -0.0001664506162865783
    roll: 1.568988140620516
    pitch: 3.141592653589793
    yaw: -3.141592642624424
  hash: calib_15915879218217949005

robot_description:

[robot_description_ur3e.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/3902196/robot_description_ur3e.zip)

fmauch, 2019-11-28:
I really can't get behind what's going on. It might be necessary to throw in full debugging: 

 * A `rosparam dump` (see [rosparam wiki entry](http://wiki.ros.org/rosparam#rosparam_dump)) of all parameters might help
 * A [rqt_graph](http://wiki.ros.org/rqt_graph) export of everything running (When the `rqt_joint_trajectory_controller` is actually running)
 * A short [rosbag](http://wiki.ros.org/rosbag/Commandline#record) to see which messages are being published

bodon97, 2019-12-05:
Rosbag:

Usage: rosbag <subcommand> [options] [args]

A bag is a file format in ROS for storing ROS message data. The rosbag command can record, replay and manipulate bags.

Available subcommands:
   check  	Determine whether a bag is playable in the current system, or if it can be migrated.
   compress  	Compress one or more bag files.
   decompress  	Decompress one or more bag files.
   decrypt  	Decrypt one or more bag files.
   encrypt  	Encrypt one or more bag files.
   filter  	Filter the contents of the bag.
   fix  	Repair the messages in a bag file so that it can be played in the current system.
   help  
   info  	Summarize the contents of one or more bag files.
   play  	Play back the contents of one or more bag files in a time-synchronized fashion.
   record  	Record a bag file with the contents of specified topics.
   reindex  	Reindexes one or more bag files.

For additional information, see http://wiki.ros.org/rosbag

[image]

[rosparam dump.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/3926032/rosparam.dump.zip)

fmauch, 2019-12-06:
For creating the rosbag, please do the following:

```
rosbag record -a --duration 10 -O driver_data.bag
```
This will create a file named `driver_data.bag` in the directory where you are running it from. To see, whether everything worked as expected, you can check the created file:

```
rosbag info driver_data.bag
```

The output should look similar to this:

```
path:        driver_data.bag
version:     2.0
duration:    10.0s
start:       Dec 06 2019 09:19:31.14 (1575620371.14)
end:         Dec 06 2019 09:19:41.09 (1575620381.09)
size:        7.0 MB
messages:    31707
compression: none [9/9 chunks]
types:       actionlib_msgs/GoalStatusArray              [8b2b82f13216d0a8ea88bd3af735e619]
             control_msgs/JointTrajectoryControllerState [10817c60c2486ef6b33e97dcd87f4474]
             geometry_msgs/WrenchStamped                 [d78d3cb249ce23087ade7e7d0c40cfa7]
             rosgraph_msgs/Log                           [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/JointState                      [3066dcd76a6cfaef579bd0f34173e9fd]
             std_msgs/Bool                               [8b94c1b53db61fb6aed406028ad6332a]
             std_msgs/Float64                            [fdb28210bfa9d7c91146260178d9a584]
             tf2_msgs/TFMessage                          [94810edda583a504dfda3829e70d7eec]
             ur_dashboard_msgs/RobotMode                 [73b72d58742c4889c47118411b03a3b4]
             ur_dashboard_msgs/SafetyMode                [5da08725a63d4237bad689481131a84b]
             ur_msgs/IOStates                            [3033784e7041da89491b97cc4c1105b5]
             ur_msgs/ToolDataMsg                         [404fc266f37d89f75b372d12fa94a122]
topics:      /joint_states                                                4446 msgs    : sensor_msgs/JointState
             /pos_traj_controller/follow_joint_trajectory/status            50 msgs    : actionlib_msgs/GoalStatusArray
             /rosout                                                        16 msgs    : rosgraph_msgs/Log                           (6 connections)
             /scaled_pos_traj_controller/follow_joint_trajectory/status     50 msgs    : actionlib_msgs/GoalStatusArray
             /scaled_pos_traj_controller/state                            4435 msgs    : control_msgs/JointTrajectoryControllerState
             /speed_scaling_factor                                        4428 msgs    : std_msgs/Float64
             /tf                                                          4907 msgs    : tf2_msgs/TFMessage                          (2 connections)
             /tf_static                                                      1 msg     : tf2_msgs/TFMessage
             /ur_hardware_interface/io_states                             4445 msgs    : ur_msgs/IOStates
             /ur_hardware_interface/robot_mode                               1 msg     : ur_dashboard_msgs/RobotMode
             /ur_hardware_interface/robot_program_running                    1 msg     : std_msgs/Bool
             /ur_hardware_interface/safety_mode                              1 msg     : ur_dashboard_msgs/SafetyMode
             /ur_hardware_interface/set_mode/status                         49 msgs    : actionlib_msgs/GoalStatusArray
             /ur_hardware_interface/tool_data                             4448 msgs    : ur_msgs/ToolDataMsg
             /wrench
```

Especially, the number of messages on each topic should be in the same magnitude.

Then, please attach the generated `driver_data.bag` file. (You might need to zip it first)

fmauch, 2020-09-25:
Closing this in ~1week if no updates arise.

fmauch, 2020-11-16:
Closing due to inactivity. Feel free to comment or reopen if anything new comes up.

closed 2020-11-16 by fmauch