UniversalRobots/Universal_Robots_ROS_Driver #51: Add documentation about different modes

fmauch, 2019-11-27:
We currently have two modes: "Normal" mode which requires the URCap being installed on the robot and the "External Control" program node running inside a program. The "headless" mode on the other side sends the primary URScript for execution to the robot directly.

An orthogonal mode to those two is remote-control mode. e-Series robots have to be put into that mode to make certain features such as sending script code and most of the dashboard commands work.

All this is not documented all too well currently.

joaocabogon, 2020-02-03:
How can one put the robot in remote-control mode without the Teach-Pendant?

fmauch, 2020-02-04:
You can't do that. The point of the remote control mode is to have the single point of control on the robot. This single point of control can decide to allow control from a remote point, but it has to be the tp allowing this.
........................................................
M.Sc. Felix Exner
Wissenschaftlicher Mitarbeiter | Research Scientist
Intelligent Systems and Production Engineering (ISPE)Interaktive Diagnose- und 
Servicesysteme (IDS)

FZI Forschungszentrum Informatik
Haid-und-Neu-Str. 10–14 
76131 Karlsruhe, Germany
Tel.: +49 721 9654-386
Fax: +49 721 9654-387

exner@fzi.de
www.fzi.de | www.youtube.com/FZIchannel
www.fzi.de/mitarbeiter/exner

.........................................................
FZI Forschungszentrum Informatik
Stiftung des bürgerlichen Rechts
Stiftung Az: 14-0563.1 Regierungspräsidium Karlsruhe
Vorstand: Prof. Dr. Andreas Oberweis, Jan Wiesenberger, Prof. Dr.-Ing. J. 
Marius Zöllner
Vorsitzender des Kuratoriums: Ministerialdirigent Günther Leßnerkraus
.........................................................

PR #179 (2020-05-20): Improve documentation
This MR aims at collecting a couple of documentation improvements.

Closes #140 

Closes #38  

Closes #51
closed 2020-05-20 by fmauch