UniversalRobots/Universal_Robots_ROS_Driver #52: 1MBaud not a supported rate?

gavanderhoorn, 2019-11-28:
Here:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/b531d2501b19a6f4a43b191085acb99b4f907c8f/ur_robot_driver/include/ur_robot_driver/ur/tool_communication.h#L231

it looks like there is a typo in the rate for 1MBaud. Unless 10MBaud is actually supposed to be there, I believe there is a `0` too many.

`10 000 000` vs `2 000 000`.

Scientific notation (`1e6`,  `2e6`, etc) could be used to avoid this.

gavanderhoorn, 2019-11-28:
Looked at this because of [Cannot launch /universal_robots_ROS_driver with custom tool](https://answers.ros.org/question/338764) on ROS Answers.

fmauch, 2019-11-28:
Indeed, thanks for reporting this and proposing the scientific notation approach.

gavanderhoorn, 2019-11-28:
I can submit a PR to fix this, but wanted to make sure I'd understood correctly what the intent was of that array.

fmauch, 2019-11-28:
Actually, scientific notation doesn't work in the initializer list, as this produces a double value...

gavanderhoorn, 2019-11-28:
You'll need to include a cast.

But I believe it's worth it, as it makes it much easier to spot 'too many zeros'.

fmauch, 2019-11-28:
The simplest solution I can come up with:
```c++
  const std::set<uint32_t> baud_rates_allowed_{ 9600,
                                                19200,
                                                38400,
                                                57600,
                                                115200,
                                                static_cast<uint32_t>(1e6),
                                                static_cast<uint32_t>(2e6),
                                                static_cast<uint32_t>(5e6) };
```

gavanderhoorn, 2019-11-28:
Yes, that would be ok.

Although I would probably use:

```c++
std::uint32_t(1e6), ..
```

fmauch, 2019-11-28:
I actually prefer static casts over cstyle casts, so I'll stick with that one.

gavanderhoorn, 2019-11-28:
Fine by me :)

PR #53 (2019-11-28): Fix faulty 1MBaud rate
It actually had a 0 too much. We use scientific notation to make this more clearly visible in future.

Fixes #52
closed 2019-11-28 by fmauch