UniversalRobots/Universal_Robots_ROS_Driver #54: Missing imaginary_driver pkg

kerliang0801, 2019-11-29:
Hi,

First and foremost, thanks for the amazing work here. I'm trying to follow the guide [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/setup_tool_communication.md) to setup tool communication on an ur10e 

I might be missing something but where exactly is "imaginary_driver"?

fmauch, 2019-11-29:
:-)

This is just supposed to be the ros driver of the device connected to the robot's RS485 communication. It might be a gripper, a sensor or any other device communicating over RS485.

Perhaps, something like
```
rosrun <device_driver_package> <device_driver_node> device:=/tmp/ttyUR
```
Would be less misleading?

kerliang0801, 2019-11-29:
Ahh, understood. Now, that was quite a dumb ques to ask. Thanks for replying!

closed 2019-11-29 by kerliang0801

fmauch, 2019-11-29:
Nothing is a really dumb question to ask. If things are unclear from the docs, we might need to improve the docs. Which is why I suggested the change above. Would that improve the docs in your perspective?

kerliang0801, 2019-11-29:
Yes, of course, it's definitely much clearer. However, I just re-read the docs. It's more of my fault for skipping the most important descriptive first paragraph (out of pure clumsiness and laziness , sorry!).