UniversalRobots/Universal_Robots_ROS_Driver #55: Not able to control the UR10e, issue with controllers

tanyouliang95, 2019-12-04:
Hi, Just recently I started to use the new ur driver for my ur10e. I followed the steps of setting up the UR robot driver for UR10e. After installing the `external control.urcap` and also a bunch of relevant ROS packages, I am able to run the `ur_robot_driver`'s ur10e bringup. Iam able to visualize the current robot state with RVIz. 

However the problem comes when I send in a planned trajectory to the `ur_robot_driver` from moveit (melodic). My "ros PC" is not able to control the robot. Here's the printout.

**These are the printouts during startup**
```
[INFO] [1575439049.914691]: Loading controller: speed_scaling_state_controller
[INFO] [1575439049.922537]: Loading controller: force_torque_sensor_controller
[ERROR] [1575439049.925483887]: Could not load controller 'force_torque_sensor_controller' because controller type 'force_torque_sensor_controller/ForceTorqueSensorController' does not exist.
[ERROR] [1575439049.925529312]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ INFO] [1575439049.948150506]: Robot's safety mode is now NORMAL
[ INFO] [1575439049.949624652]: Robot mode is now RUNNING
[ERROR] [1575439050.925985]: Failed to load force_torque_sensor_controller
[INFO] [1575439050.927156]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller
[INFO] [1575439050.930308]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller
[ INFO] [1575439051.858623603]: Loading robot model 'ur10e'...
[ WARN] [1575439051.858674367]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1575439051.858696881]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1575439051.898240782]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_dp2workcell_NUC7i5BNH_10527_4730851679228347204/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1575439052.027892145]: Starting planning scene monitor
[ INFO] [1575439052.030964174]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1575439052.032057688]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1575439053.092445076]: Waiting for /follow_joint_trajectory to come up
[ INFO] [1575439057.033862535]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-melodic-moveit-ros-planning-1.0.2/planning_scene_monitor/src/planning_scene_monitor.cpp:495
[ INFO] [1575439057.651705875]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ WARN] [1575439059.092661121]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1575439064.552758494]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[ERROR] [1575439065.092905191]: Action client not connected: /follow_joint_trajectory

```

**These are the printout when a trajectory srv is received**
```
[ INFO] [1575437926.650551896]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1575437926.650757552]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1575437926.653505540]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1575437926.653893412]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1575437926.667716032]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1575437926.669611105]: RRTConnect: Created 7 states (3 start + 4 goal)
[ INFO] [1575437926.669743583]: ParallelPlan::solve(): Solution found by one or more threads in 0.022276 seconds
[ INFO] [1575437926.671988970]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1575437926.688229656]: RRTConnect: Created 7 states (3 start + 4 goal)
[ INFO] [1575437926.688540964]: ParallelPlan::solve(): Solution found by one or more threads in 0.018659 seconds
[ INFO] [1575437926.707053023]: SimpleSetup: Path simplification took 0.018263 seconds and changed from 3 to 2 states
[ INFO] [1575437926.710855373]:  [Arm Controller: /] Executing Joint Space Motion...
[ INFO] [1575437926.713613360]: Execution request received
[ INFO] [1575437926.713994353]: Returned 0 controllers in list
[ERROR] [1575437926.714121772]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
[ERROR] [1575437926.714171562]: Known controllers and their joints:

[ INFO] [1575437926.714858177]: ABORTED: Solution found but controller failed during execution
[ERROR] [1575437926.715035262]:  Failed in executing planned motion path
[ERROR] [1575437926.715196832]: 

```

While follow the steps, the problem here seems to be on the ros controller node. Also, fyi i used my same code and it works while using the depreciated [ur_modern_driver](https://github.com/AdmiralWall/ur_modern_driver) PR with minimal setup, 

Hope someone can help me with this issue. Thanks!!!

fmauch, 2019-12-04:
The output looks like your installation is missing a couple of controllers. As the output states, could you please post the output of
```
rosservice call controller_manager/list_controller_types
```

Just to clarify: During setup you executed the command
```
rosdep install --from-path src --ignore-src -y
```
is this correct?

I'm asking, because for instance the `force_torque_sensor_controller` seems to be missing in your installation. However, it should get installed by the `rosdep install` command above.

iretiayo, 2019-12-05:
I have the same problem. I confirmed that I have all the required dependencies.

```
 rosservice call controller_manager/list_controller_types

types: [diff_drive_controller/DiffDriveController, effort_controllers/JointEffortController,
  effort_controllers/JointGroupEffortController, effort_controllers/JointGroupPositionController,
  effort_controllers/JointPositionController, effort_controllers/JointTrajectoryController,
  effort_controllers/JointVelocityController, force_torque_sensor_controller/ForceTorqueSensorController,
  joint_state_controller/JointStateController, pos_vel_acc_controllers/JointTrajectoryController,
  pos_vel_controllers/JointTrajectoryController, position_controllers/JointGroupPositionController,
  position_controllers/JointPositionController, position_controllers/JointTrajectoryController,
  position_controllers/ScaledJointTrajectoryController, ur_controllers/SpeedScalingStateController,
  velocity_controllers/JointTrajectoryController]
base_classes: ['controller_interface::ControllerBase', 'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase']

```
The [README](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#quick-start) says: To control the robot using ROS, use the action server on `/scaled_pos_traj_controller/follow_joint_trajectory`. Not sure what this statement means but looks like it might be related.

Any thoughts on how to fix the issue?

iretiayo, 2019-12-05:
I fixed it using the Troubleshooting section in the ReadMe:

- install the URCap on the robot and don't forget the last step of starting the External Control program on the teach pendant.
- for moveit, set the action controller namespace (e.g. [here](https://github.com/fmauch/universal_robot/blob/kinetic-devel/ur5_moveit_config/config/controllers.yaml#L3)) to `scaled_pos_traj_controller/follow_joint_trajectory`

fmauch, 2019-12-06:
Yes, this is definitely necessary, which is why I put that part into the README. However, I think @tanyouliang95 is still missing the force_torque_sensor_controller.

Remapping the controller namespace with the default UR MoveIt! configs seems to be necessary. Personally, I think that the controller namespace should be a launchfile parameter, but we could add MoveIt! setup to the troubleshooting section.

tanyouliang95, 2019-12-06:
Hi, yeap!!!  Changing the `action_ns` to `scaled_pos_traj_controller/follow_joint_trajectory` in `controller.yaml` did help ! Sorry for not reading the README throughfully....

Also, different to ur_modern_driver, instead of operating in remote mode, I will just need to run the URcap in local mode, and start it by clicking the play button below. Hope this issue will others in terms of migrating from `ur_modern_driver` to `ur_robot_driver`! Thanks alot   :tada:

jbcpollak, 2019-12-17:
> Changing the action_ns to scaled_pos_traj_controller/follow_joint_trajectory in controller.yaml

This issue took me all day to sort out. the README is not clear. I know it says:

> To control the robot using ROS, use the action server on
> `/scaled_pos_traj_controller/follow_joint_trajectory`
> Use this with any client interface such as MoveIt! 

but that meant nothing to me until I found @iretiayo 's comment (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-562215033) above where he explicitly linked to where to set the action server. Why isn't that the default action server in the controllers.yaml file? Isn't that UR-specific repo? What's the purpose of the existing default one? Can an explanation please be added to the documentation?

For anyone else who is searching for this problem, it happened for me on a UR10

gavanderhoorn, 2019-12-18:
I sympathise with the reporters here (and I'm not advocating for not adding more documentation or ignoring this), but in general I would say this is not something specific to this driver.

Configuring MoveIt packages for use with specific drivers (ie: action servers) is something that always needs to be done with care. It cannot be automated nor generalised. To be specific:

> Why isn't that the default action server in the controllers.yaml file? Isn't that UR-specific repo?

because that action topic is specific to `ur_robot_driver`, and there are more drivers for UR robots.

If those packages would be updated to assume use with `ur_robot_driver`, users of other drivers would start asking the exact same question.

> What's the purpose of the existing default one?

If with "default one" you are referring to the MoveIt configurations: to provide you with a simple testing ground. For any serious use, you are expected to create customised MoveIt configurations, taking your specific robot setup (ie: EEFs and other devices) and workcell into account. A MoveIt configuration with a robot in empty space is almost unusable.

> For anyone else who is searching for this problem, it happened for me on a UR10

As I wrote above: this is not specific to any one robot, nor any specific driver.

---

Having written all of this: I assume additional explanation of this part of the configuration will be added.

@fmauch: we may also consider remapping the `scaled_pos_trajectory_controller` ns to a regular one, to make it more plug-and-play. But I'm assuming this was intentionally not done to make handling support requests easier.

jbcpollak, 2019-12-18:
>>  Why isn't that the default action server in the controllers.yaml file? Isn't that UR-specific repo?
> because that action topic is specific to ur_robot_driver, and there are more drivers for UR robots.

ok, I get that, but this is the _Quick Start_ section on a specific driver's README. I think its safe to assume newbies without familiarity with ROS or UR robots are going be follow that guide in the hopes of getting stuff setup.

>>  What's the purpose of the existing default one?
> If with "default one" you are referring to the MoveIt configurations: to provide you with a simple testing ground. 

by "default one" I meant specifically [this line](https://github.com/fmauch/universal_robot/blob/c5a65a5d4dc9a5d53d6db06a3542464db3b34885/ur5_moveit_config/config/controllers.yaml#L3), which needs to be updated to work with the the `ur_robot_driver`. I assume the default is for _ur_modern_driver_ or something, but since that driver is deprecated and this current repo is the recommended one to use, its not clear to me why the mismatch is here.

Re the expectation that people write their own MoveIt configurations - that makes sense, but does that mean they would need to write their own action controllers as well? 

One last comment on the documentation, which is a bit off this specific issue, but is relevant - The Quickstart is great, but a pointer on where to go next would be useful. Perhaps that's just a pointer to the MoveIt documentation or something, but now that I have ROS control of the arm, it's not clear how I start actually doing something with it....

Thanks for hearing me out!

fmauch, 2019-12-19:
> @fmauch: we may also consider remapping the `scaled_pos_trajectory_controller` ns to a regular one, to make it more plug-and-play. But I'm assuming this was intentionally not done to make handling support requests easier.

@gavanderhoorn I agree that the situation is non-optimal. However, I do not completely understand your suggestion (specifically the "to a regular one" part). From my point of view, it seems a bit odd (probably there are legacy reasons from before-ROS-Control times) that the default MoveIt! configuration expects a controller action topic running at `/follow_joint_trajectory`. As far as I understood ROS-Control, controller action servers will always live at `/<controller_name>/<action_name>` which will result in a `/<controller_name>/follow_joint_trajectory` for any *joint_trajectory_controller* type.

To get a `/follow_joint_trajectory` action server the controller would have to have an empty name which is to my knowledge not possible. Adding a remap such as 

```xml
<remap from="/scaled_pos_traj_controller/follow_joint_trajectory" to="/follow_joint_trajectory" />
```

e.g. to [ur_control.launch](../blob/master/ur_robot_driver/launch/ur_control.launch) also feels a bit odd to me (@gavanderhoorn  is that what you meant?). I usually would expect to have a launch file parameter in the moveit configuration launchfile where I specify the controller / action server that should be used.
It used to work with both the [ur_driver](http://wiki.ros.org/ur_driver) and the [ur_modern_driver](http://wiki.ros.org/ur_modern_driver) as they did not use ROS-Control (At least in the default settings) but they rather implemented their own action server.

I'm not voting against any change here, I just don't really see "The right way" to go with this. Which is why all of your input is very valuable to me.

Note: Please consider that I have absolutely no knowledge about MoveIt!, so if I'm saying wrong things about MoveIt!, please correct me.

> One last comment on the documentation, which is a bit off this specific issue, but is relevant - The Quickstart is great, but a pointer on where to go next would be useful. Perhaps that's just a [...]

We already have a Link to MoveIt! at the end of the quick start section. Feel free to contribute with a short tutorial on how to use it with this particular driver. As stated above, I am not the right person to write a tutorial on how to use MoveIt!

tt2cloud, 2020-06-22:
Hi,dear all,I have a problem,my robot is UR10e,I followed the steps above to modify the controllers.yalm file
[image]
But ,when I moved the rviz, the real Hardware cannot move.
terminal1:roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.1.10
ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [8384]
process[ur_hardware_interface-2]: started with pid [8385]
process[ros_control_controller_spawner-3]: started with pid [8386]
[ INFO] [1592797670.344381530]: Main thread: SCHED_FIFO OK
[ INFO] [1592797670.344479463]: Main thread priority is 99
process[ros_control_stopped_spawner-4]: started with pid [8408]
[ INFO] [1592797670.356464561]: Initializing dashboard client
[ INFO] [1592797670.358266500]: Connected: Universal Robots Dashboard Server

process[controller_stopper-5]: started with pid [8443]
[ INFO] [1592797670.373468917]: Initializing urdriver
[ INFO] [1592797670.374089713]: Checking if calibration data matches connected robot.
[ INFO] [1592797670.374689783]: Producer thread: SCHED_FIFO OK
[ INFO] [1592797670.374725112]: Thread priority is 99
process[ur_hardware_interface/ur_robot_state_helper-6]: started with pid [8465]
[ INFO] [1592797670.383746952]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1592797670.384315726]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ERROR] [1592797670.732675171]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [TODO Link to documentation] for details.
[INFO] [1592797670.827828]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1592797670.856582]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1592797671.434940121]: Producer thread: SCHED_FIFO OK
[ INFO] [1592797671.435011387]: Thread priority is 99
[ INFO] [1592797671.791331044]: Setting up RTDE communication with frequency 500.000000
[ INFO] [1592797672.829602427]: Producer thread: SCHED_FIFO OK
[ INFO] [1592797672.829628161]: Thread priority is 99
[ INFO] [1592797672.829863525]: Loaded ur_robot_driver hardware_interface
[ INFO] [1592797672.885147966]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1592797672.885178731]: Service available.
[ INFO] [1592797672.885193496]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1592797672.885982928]: Service available.
[INFO] [1592797672.947910]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1592797672.949247]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1592797672.950428]: Loading controller: joint_state_controller
[INFO] [1592797672.957540]: Loading controller: scaled_pos_traj_controller
[INFO] [1592797672.976145]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1592797672.977477]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1592797672.978721]: Loading controller: pos_traj_controller
[INFO] [1592797673.007316]: Loading controller: speed_scaling_state_controller
[INFO] [1592797673.046392]: Loading controller: joint_group_vel_controller
[ INFO] [1592797673.047499093]: Robot's safety mode is now NORMAL
[ INFO] [1592797673.047703390]: Robot mode is now RUNNING
[INFO] [1592797673.051587]: Loading controller: force_torque_sensor_controller
[INFO] [1592797673.059231]: Controller Spawner: Loaded controllers: pos_traj_controller, joint_group_vel_controller
[INFO] [1592797673.063381]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1592797673.065233]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller

terminal2:roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch limited:=true
ROS_MASTER_URI=http://localhost:11311

process[move_group-1]: started with pid [9208]
[ INFO] [1592797854.051201196]: Loading robot model 'ur10e'...
[ WARN] [1592797854.051295497]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1592797854.051308504]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1592797854.125901850]: Loading robot model 'ur10e'...
[ WARN] [1592797854.125926664]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1592797854.125937078]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1592797854.185997114]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1592797854.188506916]: MoveGroup debug mode is OFF
Starting context monitors...
[ INFO] [1592797854.188543285]: Starting scene monitor
[ INFO] [1592797854.190988216]: Listening to '/planning_scene'
[ INFO] [1592797854.191010974]: Starting world geometry monitor
[ INFO] [1592797854.193649727]: Listening to '/collision_object' using message notifier with target frame '/world '
[... cut ...]
[ WARN] [1592797872.815150812]: Interactive marker 'EE:goal_ee_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
[ INFO] [1592797980.078785569]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592797988.853524536]: ABORTED: Solution found but controller failed during execution
[image]

Could you help me !  I want to move the real Hardware UR10e under the rviz! Thanks

anton-brinster, 2020-06-23:
Hello everyone, I am facing same problem at the point of using the action server on `/scaled_pos_traj_controller/follow_joint_trajectory` to control the robot/joints.

**My steps:** (with significant messages)

1. set the action controller namespace to `scaled_pos_traj_controller/follow_joint_trajectory`
2. roscore
3. roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.0.81 + Start program on TP 

> Robot ready to receive control commands.

4. roslaunch ur3_moveit_config ur3_moveit_planning_execution.launch

> [ WARN] [1592932372.503733660]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
> [ WARN] [1592932372.608198220]: Kinematics solver doesn't support #attempts anymore, but only a timeout. Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.

> [ WARN] [1592932378.049664926]: Waiting for scaled_pos_traj_controller/follow_joint_trajectory/follow_joint_trajectory to come up
> [ERROR] [1592932390.050209945]: Action client not connected: scaled_pos_traj_controller/follow_joint_trajectory/follow_joint_trajectory
> You can start planning now!

5. roslaunch ur3_moveit_config moveit_rviz.launch config:=true

> [ WARN] [1592932381.425419411]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
> [ INFO] [1592932381.425564176]: No root/virtual joint specified in SRDF. Assuming fixed joint
> [ WARN] [1592932381.496902591]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
> Please remove the parameter '/rviz_labor_8661_1211045767567327262/manipulator/kinematics_solver_attempts' from your configuration.
> [ INFO] [1592932433.795708751]: ABORTED: Solution found but controller failed during execution

In Terminal of "planning execution:

> [ERROR] [1592932436.183698945]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
> [ERROR] [1592932436.183751039]: Known controllers and their joints:
> [ERROR] [1592932436.183803715]: Apparently trajectory initialization failed

I would appreciate a detailed description of how to handle the action server practically.

Thanks and best regards
Anton

fmauch, 2020-06-24:
> [ WARN] [1592932378.049664926]: Waiting for scaled_pos_traj_controller/follow_joint_trajectory/follow_joint_trajectory to come up
> [ERROR] [1592932390.050209945]: Action client not connected:

This doesn't look correct to me.

fmauch, 2020-06-24:
You would have to modify the [`controllers.yaml`](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10_e_moveit_config/config/controllers.yaml) like this:

```yaml
controller_list:
  - name: ""
    action_ns: /scaled_pos_traj_controller/follow_joint_trajectory
    type: FollowJointTrajectory
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
```

With that setup I just tested it and it seems to work.

tt2cloud, 2020-06-25:
> You would have to modify the [`controllers.yaml`](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10_e_moveit_config/config/controllers.yaml) like this:
>
> ```yaml [...]
yeah,you're right,, I successed. Thank you!!

tt2cloud, 2020-07-01:
Dear fmauch,I have a new problem,I am trying to attach a Robotiq 2f AG-85 gripper to a UR10e arm in Rivz,but cannot get the two components to properly connect to one another,I created the xacro file,but only the gripper appears when I launch the MoveIt Setup Assistant and load the xacro.I don't see any errors.can you solve my problems,Thanks in advance! or was there a similar packages?
the gripper packages downloaded in (https://github.com/ros-industrial/robotiq)

fmauch, 2020-07-01:
Dear @tt2cloud please accept, that this is not a ROS helper forum. This is an issue tracker for the Universal Robots ROS driver. For questions about ROS, please go to ROS Answers.

I'll close this, as the initial problem seems to be solved by the workaround mentioned in #55

closed 2020-07-01 by fmauch

NZdreamer, 2020-08-06:
Hello, dear all, I have the same problem and still not fix it after modifing the `controllers.yaml` as guided. When launching moveit! like `roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch`, I got these:
```
[ WARN] [1596687278.147137989]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ WARN] [1596686938.792754798]: Waiting for /scaled_pos_traj_controller/follow_joint_trajectory to come up
[ WARN] [1596686944.794059622]: Waiting for /scaled_pos_traj_controller/follow_joint_trajectory to come up
[ERROR] [1596686950.795850394]: Action client not connected: /scaled_pos_traj_controller/follow_joint_trajectory
```
Then trying to plan and execute in rviz, the actual UR5 didn't move and the terminal shows:
```
[ERROR] [1596687473.671224078]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
[ERROR] [1596687473.671365899]: Known controllers and their joints:

[ERROR] [1596687473.671471272]: Apparently trajectory initialization failed
```
which is exact the same errors as before, but my `controllers.yaml`  does look as ame as @fmauch suggested. Could anyone help me solve the problem, thanks in advance!

fmauch, 2020-08-06:
@NZdreamer please confirm that the driver is running correctly by using the `rqt_joint_trajectory_controller` as noted in the [quick start section](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#quick-start).

For this, please also see the most common [troubleshooting question](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot).

NZdreamer, 2020-08-06:
@fmauch Thank you for reply! I had followed the steps in quick start section. The driver is running correctly by `rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`, UR5 can be controled nicely. 
However, I'm facing the problem above when trying to use moveit!. I've been stuck for a few days but still cannot figure it out.

fmauch, 2020-08-06:
Ah, looking back at your comment: Do you use a recent version of the driver? The controller got renamed to `scaled_pos_joint_traj_controller` a couple of commits back. You are still requesting `scaled_pos_traj_controller`.

NZdreamer, 2020-08-06:
@fmauch That's the point! I cloned the driver few weeks ago. Now I fix it, thanks a lots!

gokhansolak, 2020-08-26:
I get the same `Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]` error when I execute a Moveit plan.

* I have the *external-control* URCap program started.
* rosdep install gives "#All required rosdeps installed successfully"
* In controllers.yaml file I tried each of the following namespace options, all give the same error.
  - `follow_joint_trajectory`
  - `scaled_pos_traj_controller/follow_joint_trajectory`
  - `scaled_pos_traj_controller/scaled_pos_joint_traj_controller`
  - `scaled_pos_traj_controller`
  - `controller_manager/scaled_pos_traj_controller` (from rqt_joint_trajectory_controller)
* I can move the joints using *rqt_joint_trajectory_controller*.
* Output of `rosservice call controller_manager/list_controller_types`:
<details>
  <summary>[Click to expand]</summary>
types: [effort_controllers/JointEffortController, effort_controllers/JointGroupEffortController,
  effort_controllers/JointGroupPositionController, effort_controllers/JointPositionController,
  effort_controllers/JointTrajectoryController, effort_controllers/JointVelocityController,
  force_torque_sensor_controller/ForceTorqueSensorController, industrial_robot_status_controller/IndustrialRobotStatusController,
  joint_state_controller/JointStateController, pos_vel_acc_controllers/JointTrajectoryController,
  pos_vel_controllers/JointTrajectoryController, position_controllers/JointGroupPositionController,
  position_controllers/JointPositionController, position_controllers/JointTrajectoryController,
  position_controllers/ScaledJointTrajectoryController, ur_controllers/SpeedScalingStateController,
  velocity_controllers/JointGroupVelocityController, velocity_controllers/JointPositionController,
  velocity_controllers/JointTrajectoryController, velocity_controllers/JointVelocityController,
  velocity_controllers/ScaledJointTrajectoryController]
base_classes: ['controller_interface::ControllerBase', 'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase']
</details>

What is the most recent namespace? The README currently says "scaled_pos_traj_controller/follow_joint_trajectory" but it gives error. 

Also, it may be handy to mention the controllers.yaml for the users who are not accustomed to moveit config.

fmauch, 2020-08-26:
@gokhansolak please don't dig out old issues if you have an issue that might be related with this one.

Please open a new issue where clearly describe what you have done.
 - Which launchfiles / nodes did you start?
 - What are you doing to interact with the driver?
 - What would you expect to happen?
 - What is the error that you get?

mn-banjar, 2020-10-19:
> Hello, dear all, I have the same problem and still not fix it after modifing the `controllers.yaml` as guided. When launching moveit! like `roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch`, I got these:
> 
> ```
> [ WARN] [1596687278.147137989]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
> [ WARN] [1596686938.792754798]: Waiting for /scaled_pos_traj_controller/follow_joint_trajectory to come up
> [ WARN] [1596686944.794059622]: Waiting for /scaled_pos_traj_controller/follow_joint_trajectory to come up
> [ERROR] [1596686950.795850394]: Action client not connected: /scaled_pos_traj_controller/follow_joint_trajectory
> ```
> 
> Then trying to plan and execute in rviz, the actual UR5 didn't move and the terminal shows:
> 
> ```
> [ERROR] [1596687473.671224078]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
> [ERROR] [1596687473.671365899]: Known controllers and their joints:
> 
> [ERROR] [1596687473.671471272]: Apparently trajectory initialization failed
> ```
> 
> which is exact the same errors as before, but my `controllers.yaml` does look as same as @fmauch suggested.
> The output of running `rosservice call controller_manager/list_controller_types`is:
> 
> ```
> types: [ackermann_steering_controller/AckermannSteeringController, controller_manager_tests/EffortTestController,
>   controller_manager_tests/MyDummyController, controller_manager_tests/PosEffController,
>   controller_manager_tests/PosEffOptController, controller_manager_tests/VelEffController,
>   diff_drive_controller/DiffDriveController, effort_controllers/GripperActionController,
>   effort_controllers/JointEffortController, effort_controllers/JointGroupEffortController,
>   effort_controllers/JointGroupPositionController, effort_controllers/JointPositionController,
>   effort_controllers/JointTrajectoryController, effort_controllers/JointVelocityController,
>   force_torque_sensor_controller/ForceTorqueSensorController, imu_sensor_controller/ImuSensorController,
>   industrial_robot_status_controller/IndustrialRobotStatusController, joint_state_controller/JointStateController,
>   pos_vel_acc_controllers/JointTrajectoryController, pos_vel_controllers/JointTrajectoryController,
>   position_controllers/GripperActionController, position_controllers/JointGroupPositionController,
>   position_controllers/JointPositionController, position_controllers/JointTrajectoryController,
>   position_controllers/ScaledJointTrajectoryController, ur_controllers/SpeedScalingStateController,
>   velocity_controllers/JointGroupVelocityController, velocity_controllers/JointPositionController,
>   velocity_controllers/JointTrajectoryController, velocity_controllers/JointVelocityController,
>   velocity_controllers/ScaledJointTrajectoryController]
> base_classes: ['controller_interface::ControllerBase', 'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase']
> ```
> 
> Could anyone help me solve the problem, thanks in advance!

I faced the same issue when I was trying to use an example package that uses Moveit.

In my case, I forget to install Moveit package on the ROS. so I solved the issue by running:

`sudo apt-get install ros-kinetic-moveit`

chentongtong121, 2020-12-09:
> I faced the same issue when I was trying to use an example package that uses Moveit.
>
> In my case, I forget to install Moveit package on the ROS. so I solved the issue by running:
>
> `sudo apt-get [...]

hi  I faced the same  problem for several days  have you  fixed it ？ I have  changed the action_ns,but  it didnot work , could anyone help?

chentongtong121, 2020-12-09:
Hello, dear all, I have the same problem and still not fix it after modifing the controllers.yaml as guided. When launching moveit! like roslaunch ur3_e_moveit_config ur3_e_moveit_planning_execution.launch, I got these:

[ WARN] [1596687278.147137989]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ WARN] [1596686938.792754798]: Waiting for /follow_joint_trajectory to come up
[ WARN] [1596686944.794059622]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1596686950.795850394]: Action client not connected: /follow_joint_trajectory

I have changed the action_ns: /scaled_pos_traj_controller/follow_joint_trajectory , but  from the WARN above it seems that the change dose not work , that confused me  for several days , i have tried "test_move.py" it works well, but  the  move it  dose not work ,could  anyone can help ? thanks a lot  in advance

abdul-mannan-khan, 2020-12-15:
First of all, thank you all for all of your efforts in this blog. I appreciate it. 

I am facing many problems related to this section. 

Question 1: My question is, should I use 192.168.56.101 or 192.168.0.100? If we have to use 192.168.56.101, why are we configuring IP address 192.168.56.1 in URCap (as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md)) and why are using 192.168.56.101 and not 192.168.56.1? Please, read details below so that my problem could be cleared. I appreciate for your precious time. Thank you. Here is detail. 

I am confused about IP address as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#setting-up-a-ur-robot-for-ur_robot_driver) that I need to run the following to get configuration file

> $ roslaunch ur_calibration calibration_correction.launch robot_ip:=<robot_ip> target_filename:="${HOME}/my_robot_calibration.yaml"
My IP network IP for UR5 is 192.168.0.100. I replace <robot_ip> with my IP and it works. Next, I installed URCap as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md) where I set the IP address: 192.168.56.1. However, as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#setting-up-a-ur-robot-for-ur_robot_driver) in **Quick start** 

> Once the driver is built and the externalcontrol URCap is installed on the robot, you are good to go ahead starting the driver. (Note: We do recommend, though, to extract your robot's calibration first.)
> 
> To actually start the robot driver use one of the existing launch files
> 
> $ roslaunch ur_robot_driver <robot_type>_bringup.launch robot_ip:=192.168.56.101

If I run `$ roslaunch ur_calibration calibration_correction.launch robot_ip:=192.168.56.101 target_filename:="${HOME}/my_robot_calibration.yaml"`, I get the following warnings 
```
[WARN] [1607997530.263981]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1607997530.270256]: Controller Spawner couldn't find the expected controller_manager ROS interface.
```
However, I do not get any warnings if I run `$ roslaunch ur_calibration calibration_correction.launch robot_ip:=192.168.0.100 target_filename:="${HOME}/my_robot_calibration.yaml"`. Also my actual robot configuration and configuration in RViz is same if after this command, `$ roslaunch ur_calibration calibration_correction.launch $ roslaunch ur_calibration calibration_correction.launch **robot_ip:=192.168.0.100** target_filename:="${HOME}/my_robot_calibration.yaml" target_filename:="${HOME}/my_robot_calibration.yaml"`, I run these commands:
```
$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true
$ roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```
However, with this command, `$ roslaunch ur_calibration calibration_correction.launch robot_ip:=192.168.56.101 target_filename:="${HOME}/my_robot_calibration.yaml"`, my robot in RViz is just lying on the ground. 

My question is should I use 192.168.56.101 or 192.168.0.100? If we have to use 192.168.56.101, why are we configuring IP address 192.168.56.1 in URCap (as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md)) and why are using 192.168.56.101 and not 192.168.56.1? Thank you for reading my question. 

Question 2: I believe that my second problem is due to the first question. I receive the error message as described in your blog [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#when-starting-the-program-on-the-tp-i-get-an-error-the-connection-to-the-remote-pc-could-not-be-established). In particular, I receive the error message on my Touch Panel as 

> The connection to the remote PC at 192.168.56.1:50002 could not be established. Reason: No route to host. 

I believe, if you could answer my Question 1, I would be able to solve this problem. I have tried all IPs. 192.168.56.1, 192.168.0.100, 192.168.56.101.  

Many thanks for your respone.

robint-XNF, 2021-04-23:
even though my controllers.yaml does look as same as @fmauch suggested.
and run:
 `sudo apt-get install ros-kinetic-moveit`
i still have the problem ,can anyone help me?

fmauch, 2021-05-01:
> even though my controllers.yaml does look as same as @fmauch suggested.
> and run:
> `sudo apt-get install ros-kinetic-moveit`
> i still have the problem `[ERROR] [1596686950.795850394]: Action client not connected: /scaled_pos_traj_controller/follow_joint_trajectory,`
> can anyone help me?

See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-669733953, the trajectory controller got renamed to match the other controllers.

fmauch, 2021-05-01:
@abdul-mannan-khan 

Sorry, your comment slipped through my mails, before. If your problems / questions still exist, could you please open a new issue? This seems rather unrelated to this particular issue in first place.

break-seven, 2021-10-17:
If you use google translate, don't forget close it and read it carefully "install the URCap on the robot and don't forget the last step of starting the External " above. Yep, it is the small  start key In the lower left corner of the screen. This is my problem,  maybe it could help you.