UniversalRobots/Universal_Robots_ROS_Driver #56: Could not get fresh data package from robot

megantuttle, 2019-12-05:
Upon attempting to set the tool digital output via the `/ur_hardware_interface/set_io` service, the driver throws an error:
``` [ERROR] [1575562385.903592318]: Could not get fresh data package from robot ```
and the output remains the same, although the service call returns `success: True`.

### Versions
 - Affected Robot Software Version(s): 3.11.0.82155
 - Affected Robot Hardware Version(s): UR3 CB-series

### Use Case and Setup
 - ROS Kinetic
 - Ubuntu 16.04.6

### Issue details
Terminal output from driver launch:
```
arw@arw-pc:~$ roslaunch ur_robot_driver ur3_bringup.launch
... logging to /home/arw/.ros/log/04cf3074-177a-11ea-9bba-7085c275eb55/roslaunch-arw-pc-121002.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://arw-pc:36045/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 125
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /scaled_pos_traj_controller/action_monitor_rate: 10
 * /scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
[... 105 lines cut ...]
[ INFO] [1575562347.639440003]: Robot's safety mode is now NORMAL
[ INFO] [1575562347.639986418]: Robot mode is now RUNNING
[INFO] [1575562347.670039]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1575562347.672224]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1575562347.673714]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1575562347.674900]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1575562347.676547]: Loading controller: joint_state_controller
[INFO] [1575562347.677481]: Loading controller: pos_traj_controller
[INFO] [1575562347.693851]: Loading controller: scaled_pos_traj_controller
[INFO] [1575562347.805735]: Controller Spawner: Loaded controllers: pos_traj_controller
[INFO] [1575562348.021409]: Loading controller: speed_scaling_state_controller
[INFO] [1575562348.029785]: Loading controller: force_torque_sensor_controller
[INFO] [1575562348.045608]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1575562348.053596]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ERROR] [1575562362.355470964]: Could not get fresh data package from robot
```
Terminal output from service call:
```
arw@arw-pc:~$ rosservice call /ur_hardware_interface/set_io "fun: 1
pin: 16
state: true" 
success: True
```
I also tried pin 17 with the same result.
The driver appears to be getting data from the robot because IO states are being published on the `/ur_hardware_interface/io_states` topic.

megantuttle, 2019-12-05:
I resolved this issue. I have Profinet enabled on the robot.  Evidently you cannot control the tool digital outputs via the RTDE interface while Profinet is enabled.  I looked in the [UR documentation on the RTDE interface](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/real-time-data-exchange-rtde-guide-22229/) and didn't find anything that suggested the Profinet interface would take priority or prevent certain outputs from being accessed via the RTDE interface.  I posted this issue to the driver just in case someone else runs into the same problem.  Perhaps the phenomenon could be addressed in the driver by giving a more informative error output; or maybe it could be included in the documentation.

fmauch, 2019-12-06:
This is someting not yet known to us and should definitely be tackled.

Unfortunately, I don't have any PROFINET device here. I cannot reproduce this in URSim. Does this only occur with a connected PROFINET device or can I test this on a real system without a PROFINET device attached?

gavanderhoorn, 2019-12-06:
@fmauch: as we also have #28 there may be some unexpected interactions between the driver and other 'users' of RTDE?

fmauch, 2019-12-06:
Well, #28 was about the missing fieldbus connection ended the program. This one here seems to directly interfer with RTDE data transport in some way, so I think, this is something different.

gavanderhoorn, 2019-12-06:
True.

It would perhaps be good to verify that using EtherNet/IP while the driver is active does not cause similar issues though.

fmauch, 2019-12-06:
> True.
> 
> It would perhaps be good to verify that using EtherNet/IP while the driver is active does not cause similar issues though.

Also true.

megantuttle, 2019-12-10:
@fmauch 
> This is someting not yet known to us and should definitely be tackled.
>
> Unfortunately, I don't have any PROFINET device here. I cannot reproduce this in URSim. Does this only occur with a connected PROFINET device or can I test this on a real system without a PROFINET device attached?

I was able to reproduce the issue by deleting the PROFINET device from the project and simply having PROFINET enabled on the robot.

I also just noticed that enabling PROFINET while the driver is connected causes an error message to appear on the TP.
[image]
The error message goes away once I shut down the driver. I'm not sure if that's useful information or not.

fmauch, 2019-12-12:
We checked back and there seem to be conflicts on the IO write access here. Unfortunately, I can't provide an ad-hoc solution for this, but we'll keep following this.

fmauch, 2019-12-16:
Unfortunately I am still not able to reproduce this. As I don't have any PROFINET hardware available, I cannot go any further than enabling PROFINET which will only complain about a "IO device module IO not plugged in" and "IO device module Registers not plugged in".

According to the [UR Profinet Guide](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/profinet-guide-20596/) and the [RTDE documentation](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/real-time-data-exchange-rtde-guide-22229/) the only conflicts could be raised by the `<input/output>_<type>_register_x` fields, which we currently do not use inside the driver.

With implementing #24 conflicts should be visible more clearly, we hope to finish a first implementation within this year.

fmauch, 2020-09-25:
Closing this in ~1week if no new information is coming in.

closed 2020-11-06 by fmauch

abdul-mannan-khan, 2020-12-08:
I have a another problem. My PROFINET is disabled still I am getting this error. 

Here is what I am doing:
```
$ roslaunch ur_calibration calibration_correction.launch \robot_ip:=192.168.0.100 \target_filename:="$(rospack find ur_calibration)/etc/ex-ur5_calibration.yaml"
$ roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.168.0.100 [reverse_port:=REVERSE_PORT] \kinematics_config:=$(rospack find ur_calibration)/etc/ex-ur5_calibration.yaml
```

But after sometime, I get this error "Could not get fresh data package from robot". My PROFINET is disabled. Any suggestion please? Thank you for reading my query and your interest.

fmauch, 2020-12-09:
@abdul-mannan-khan This could be related, but not necessarily is. Please open a new issue instead of restarting a closed issue with maybe a different topic.