UniversalRobots/Universal_Robots_ROS_Driver #58: unable to control UR5e robot

liangxiao010, 2019-12-10:
*Introduction to the issue*
I downloaded and built the universital_robot_ros_driver package and the fmatch_universial_robot package .(and also the external control URcap) And I can see the real pose of the robot, but it cannot execute through rqt or moveit(I mean the robot doesn't move at all.)

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s): 5.3.0
 - Affected Robot Hardware Version(s): Ur5e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): external control 1.0.1

#my procesures
1. Use the cable to connect the PC and the robot, set the correct IP address and ping it successfully. 
2. launch the ur5e bringup file.
3. since I can't press the program play button after turning into remote control or turn into remote control after playing the program, i turn into remote mode first and then use the sockettest software to communicate with the robot, load and play the program. Then I see  "Robot ready to receive control commands."
4. In this step I also tried the command
 **rosservice call /controller_manager/switch_controller "{start_controllers: [ 'scaled_pos_traj_controller'], stop_controllers: [ 'scaled_pos_traj_controller'], strictness: 1}"** 

and received "ok:ture"
With or without this command, I didn't see any difference in the rosservice list. Just a try.

5. I tried the      **rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller**
command with the error 
**Traceback (most recent call last):
File "/home/bodon/catkin_ws/src/ros_controllers/rqt_joint_trajectory_controller/src/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 416, in _update_cmd_cb
pos = self._joint_pos[name]['position']
KeyError: 'position'**
I have no idea about how to solve this.
6. Then I tried the moveit. But the ur5_e_moveit_planning_execution.launch has error about this:
**[ERROR] [1575962118.560119003]: Exception while loading move_group capability 'move_group/MoveGroupExecuteService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupExecuteService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()**

I changed the **move_group/MoveGroupExecuteService** to **MoveGroupExecuteTrajectoryAction** in move_group.launch, the error disappeared. 
Then I tried to move the robot in rviz interface,  the plan successed, but the execute failed with error
**ABORTED: Solution found but controller failed during execution**
Also, at the first, I have already changed to action_ns to scaled_pos_traj_controller/follow_joint_trajectory in controller.yam.

7. Actually, I have another moveit_config package which i used before to do the rviz simulation. I use this package to contain the urdf from the fmatch_universial_robot package, and do the moive it in rviz. No errors occured, even it said execute successfully. But the real robot didn't move, either. Then i found maybe I should also change to  scaled_pos_traj_controller/follow_joint_trajectory in this package's controller.yam. Then error occured, 
 **Action client not connected: /scaled_pos_traj_controller/joint_trajectory_action**

**Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
[ERROR] [1575966062.495484247]: Known controllers and their joints:**

liangxiao010, 2019-12-10:
@fmauch I will appreciate it if you can help me with that.

liangxiao010, 2019-12-10:
Another question is when i use the sockettest to communicate with the robot, I type in port 29999 in the software which is following the UR website. Should I still use 50001 on the teaching panel (installation -> urcaps). It seems no wrong with this port to play the program successfully.

liangxiao010, 2019-12-10:
https://drive.google.com/open?id=1SpwJ8-SEQw7E5msEogW5Y6HcxdW6I_xh

this is the ros serveice param and topic list when doing the step 6 Ur5e_moiveit

https://drive.google.com/open?id=1NHPLleYA8yvYB62Sog7XtIP1Ss0ESvff
this link is all the package files in my catkin_ws/src， where the fake_ar_publisher and the three myworkcell packages are my previous work. I use two commands to move the robot, 
**roslaunch myworkcell_moveit_config myworkcell_planning_execution.launch
 roslaunch myworkcell_support workcell.launch**

gavanderhoorn, 2019-12-10:
I'm not @fmauch, but I would suggest to focus on one particular problem at a time.

Let's forget about MoveIt, `rqt_joint_trajectory_controller` and all the rest.

I would first get the basis driver working, including starting things on the teach pendant and configuring everything correctly.

fmauch, 2019-12-10:
If I understand correctly, you managed to setup the robot and the driver, as you said that you see the robot model in RViz.

As you managed to get the output "Robot ready to receive control commands." and the program keeps running on the controller, it looks like your setup is correct so far.

By the way, you don't need to use remote-control mode to run the driver. However, you can also use the inbuilt Dashboard client to load and start the program. See the services in the `dashboard` namespace.

The `rqt_joint_trajectory_controller` thing could be related to #47.

To control the robot using Moveit!, please check the solutions proposed inside #55. This will be moved into the documentation.

liangxiao010, 2019-12-10:
Thanks for your reply. 
I am just confused why the remote control is not necessary. Can the robot receive command from pc under local mode？

liangxiao010, 2019-12-10:
Actually, I have tried the inbuilt dashboard. When i try 
**rosservice call /ur_hardware_interface/dashboard/get_loaded_program**
it works well to get the name. 
But when I try the 
**rosservice call /ur_hardware_interface/dashboard/play**
It got error that it cannot get the command from the server, something like this. 
For #47, I didn't get enough information about how to solve it.
For the #55, I have already followed it to set the scaled_pos_traj_controller/follow_joint_trajectory in controller.yam.

fmauch, 2019-12-10:
Remote control-mode is not needed to run the `External Control`, as it will open up a custom connection to the remote PC.

For using the dashboard client, for most calls remote-control mode is indeed needed. We will hopefully improve documentation on this.

And yes, #47 is not yet solved.

liangxiao010, 2019-12-10:
ok，so for my case，I just want to use ros to control the robot with cable connecting PC and robot. Only the external control is enough，right？
Do you think the errors of rqt and move it comes from the same issue？Because my final goal is to use move it. Until now，from the above dialogue，I am still confused about what I should do. Could you please give me some specific suggestions?  or if possible，can you help me to check my files in the links I given?  Since we are not very familar with the configuration of ros， my partner and I have been struggled in this problem for almost two weeks.

liangxiao010, 2019-12-12:
All the problems are solved by changing the port from 50001 to 50002. Thank you for your help!

fmauch, 2019-12-12:
Do I understand correctly, that you configured the URCap on the robot to use port 50001 instead of 50002 before and changing that back solved the problem?

liangxiao010, 2019-12-12:
Yes

liangxiao010, 2019-12-12:
The default port value on the panel is 50001 at the beginning.  We have also tried other values before，such as 30001,29999. Today we tried 50002，it worked.

fmauch, 2019-12-12:
The default value should be 50002 by default, if I'm not mistaken.

liangxiao010, 2019-12-12:
Maybe someone changed it before. I don’t know.  When I used it at the first time，it was 50001.

fmauch, 2019-12-16:
I'll go ahead and close this as it seems to work now.

closed 2019-12-16 by fmauch