UniversalRobots/Universal_Robots_ROS_Driver #59: "Received new trajectory execution service request..." get stuck and robot didn't movet

imba798, 2019-12-11:
# Summary

It seems that the ur5e received trajectory request but didn't execute it.

# Versions
 - ROS Driver version: Kinetic+Ubuntu 16.04
 - Affected Robot Software Version(s): 5.3.0
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):external control 1.0.1

# Procedure
I followed the README for Universal_Robots_ROS_Driver, including installing external control, set up IP address, extracting callibration, changing action_ns in controller.yaml, etc.

## Bring up
In the first terminal I run

`roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=MY_UR5E_IP`

And I run External Control program on the ur5e,

In the terminal I get,

>[INFO] [1576050304.674887]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
>[INFO] [1576050321.150582877]: Robot ready to receive control commands.

Everything seems correct so far, then I try Moveit! to move the robot arm.

## Moveit package

In the second terminal, I run

`roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch limited:=true`

By the way, the moveit package is from

`git clone -b calibration_devel https://github.com/fmauch/universal_robot.git`

In the terminal I get,

>[ INFO] [1576049741.049094121]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
>[ INFO] [1576049741.049121529]: MoveGroup context initialization complete
>
>You can start planning now!

Everything seems correct so far, then I try to launch RViz

## Lauching RViz

In the third terminal, I run

`roslaunch ur5_e_moveit_config moveit_rviz.launch config:=true`

No bugs happened in the terminal.

When I move the robot with freedrive, the robot in the RViz keeps the same posture with real robot.

## Plan and Execute
I use the GUI of Rviz to plan and execute a trajectory.

Trajectory can be sucessfully planned but when I click `execute`,

In the second terminal, it says

>[ INFO] [1576049754.799267993]: Received new trajectory execution service request...

The terminal gets stuck here and robot doesn't move.

What's more, there are no bugs shown in all three terminals.

Then, in the RViz, I click `stop`,

In the third terminal, it says,

>[ INFO] [1576049799.573201610]: Received event 'stop'
>[ INFO] [1576049799.573290613]: Cancelling execution for 
>[ INFO] [1576049799.573405578]: Stopped trajectory execution.
>[ INFO] [1576049799.577856689]: Completed trajectory execution with status PREEMPTED ...
>[ INFO] [1576049799.578047524]: Execution completed: PREEMPTED

It seems that robot is able to receive execution request and stop request but cannot execute the trajectory.

Since there are no bugs shown in the terminals, I have no idea which part is wrong or how to debug it. 

I would appreciate it if someone have a idea or have solved similar problems.

fmauch, 2019-12-11:
After moving the robot in freedrive mode - did you restart the program on the panel? It has to be running to accept commands from the external source like ROS.

MoveIt! integration has been causing problems before. Have you checked #55?

imba798, 2019-12-11:
@fmauch I am sure that the external control program was running but the robot didn't execute the planned trajectory. I tried lots of times with the same result without using freedrive.
I have checked #55 , thus I changed action_ns and started the external control program.

imba798, 2019-12-12:
@fmauch The problem has been solved. There is a problem with the configuration of the port. Sorry for the distraction.

closed 2019-12-12 by imba798