UniversalRobots/Universal_Robots_ROS_Driver #61: socket_read_binary_integer:timeout

JostMat, 2019-12-11:
# Summary
Hello,
We have a problem with the connection stability and can not find any more clues how to solve this.
After running
`roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=10.2.1.26`
the connection is established successfully and at first everything works fine. 
After a couple of seconds up to about one minute the connection gets terminated, as we found out later from the tablet (see tablet log image), and rebuild again.
If the connection is stable long enough we are able to start the motion planning library and the simple control GUI with 
`roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch limited:=true`
`rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`
To control the robot with this works until the connection is lost again.
If we want to use moveit with 
`roslaunch ur10_e_moveit_config moveit_rviz.launch config:=true`
anything crashes instantly.
Since we thought the system is maybe too slow, (because of the timeout problem and the juddery movements) we tried to switch to a real time kernel which seemed to have worked (choosen kernel image is correct) but ROS reports, that it can not detect realtime capability (see warning at the top part) and the connection is lost even faster than before.
The error message about fresh packages occurs right after launching the RViz GUI.

In the same workspace as the robot package is the ros_controllers package.
We invoked catkin_make after the kernel switch.

[image]

[image]

# Versions
 - ROS Driver version:  Kinetic
 - Affected Robot Software Version(s): 3.2
 - Affected Robot Hardware Version(s): UR10e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

gavanderhoorn, 2019-12-11:
Just to make sure: is this all using a cabled network? Or are there wireless segments between the ROS PC and the UR controller?

JostMat, 2019-12-11:
Thanks for the fast reply.
I updated the question. 
You were right about the version of course and the connection is indeed wireless. 
I forgot to mention it and did not pay much attention to it because in the lab there are several robots connected to this server that are controlled with mqtt and there were no to timing issues reported.

gavanderhoorn, 2019-12-11:
The driver allows you to control the robot in real-time.

Wireless connections do not work very well for these kinds of setups.

I would not recommend using them.

Can you confirm that using a wired connection fixes the issues you were seeing?

JostMat, 2019-12-11:
I will try it and get back to you asap.
Is there a way to use the driver in a wireless setting?

fmauch, 2019-12-11:
No, this is not possible, as @gavanderhoorn explained above. And this is not only because of controlling the robot but actually sending robot data through RTDE from the robot to the ROS machine fails as the errors "Could not get fresh data package from robot" tell us. TCP connections are very error-prone when it comes to package loss and we expect to receive robot data with 500Hz.

JostMat, 2020-01-08:
First of all: Thank you. 
The hint has solved the initial problem and we figured out the rest. The driver is working great. We want to try and change the setup to use a raspberry pie in the future to have a wireless connection but for now everything works fine.

closed 2020-01-08 by JostMat