UniversalRobots/Universal_Robots_ROS_Driver #62: Publish "actual_TCP_speed" data field in hardware interface

rcorralesf, 2019-12-11:
Hardware_interface doesn't appear to include functionality to read/publish the "actual_TCP_speed" field from rtde. This data is useful when using the robot as both an actuator and a sensor during precise operations. 
I would be happy to PR a quick implementation but I am not familiar enough with the driver to suggest a ROS message type. Maybe create a quick 6 float vector message and store it somewhere in ur_robot_driver/include/ur_robot_driver/ros? [RobotStateRTMsg.msg](https://github.com/fmauch/universal_robot/blob/kinetic-devel/ur_msgs/msg/RobotStateRTMsg.msg) from ur_msgs already includes a tcp_speed field but with so many more fields in it I think it would be overkill to reuse it unless everything is implemented.

fmauch, 2019-12-11:
We are currently implementing this inside #35. Does that fit your needs?

rcorralesf, 2019-12-11:
Yes, that pretty much targets my needs. Is it currently in a usable state? 
Beyond that, thank you for your efforts. I totally agree that a proper interface for rtde will be very useful for ROS integrations. Let me know if I can be of any help and do not hesitate to ask for user feedback if you need it.

fmauch, 2019-12-12:
The state it is currently inside #55 should be working, but it is not developed on top of the latest master. We expect to refactor this as proposed inside the PR within 2019, but I can't promise anything.

Feel free to comment on the discussion inside #55 and test the features as soon as we push updates. I'll mark this issue to be closed by #55.

urfeex, 2026-01-19:
Since ROS (1) is end of life this will not be worked on anymore. See https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1648 for a discussion around this topic in ROS 2.