UniversalRobots/Universal_Robots_ROS_Driver #65: [kinematics_config] argument is needed?

AhmetMericOzcan, 2019-12-15:
Trying to use it on real robot with Moveit, and I am trying to use it with limited version since otherwise it can create unwanted dangerous situations.

However even though I can connect to robot with certain IP, when run the command suggested I got this error

`roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=10.0.0.2 limited:=true
`

```
unused args [kinematics_config] for include of [/home/meric/c_ws/src/universal_robot/ur_description/launch/ur5_upload.launch]
The traceback for the exception was written to the log file
```

In the launch file argument list it is said that I need to use it with calibration. But I dont prefer that, and in the main page it is said like calibration is an optional thing. Any Ideas?

gavanderhoorn, 2019-12-15:
Have you made sure you have the `fmauch/universal_robot` fork of `ros-industrial/universal_robot`, as explained in the [Building](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building) section of the readme?

ipa-lth, 2019-12-16:
I had the same issue, even without involving any calibration (for now).
Cloning the fmauch/universal_robot fork fixed the problem for me.

fmauch, 2019-12-16:
Yes, this is currently necessary until we got the changes merged into the mainline packages. @AhmetMericOzcan If this also resolves your problem, feel free to close this issue.

fmauch, 2020-05-12:
I'll close this as this seems to be resolved.

closed 2020-05-12 by fmauch