UniversalRobots/Universal_Robots_ROS_Driver #67: Feature: A free parameter, that the UR user can configure to select different behaviours on the ROS side

ipa-lth, 2019-12-17:
# Summary

A addition free parameter could be transmitted in the rtde protocol to allow a UR user to configure what should happen on the ROS side (encoded in e.g. an int or string). The URcap should allow configuring this parameter in the tree-view individual for every occurrence of the "external control" (not in *installation*).
In some cases the external control could then be e.g. a force control in other situations something else.
The free parameter should be transmitted somehow (rostopic?) to the ROS world, to be picked up by whatever functionality there is.
 
# Impact
The UR user could benefit from many different behaviors provided by different ROS nodes/controller, without being bothered with ROS at all.

fmauch, 2019-12-17:
Something like this will be implemented during the second ROSin phase.

ipa-lth, 2019-12-17:
Great.
Is there already a time schedule? We might can contribute.

fmauch, 2019-12-17:
We'll start January 2020 with the concept phase. Development will be quite open with that, so we'll try to keep you in the loop.

ipa-lth, 2020-06-05:
What is the status here?

fmauch, 2020-06-05:
Sorry for responding that late. We currently have a design document in the loop that we hope to publish soon. Please keep bugging me again, if I don't publish it in the next 2 weeks.

ipa-lth, 2020-06-05:
I will :-).
Thank for your help!

fmauch, 2020-06-30:
I pushed our current concept on the branch [ros_as_a_service](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/ros_as_a_service). Please, feel free to add your comments to the according PR in #212.

fmauch, 2021-04-26:
We've created a separate URCap: [ros_as_service](https://github.com/UniversalRobots/Universal_Robots_ROS_as_a_Service_URCap)

It is implementing #212 and currently allows publishing messages, reading messages and calling services. Running actions will be added soon (~EOW) and setting and reading parameters is also on the todo list.

stefanscherzinger, 2021-08-09:
@ipa-lth This functionality should be covered by the latest ROS as a service [URCap](https://github.com/UniversalRobots/Universal_Robots_ROS_as_a_Service_URCap). Feel free to re-open this issue if you think this driver needs further enhancements in that direction.

closed 2021-08-09 by stefanscherzinger