UniversalRobots/Universal_Robots_ROS_Driver #68: EM - stop doesn't clear current goal

ichumuh, 2019-12-17:
# Summary

When the robot is executing a goal from the ros action server and I press the EM-Stop and release it again, the robot executes the rest of the remaining trajectory with what appears to be maximum velocity.
This usually triggers the protective stop.
I'm surprised that I didn't find any related issues. Am I doing something wrong or is this maybe even intended?

# Versions
 - ROS Driver version: I just pulled https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/a926789adc0cefc616beb0fda15793cdcd65831b
 - Affected Robot Software Version(s): 3.11.0.82155
 - Affected Robot Hardware Version(s): UR5
 - Robot Serial Number:
 - UR+ product(s) installed: 
 - URCaps Software version(s): external control 1.0.1

# Impact
In the best case it just wastes time. In the worst case you get a heart attack and die, because its fucking scary. :(

# Issue details

- it happens both with the em stop and the safeguard stop
- canceling the goal on the ros driver doesn't fix the problem. It seems like the remaining trajectory is executing until the time step when a cancel was received by the ros driver.
- I saw that the ros driver status of the current goal switches to aborted when the em stop is pressed.

## Use Case and Setup
I don't think that's relevant here.

## Project status at point of discovered
Well, it always happens, I just didn't create an issue until now.

## Steps to Reproduce
You send a goal to the driver, wait until the arm starts moving and hit the em-stop and then release it again.
It happens both, with the em-stop and the safeguard-stop.

## Expected Behavior
I expect the current goal to get canceled when the em-stop is hit. I don't see any reason why one would want to execute the old trajectory. If you do, I think its better to have some failure handling on the client side instead.
At the very least the robot should not try to "catch up" by using max velocity.

## Workaround Suggestion
Restarting the ros driver clears the goal.

fmauch, 2019-12-17:
I think, you are using the *joint_trajectory_controller* instead of the *scaled_trajectory_controller* wich is enabled by default.

On UR robots an EM-Stop causes a "Pause" of a program. As the standard *joint_trajectory_controller* does not recognize this pausing (in contrast to the custom scaled version), the controller keeps interpolating in the background. So, when you resume the program, the robot tries to move to the last commanded position from ROS.

One main problem I see is that the ROS controller keeps sending updates even if the trajectory is being aborted due to trajectory constraints being harmed. I'm not sure whether this behavior is intended, but it is also out of scope of this project.

Could you confirm, that you are not using the scaled trajectory controller? Because with this, that should not happen.

Additional note: If you use the dashboard client to recover after an EM-Stop (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#actions) there is a parameter to stop the current program (in terms of the "External Control" component interpreting ROS commands on the robot.

And now that we clarified all this, we can come to the discussion part: I was quite surprised the first time when I realized that an EM-Stop only pauses the program on the robot instead of stopping it. With this being the case on the hardware, we decided to forward this behavior to the ROS side. One could argue that this is something not very common in ROS drivers, but EM-stop handling actually isn't that common for many robot drivers.

One possibility would be to reset the trajectory interface on an EM stop but that would also mean to remove the (maybe by some expected or desired) pausing feature. Also How would we handle Protective- or Safeguard stops in contrast? Especially for the safeguard stop it does make sense to only pause the running program and continue afterwards, as for example shown in our [release video](https://youtu.be/EfEfkXUJwP8?t=88) where we pause the robot while a laser scanner field is harmed.

I would like to hear different opinions on this topic :-)

ichumuh, 2019-12-17:
Thanks for the quick reply. Yes I was using the joint trajectory controller, because I thought it was like the scaled one set to 100%. So using the scaled one would fix my problem?

fmauch, 2019-12-17:
Using the scaled controller would continue your trajectory after pressing "play" on the TP once the EM-Stop is released. (However, with normal speed, as trajectory interpolation was paused in between).

ichumuh, 2019-12-17:
Yes I've just tried it, that's much better. Thank you very much.

closed 2020-04-03 by fmauch