UniversalRobots/Universal_Robots_ROS_Driver #69: ROS 2 support

darkdragon-001, 2019-12-17:
Opening this issue to track progress. Follow up on ros-industrial/ur_modern_driver#244

fmauch, 2019-12-19:
Thanks for raising this. Let me quickly state that we currently don't have any specific plans to develop this, but I would be happy to assist in questions raised by anyone going for this.

gavanderhoorn, 2019-12-19:
First thing to do probably would be to refactor `ur_robot_driver` into the ROS-agnostic library and the ROS package (ie: the driver). Then make the library a plain CMake package, which would make it easily consumable by both ROS 1 and ROS 2 packages.

Porting the driver (so not the library) to ROS 2 is complicated right now because of the lack of `ros_control` support in ROS 2.

There is work underway on that as well, but it's not usable right now.

robinsonmm, 2020-10-01:
Based on the comments related to ros_control support limiting ROS2 driver development, has there been any update on adding ROS2 support relative to the UR driver?

MR

gavanderhoorn, 2020-10-01:
I'll leave the details to @fmauch, but at least the refactor into separate library has been done/is in progress. See #266 and [UniversalRobots/Universal_Robots_Client_Library](https://github.com/UniversalRobots/Universal_Robots_Client_Library).

re: `ros2_control`: progress is being made (it's (partly) a ROSin FTP). Things are in flux right now though, so any effort would probably want to coordinate with the `ros2_control` devs.

fmauch, 2020-10-01:
The library in [UniversalRobots/Universal_Robots_Client_Library](https://github.com/UniversalRobots/Universal_Robots_Client_Library) should be ready to use, so if anyone wants to start building a ROS2 driver ontop of that, this should be working. The dev team from this driver currently still doesn't have any plans further following this.

bharathra, 2020-10-01:
@fmauch Understand that currently you do not have plans to develop ROS2 drivers. In the long term, is ROS2 support on your roadmap? We are currently looking at productionizing couple of robot (UR10e) cells at our company and ROS2 support is critical. (Prototype cell is running via ROS bridge). Curious to know your long term roadmap for ROS2. Thank you.

mahmoud-a-ali, 2020-10-02:
I am also interested to see ros2 driver for UR robots! Do you have any plans to support ROS2 in the future?

marioney, 2020-10-02:
I agree, support for ROS2 will be very valuable, so I'm also interested on the roadmap for ROS2

tckarenchiang, 2020-10-02:
Yes, as there are more and more ROS2 users, support for ROS2 is definitely worth consideration.

robinsonmm, 2020-10-05:
Being that there seems to be interest, and within the ROS-Industrial Consortium Americas we have membership engagement on this topic, what would it take @fmauch to get it on the UR/FZI roadmap?

JStech, 2020-10-07:
Yes, please support ROS 2!

mitchpryor, 2020-10-09:
ROS2 integration will be a must. Many of our collaborators are seeing ROS2 as a necessary upgrade to move systems from the lab to the field for a variety of good reasons.

fmauch, 2020-10-12:
Maybe @stefanscherzinger can drop a word about this once there are any news?

jliukkonen, 2020-10-12:
ROS 2 support would be much appreciated.

schornakj, 2020-10-12:
+1 from me too! My group at SwRI is using a few UR-10e's in industrial applications where most of the rest of the system consists of ROS2 nodes. Right now we use the `ros1_bridge` and a custom action bridge node to connect the ROS1 UR driver with the rest of the system, but it would be strongly preferable to have the whole system use ROS2.

BVeillette, 2020-10-13:
Another vote for ROS 2 support. We use UR5e and UR7e in production.

ipa-cmh, 2020-10-21:
With ros2_control and moveit2 upcoming, a native ROS 2 UR driver would be very cool. So +1.

wafts95, 2021-01-14:
One more vote for ROS 2 Support! We have almost 7 UR10 robots planned for use this year in production.

Can you please share with us your development timeline? Right now we are still using the urx driver, but we need to switch to the more powerful UR ROS driver in the next few months. Your timeline would be very helpful for us to decide if we wait for ROS2 support, or directly start using the outgoing UR ROS driver. Thanks!

fmauch, 2021-01-14:
Work is on its way currently. @AndyZe can probably say more about this atm.

AndyZe, 2021-01-14:
Indeed. You can keep an eye on the repo [here](https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver/). It's not ready for public use yet, though. My rough guess is 2 months for a beta release.

davetcoleman, 2021-03-10:
Update: the driver is ready for beta testers, [announcement here](https://picknik.ai/ros2/moveit2/ros2_control/2021/02/09/ur-ros2-driver-announcement.html), but still plenty of work to be done.

urrsk, 2021-05-12:
Please see the https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver

fmauch, 2023-01-23:
Closing this as the driver for ROS2 has been released since quite a while.

closed 2023-01-23 by fmauch