UniversalRobots/Universal_Robots_ROS_Driver #70: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]

Guanxiaolong0408, 2019-12-19:
I using easy_handeye to calibrate the UR5 and depth camera, all the driver has been installed ,  while I using the rqt_easy_handeye to execute the plan, the robot on the rviz could move , **but the real robot cannot move** , and there is errors as 

[ INFO] [1576755856.511674999]: Execution request received
[ INFO] [1576755856.511729324]: Returned 0 controllers in list
[ERROR] [1576755856.511757687]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
[ERROR] [1576755856.511782182]: Known controllers and their joints:

Is there any one meet this problem?

fmauch, 2019-12-19:
As far as I see, this uses MoveIt!, so it might be related to #55. If so, [this comment](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-562215033) might help you. Feel free to participate in the discussion inside #55.

fmauch, 2020-04-03:
Any news on this? Otherwise I will consider closing this in a future sweep through the issues.

closed 2020-05-12 by fmauch

mn-banjar, 2020-10-19:
> I using easy_handeye to calibrate the UR5 and depth camera, all the driver has been installed , while I using the rqt_easy_handeye to execute the plan, the robot on the rviz could move , **but the [...]

I faced the same issue when I was trying to use an example package that uses Moveit.

In my case, I forget to install Moveit package on the ROS. so I solved the issue by running:

`sudo apt-get install ros-kinetic-moveit`