UniversalRobots/Universal_Robots_ROS_Driver #71: should this be checking the first index?

jbcpollak, 2019-12-20:
Reviewing the URScript, specifically [this line](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/fad8c0b34f291280cb03bde791b71ac7a3ba1349/ur_robot_driver/resources/servoj.urscript#L83), I see:

```
    if params_mult[0] > 0:
      keepalive = params_mult[1]
      if params_mult[1] > 1: # <--- THIS LINE
        q = [params_mult[2] / MULT_jointstate, params_mult[3] / MULT_jointstate, params_mult[4] / MULT_jointstate, params_mult[5] / MULT_jointstate, params_mult[6] / MULT_jointstate, params_mult[7] / MULT_jointstate]
        set_servo_setpoint(q)
```

it seems to me the intention of the code here is to check if more than one integer was read, and if so, to assume it was a vector of points. If that is the case, shouldn't this line be:

`if params_mult[0] > 1`

If I'm mistake, please explain what I'm missing!

Thank you!

fmauch, 2019-12-20:
No, this line is actually about checking the keepalive signal sent from the external source. For instance, a connection could be established, but no controller running on ROS side. In this case, we'll jump into the else part, where we set the current position as target position.

`params_mult[0]` will only be `0` when reading from the socket failed. In this case we'll reduce the internal keepalive counter. As soon as this gets below 0, we exit as we have lost connection to the external control point. 

I hope, that explains this. If you have any comments or suggestions for the script, please let us know :-)

gavanderhoorn, 2019-12-20:
@fmauch: should we add your explanation as a comment to the script? Would avoid questions like these in the future.

fmauch, 2019-12-20:
Yes,but after merging #1.This will change a lot of things, especially in this region.

jbcpollak, 2019-12-20:
> No, this line is actually about checking the keepalive signal sent from the external source. For instance, a connection could be established, but no controller running on ROS side. In this case, we'll jump into the else part, where we set the current position as target position.

ok, I think that makes sense - let me echo that back to make sure I've got it, the semantics of socket_read_binary_integer make this a bit tough to reason about. 

In short, `params_mult[1]` is `type` value [from here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/16cc69ddc1bf5cb944f5e627aeffdaa4fa59ed7c/ur_robot_driver/include/ur_robot_driver/comm/reverse_interface.h#L80), which is effectively a command to run:

 1 = keepalive
 2 = servoj

it looks like the linked PR changes it to always be the keepalive, and appends the control mode to the end of the command. this is interesting because the next step of my project would be to look into adding additional control options to the protocol, so I'll dig into the PR.

fmauch, 2020-05-12:
This looks like it can be closed.

closed 2020-05-12 by fmauch