UniversalRobots/Universal_Robots_ROS_Driver #73: ROS support for UR16e

NielsH12, 2019-12-27:
There is still no support for the UR16e

There are two posts on the forum requesting this:

https://forum.universal-robots.com/t/ur16e-ros-driver-support/5395
https://forum.universal-robots.com/t/looking-to-buy-ur16e-what-is-current-18th-dec-2019-ros-support/5890

It seems that relatively little work is needed to support this. (Some configuration files with dimensions and such)

Can this be prioritized higher please?

rosslloydqbot, 2020-01-03:
Just to add my +1 to this request as well. It is driving a decision to purchase a UR16e vs a competitor product at the moment.

What would need to be done to get this up and running?

gavanderhoorn, 2020-01-03:
On the `universal_robot` side, `.xacro`s, config files and meshes need to be created and added. Additionally `ur_gazebo` would need to be extended, `ur_kinematics` (if we want to be thorough) and a MoveIt configuration would need to be added.

rosslloydqbot, 2020-01-03:
Thanks gavanderhoorn. Does that lean more towards a timescale of years, or months? Sorry to put you on the spot but it will help enormously with the buying decision. Thanks again.

gavanderhoorn, 2020-01-03:
I would say days. But it requires someone to do the work.

There is nothing very difficult about this. It's just grunt work.

fmauch, 2020-01-07:
I started working on that on a train trip during the Christmas holidays and quickly realized this would be getting a mess quite quickly for multiple reasons:

  - Currently my fork of the `universal_robots` repository is required to run this driver. I would prefer to get this merged before adding a ton of changes on top of that.
  - The plan is to rebase my changes ontop of ros-industrial/universal_robot#371
  - I would like to reduce the double amount of work where possible.

For the reasons above I decided to start working on top of ros-industrial/universal_robot#371 the current status is on https://github.com/fmauch/universal_robot/tree/ur16e

rosslloydqbot, 2020-01-10:
Hi Felix
Thank you so much for the update, and for the hard work you are putting into this! I appreciate this is something that is difficult to quantify, but do you have a feel for the likely timeframe for this to be completed? 

In terms of applications, there are two main things we would like to achieve with the arm. 
1) To use ROS moveit to send the robot cartesian trajectories
2) Additionally make use of the UR16e's force sensing / impedance control in all three cartesian axes through ROS. 

Will both of these be available once the work is complete?
Thanks again

fmauch, 2020-01-29:
> Hi Felix
> Thank you so much for the update, and for the hard work you are putting into this! I appreciate this is something that is difficult to quantify, but do you have a feel for the likely [...]

As you can see in testing ros-industrial/universal_robot#477, moveit support for UR16 is currently not yet implemented.

> 2. Additionally make use of the UR16e's force sensing / impedance control in all three cartesian axes through ROS.

Currently, no native support for UR's force_sensing exists. You could make use of https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers to achieve similar behavior using ROS.

> 
> Will both of these be available once the work is complete or require some coding our end?
> Thanks again

None of these are on our current schedule. Any work on this end is welcome, especially regarding MoveIt! support.

fmauch, 2020-09-30:
Added basic ur16e support in #270, therefore closing this.

closed 2020-09-30 by fmauch