UniversalRobots/Universal_Robots_ROS_Driver #74: UR10e lag problem with externalcontrol-1.0.1.urcap

gadringer, 2020-01-07:
*Introduction to the issue*
I successfully downloaded and built the ROS driver on the ROS PC. I installed the External Control URCap and created a new program ‘external_control_ros’ on the UR10e as explained in the setup guide. **However, the start of the program ‘external_control_ros’ takes around two minutes before the program actually executes** and shows that the connection to the remote PC could not be established. However, IP addresses and URCap settings should be fine because on the day before it was already working, i.e. I was able to plan and execute a trajectory with MoveIt. However, after saving the settings (installation, program) and a restart of the UR on the next day, it was not working anymore.

# Versions
 - ROS Driver version: Melodic
 - Affected Robot Software Version(s): 5.5.1 / 5.6.0
 - Affected Robot Hardware Version(s): UR10e
 - URCaps Software version(s): external control 1.0.1

# Impact and issue details
Installation of 'externalcontrol-1.0.1.urcap' leads to a lag time between pressing "play" on UR and program start. The lag is independent on the program. For example, if I start a different program (not ROS related), then the **program start takes around two minutes as well but the execution works normally**, i.e. the robot moves its desired path. If I uninstall the External Control URCap, then the lag is reduced to just a few seconds but this is still slightly slower than prior the first URCap installation. A reinstall of the URCap leads again to the same behaviour (lag of two minutes). Sometimes not just the program but also the whole software seems to lag. After the upgrade to UR Software 5.6.0 I still have the same problem. Reinstalling the URCap did not help either. Details should be shown in the log history. The problems with URCap started on 10th of December 2019 (2019-12-10)

I also tried two different SD cards, installed new UR robot images (5.5.1 & 5.6.0) on it and included the important files from the original SD card. I followed the instructions of the document 'install-new-image-all-types.pdf' (Version 0.0.2) for the data backup and image installation. However, after normal boot-up and installation of the external control URCap, i still had the same lag problem and could not establish the connection to the remote PC.

IP addresses:
- Remote PC: 192.168.10.120
- UR10e: 192.168.10.122
- External Control Host: 192.168.10.120

## Steps to Reproduce
*Install externalcontrol-1.0.1.urcap on UR10e*

Help would be very appreciated.

Thanks and best regards,
Stefan

[log_history.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4031467/log_history.txt)

fmauch, 2020-01-08:
This seems very strange. The `externalcontrol-1.0.1.urcap` should not be interfering with anything when it is not part of the current program.

gadringer, 2020-01-08:
Yeah, I would not expect that either. However, I still think that it might be a problem with the URCap itself because a very similar behaviour was reported in another forum (https://dof.robotiq.com/discussion/726/lag-time-between-pressing-play-on-ur-and-program-start-with-ucg-1-1-1) for a different URCap. Because of the similarity, I also explained my issue in this forum.

fmauch, 2020-01-08:
When saving a program, it is transformed to `URScript` code which is then being compiled before execution. It would be interesting to see what this `URScript` looks like.

Could you please
- Save the robot program from the teach pendant as a name, lets call it ''\<robotProgramName>''
- Save the installation from the teach pendant, the default name is ''default.installation''

and attach the following files from the robot to this issue:
 - The log: ''/root/log_history.txt''
 - The robot program: ''/programs/\<robotProgramName>.urp''
 - The generated script: ''/programs/\<robotProgramName>.script''
 - The installation: ''/programs/default.installation''

You can copy the files using scp with user: root, pw: easybot

gadringer, 2020-01-08:
Thanks for the prompt reply. These are the files: 
[filesUR10e.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4034628/filesUR10e.zip)

fmauch, 2020-01-08:
On a first glance the produced files don't show anything weird. The generated script only contains the default initialization code and the code coming from our URCap where the ROS driver was not connected while saving the program.

From your logs it is also visible that the program is running last on the 9th of December:
```
3.5 :: 0000d10h07m29.334s :: 2019-12-09 17:47:45.695 :: -3 :: C0A0:0 :: null :: 1 :: steptime=0.002 ::  :: null
3.5 :: 0000d10h07m29.336s :: 2019-12-09 17:47:45.695 :: -3 :: C0A0:0 :: null :: 1 :: External control active ::  :: null
```

After that the program can't connect to the ROS machine anymore. However, I can't see any hint of the 2min delay:

```
3.5 :: 0000d10h38m26.503s :: 2019-12-10 10:00:30.679 :: -5 :: C0A0:7 :: null :: 1 ::  :: Programm external_control_ros wird gestartet... (Zuletzt gespeichert: 2019-12-10 09:55:46) :: null
3.5 :: 0000d10h38m39.973s :: 2019-12-10 10:00:30.795 :: -3 :: C0A0:7 :: null :: 1 :: external_control_ros :: Programm external_control_ros gestartet :: null
3.5 :: 0000d10h38m43.093s :: 2019-12-10 10:00:33.856 :: -3 :: C0A0:7 :: null :: 1 :: external_control_ros :: Programm external_control_ros gestoppt :: null```
```

This shows that the program is being started right after the start is requested and is shutting down rougly 2 seconds afterwards as this is the timeout for connecting to the external PC.

I cannot say anything to the 2min delay, I'll try to come back on that later. First, let's find out why connection cannot be established.

While you already know how to connect to the robot via ssh could you try to ping your ROS pc from the robot controller?

When the ROS driver is running, what is the output of `sudo netstat -tlpn | grep 5000` (on the ROS pc)? It should be similar to the following:

```bash
> $ sudo netstat -tlpn | grep 5000
tcp        0      0 127.0.0.1:5000          0.0.0.0:*               LISTEN      1884/python3        
tcp        0      0 0.0.0.0:50001           0.0.0.0:*               LISTEN      7391/ur_robot_drive 
tcp        0      0 0.0.0.0:50002           0.0.0.0:*               LISTEN      7391/ur_robot_drive
```

While we're at it: You can try connecting to the ROS driver from the robot controller manually (inside an ssh session) by executing `nc 192.168.10.120 50002`. Once this is started, every character you type will be sent to the ROS machine on the respective socket. Try requesting the program by typing `request_program` followed by an <kbd>Enter</kbd> keystroke. You should get a whole lot of URScript code.

If the test above is working I don't see any reason why the driver should not be working. If this doesn't work there's probably a problem with your network setup (e.g. a firewall?)

gadringer, 2020-01-08:
Test 1: Ping of the ROS pc from the robot controller is possible (ping_result.txt)
Test 2: Netstat should be fine too (netstat_result.txt)
Test 3: Manual connection from controller to ROS PC is not possible (manual_connection_result.txt). The connection timed out after 2 minutes and 10 seconds.
[manual_connection_result.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4035070/manual_connection_result.txt)
[netstat_result.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4035071/netstat_result.txt)
[ping_result.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4035072/ping_result.txt)
[ur_robot_driver_log.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4035073/ur_robot_driver_log.txt)

fmauch, 2020-01-08:
Please post the results inline the next time. That makes it easier to follow this thread.

From your tests it looks as if your machine is blocking connection attempts. Probably you have a firewall running on your ROS machine. Please try to fix that first such that this connection works. Otherwise the URCap will also not be able to communicate with the ROS driver.

gadringer, 2020-01-08:
Ok, sorry for that. I will post everything inline next time.

I have a direct connection to the UR controller with the LAN cable and I should not have any firewall enabled (Ubuntu 18.04). Anyway, I will do some tests with my colleague and will let you know about the progress.

gadringer, 2020-01-08:
Good news, it is working now :)

You were right! After some tests with my colleague, we found out that pretty much all ports were closed on my ROS PC (`sudo nmap -p 1-65535 MY_IP_Address`) because a firewall was active (`sudo ufw status`). I really do not know why a firewall was installed and why it was even enabled...

However, I just needed to disable the firewall UFW (Uncomplicated Firewall) to get it working (`sudo ufw disable`). Now I can start my external control program and the connection to my ROS PC is established instantly. I can also start a ROS independent program without any lag. That`s great! However, it is very strange that my firewall had such a big effect on the UR.

@fmauch: Thank you very much for your support. I really had no idea anymore how I would be able to solve this.

fmauch, 2020-01-08:
That's great news. Ill go ahead and close this, then.

closed 2020-01-08 by fmauch

gavanderhoorn, 2020-01-08:
> However, it is very strange that my firewall had such a big effect on the UR.

@fmauch: this is indeed strange.

Is the urcap trying to connect to the PC, even if it's not part of the program / active?

Could it be trying to connect regardless, and with closed ports, get a failure immediately, but with firewalls, has to wait for connection attempts to time-out (as FWs typically `DROP`, instead of `REJECT`)?

fmauch, 2020-01-08:
@gavanderhoorn Thanks for pointing that out again. I created an issue in the external_control repo.