UniversalRobots/Universal_Robots_ROS_Driver #76: Add missing calls from dashboard server

fmauch, 2020-01-10:
There are a couple of calls missing from the dashboard server, especially since Versions 3.12/5.6

 - [ ] PolyscopeVersion
 - [ ] setUserRole <role>
 - [ ] get serial number
 - [ ] get robot model
 - [ ] get operational mode
 - [ ] is in remote control
 - [ ] set operational mode <mode>
 - [ ] clear operational mode

MinhDangTG, 2020-05-04:
Hello fmauch,

it seems liked an already-answered question, but do we have a service/method to turn on the electricity of the control box instead of pressing the power button on the teach pendant? since we had the shutting down service, I though that it would be handy if we had a similar method of the other way around. 

It seems almost impossible since the shutting-down will lose all the connection between the robot and ROS. A solution may be connecting another button or an arduino to pins in the control box (liked the safeguard stop) to booting the control box. 

Thanks.

fmauch, 2020-05-04:
> Hello fmauch,
>
> it seems liked an already-answered question, but do we have a service/method to turn on the electricity of the control box instead of pressing the power button on the teach pendant? [...]

I think, this is not possible to do. 

In the scope of this issue it is absolutely impossible, as obviously if the robot is switched off, there is no dashboard server to send a command to.

A physical wiring as you suggested it might work, if you can get access to the necessary pins inside the controller box. I guess you can, as I have seen UR arms integrated on mobile robot platforms, but there was still another physical button to press.

Another solution might be Wake-On-LAN, but I am not sure whether this is supported by the controller PC / would require modification of the controller PC.

Please let's not discuss this further inside this issue, as this is out of scope for the dashboard server.