UniversalRobots/Universal_Robots_ROS_Driver #77: Sending URScript commands for gripper have no effect

ingmar12345, 2020-01-13:
# Summary
I am trying to control the Robotiq 2f-85 gripper on the UR5 by sending URScript commands.

I am publishing the commands to the /ur_hardware_interface/script_command topic

Example:

rostopic pub -1 /ur_hardware_interface/script_command std_msgs/String "rq_open()"

rq_open() and rq_close() are used to open and close the gripper. The gripper is activated before exetuting these commands.

The script commands work fine when when executing them as separate Scipt Code nodes on the PolyScope tablet.

Why won't the commands work when publishing them to the /ur_hardware_interface/script_command topic?

The ROS driver is running on both the robot side and PC side.

I am running ROS Kinetic on Ubuntu 16.04 Xenial natively on a laptop PC.

# Versions
 - ROS Driver version: 0.0.3
 - Affected Robot Software Version(s): 3.9.1
 - Affected Robot Hardware Version(s): CB3.0
 - Robot Serial Number: 2015351381
 - UR+ product(s) installed:
 - URCaps Software version(s): Robotiq_grippers 1.2.1, External Control 1.0.1

# Impact
The coupling between the UR5 and the Robotiq gripper will be upgraded soon to a wireless interface through the wrist socket, which results that the gripper cannot be controlled through the official Robotiq driver (requires separate cable).

# Issue details
*This part is important in order to clarify the severity of the issue.*

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
I expected the gripper to perform opening and closing operations.

## Actual Behavior
The gripper didn't move at all.

## Workaround Suggestion
Using the Robotiq driver (http://wiki.ros.org/robotiq), please note that this driver is not compatible with the wireless wrist socket coupling interface for the Robotiq gripper and the UR5.

fmauch, 2020-01-13:
I don't see a reason, why this should not be working. Just to make sure: 
 - You are running a driver version more recent than b22365550fa31199deba4ff2bbe7a28479ea7f8f
 - Are other script commands working?

You could try 
```
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'popup(\"hello\")'"
```

ingmar12345, 2020-01-14:
Hi,
I tried other script commands and they are working perfectly fine.
It is still strange behaviour that the gripper script commands are not working.

fmauch, 2020-01-14:
I just realized the following: Probably the gripper URCap generates script in its installation part of program generation. If you don't have a program running (as in a program coming from the TP), then this code doesn't get injected. Probably this code generation creates the functions that you'd like to call.

As soon as you are sending those script commands, any running programs will be stopped and your new script command will be executed as a standalone program. You can try to send the commands as a [secondary program](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/secondary-program-17257/). Depending on the actual implementation this could work. Otherwise, you'd have to include the function definitions inside the script that you are sending through the ROS driver. Note, you can extract the generated script code after saving a program on the TP. See [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829) for details.

gavanderhoorn, 2020-01-14:
@fmauch: would this be something to add to the FAQ? Or perhaps a more extensive doc that explains why things are the way they are and work the way they work?

fmauch, 2020-01-14:
Yes, that seems reasonable.

carofel, 2020-02-17:
Hi, 
I am struggeling with the same issue. I tried to adapt the ros_control.urscript with a secondary programme. Is there an example how to add an additional function and how to handle it with the threads? It is the correct procedure? Hope it didnot miss anything. Thanks in advance.

fmauch, 2020-02-17:
@carofel Depending on your use-case this might be the correct attempt, but it might as well not be. If you're unfamiliar with URScript, I suggest you start by reading the URScript manual. You can get it from here: https://www.universal-robots.com/download/

However, please let me quickly note that adapting the URScript file to your needs goes beyond the scope of an issue on this driver.

ingmar12345, 2020-02-17:
Secondary programs may not take any time to execute (https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/secondary-program-17257/)

I’ve tried to pass both the function definitions for the gripper and the external control program to the secondary port in the UR5.
This results in a crash, the fans in the controller stop spinning and I have to do a hard reset.

Karthik-Ranga, 2020-03-02:
I have Robotiq gripper connected to wrist socket. I tried sending the gripper functions through secondary programs but nothing happened. But when I sent the same script file when in remote control mode(through socket), the script executed successfully. 

This might be a dumb question. Where to keep/call the script function from? Is there a folder where you need to have script files so the urscript looks for?

when 
```
rostopic pub -1 /ur_hardware_interface/script_command std_msgs/String " "
```
called
should I call the script file name or the function name within the quotes?

Karthik-Ranga, 2020-03-02:
With use_tool_communication:=false 
and sending the script file, I get the below error though. 

```
[ERROR] [1583179698.329217700]: Sending data through socket failed.
[ERROR] [1583179698.329367664]: Could not send program to robot
[ERROR] [1583179698.329474076]: Error sending script to robot
```

felixvd, 2020-03-10:
We use a Robotiq 2F-85 gripper with a UR5 that is connected to the UR controller (not the wrist connector), which we used to control through UR scripts. I can confirm that those scripts do not work as secondary programs, resulting in a hard crash as reported earlier.

Our workaround (which has been working fine) is to restart the program through the `dashboard` services.

It would be great if the gripper could be controlled while the robot is moving, but I guess it is not very important for most people.

Karthik-Ranga, 2020-03-10:
> We use a Robotiq 2F-85 gripper with a UR5 that is connected to the UR controller (not the wrist connector), which we used to control through UR scripts. I can confirm that those scripts do not work [...]

Did any other scripts worked for you? Like controlling I/O pins or something like that? 
In my case, no script works. Sending the script did not crash the robot, but sending through a socket program did result in crash. 
I'll try your suggested method on restarting through dashboard services.

felixvd, 2020-03-10:
I haven't tried any other scripts as secondary programs, no. I am not sure what you mean by "sending through a socket program", but I would hope that socket communication and setting I/O would work in a secondary program.

fmauch, 2020-03-27:
setting I/Os from a secondary program should definitely work.

What you can always do is send arbitrary (working) script code as a primary program and then start the program containing the external control node again using the dashboard client as described by @felixvd above.

If you're not able to execute script code at all, that's something we should take care of. I've lost track of all the different discussions going on here with different problems.

To get back to the original question:
 * We confirmed, that you can indeed send and execute script code to the robot.
 * As far as I can see, the robotiq URCap defines a couple of custom functions such as `rq_open()` which are being added to each program running on the TP by the respective Installation Node. You will find this script code if you have a look at the generated script code as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829). When you send script code to the robot, this is being executed similar to starting a new shell. The custom functions are not defined here, so the calls don't work.
 * If you want this to work you have to make sure that the function definitions are contained in the script code you send to the robot. Whether it will be possible to execute this as a secondary program or not will depend on the specific script code.

fmauch, 2021-03-11:
There is one year gone by with this. I'm closing this for now. If anything new arises, feel free to reopen.

closed 2021-03-11 by fmauch

MilanPadhiyar027, 2022-08-01:
Hi, I am facing a similar issue with controlling the Robotiq end-effector using URScript commands through the ROS interface. Do we have any useful workaround or solution now?

MilanPadhiyar027, 2022-08-01:
> I have Robotiq gripper connected to wrist socket. I tried sending the gripper functions through secondary programs but nothing happened. But when I sent the same script file when in remote control [...]

Hi,
can you please tell how did you send the script in remote control mode?

fmauch, 2022-08-01:
> Hi,
> can you please tell how did you send the script in remote control mode?

Regarding remote_control mode, please see https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/README.md#remote-control-mode

Karthik-Ranga, 2022-08-01:
> Hi, can you please tell how did you send the script in remote control mode?

I have written a blog post on our company website on how to send gripper script commands in remote control mode. 
http://axisnj.com/controlling-a-universal-robots-cobot-using-python/

If you have any questions please let me know!

Yeesha-R, 2024-03-28:
@Karthik-Ranga , the link to the blog post does not work. Could you please provide a working link to the blog post? Thanks

Yeesha-R, 2024-04-02:
@Karthik-Ranga thanks for sharing the link