UniversalRobots/Universal_Robots_ROS_Driver #78: Wrong robot model for ur10

eirik-strand, 2020-01-14:
I noticed that the 3d model for the simulator does not match the one in rviz. I think this can be an issue with the driver or just the urdf. I have not made any changes to fmauch or the driver. The picture shows how the base link is 180 degrees wrong
[image]

fmauch, 2020-01-14:
Thanks for noticing! I thought, I checked that, but I will check again.

gavanderhoorn, 2020-01-14:
At least in `ros-industrial/universal_robot`  this is a known issue.

See also this note in the xacro (from [here](https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_description/urdf/ur10.urdf.xacro#L346-L356)):

```xml
  <!-- NOTE: this rotation is only needed as long as base_link itself is
             not corrected wrt the real robot (ie: rotated over 180
             degrees)
  -->
```

fmauch, 2020-01-14:
That rings a bell... Which is why I decided to [do it the same](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_description/urdf/ur10.urdf.xacro#L61-L62).

gavanderhoorn, 2020-01-14:
I actually don't know why the mesh is wrong. It's a very old package (2012) and it has always been like this. It's one of the things which https://github.com/ros-industrial/universal_robot/pull/371 should fix.

ndaley7, 2022-08-02:
> The rotation should not be there.
> 
> The mesh / kinematic chain should be fixed, then the compensation should not be necessary any more.

Good afternoon, Does this mean that this issue is fixed in the current version of the repository?

fmauch, 2024-06-12:
This should be resolved by now...

closed 2024-06-12 by fmauch