UniversalRobots/Universal_Robots_ROS_Driver #79: Unable to control I/O states on UR10 (Python)

Sammysedo, 2020-01-16:
Hi. I'm working on a project where I need to pick up objects using Universal Robots UR10 and a vacuum gripper. So far I am seeing positive results in executing move commands using movegroup python interface. I am, however, experiencing problems in controlling digital output states using my script. I am using the driver in this repository along with the [ROS-I Universal Robots package](https://github.com/ros-industrial/universal_robot).

Here's my import list:
```
import sys
import copy
import rospy
import tf2_ros
import tf
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
#from math import pi
import math
from std_msgs.msg import String
from moveit_commander.conversions import pose_to_list
from ur_msgs.msg import SetIO
```
My function which I'm calling in the main function:
```
  def apply_vacuum(self):
    rospy.wait_for_service('/ur_hardware_interface/set_io')
    try:
        set_io = rospy.ServiceProxy('/ur_hardware_interface/set_io',SetIO)
        set_io(fun = 1,pin = 2,state = 1)
    except rospy.ServiceException, e:
        print "Something went wrong: %s"%e
```
When I run the script I get no rospy.ServiceException. On my ur10_bringup.launch terminal I'm getting this error whenever I run the script above:
`[ERROR] [1579178138.697529807]: Could not get fresh data package from robot`

Running `$rosservice list` yields `/ur_hardware_interface/set_io`. I can even do a rosservice call, which results in True, although nothing seems to happen:
```
rosservice call /ur_hardware_interface/set_io 1 2 1
success: True
```

I'd like to know if this is a known issue or whether there's a workaround or I'm missing something. I'm running ROS Kinetic on Ubuntu 16.04 LTS with Lenovo Thinkpad X260. My gripper is properly wired and connected and I can directly control the said output from the I/O tab on the teach pendant.

This issue has also been posted at answers.ros.org:
[https://answers.ros.org/question/341865/unable-to-control-io-states-on-ur10-python/](https://answers.ros.org/question/341865/unable-to-control-io-states-on-ur10-python/)

megantuttle, 2020-01-16:
Do you have PROFINET or Ethernet enabled on the UR?  This sounds a lot like #56.

Sammysedo, 2020-01-16:
> 
> 
> Do you have PROFINET or EtherNet/IP enabled on the UR? This sounds a lot like #56.

I'm using EtherNet/IP. So far the communication has worked perfectly.

megantuttle, 2020-01-16:
I was experiencing the same issue, but with PROFINET enabled.  This is a known problem; there are IO conflicts with the RTDE interface.  See [this post](https://forum.universal-robots.com/t/rtde-inputs-with-ethernet-ip-activated-problem/3655) on the UR developer forum.

Sammysedo, 2020-01-16:
I have to check that out. I will post further updates as soon as I get back to the robot and begin backtracing the issue. Have you tried using the driver included within the [ros-i package](https://github.com/ros-industrial/universal_robot) to fix the issue? I thought it had been deprecated long ago but apparently it's still being updated regularly.

gavanderhoorn, 2020-01-16:
> Have you tried using the driver included within the [ros-i package](https://github.com/ros-industrial/universal_robot) to fix the issue? I thought it had been deprecated long ago but apparently it's still being updated (somewhat) regularly.

As the (main) maintainer of those packages I can assure you that `ur_driver` is fully deprecated and *not* being updated at all.

It also will not work (reliably) with anything but CB2 controllers with Polyscope `1.8.x` max.

fmauch, 2020-01-17:
@megantuttle  We are still lacking of understanding #56. As stated there, field conflicts between rtde and field busses should only affect fields currently not being used. But yes, it would be interesting, whether disabling field busses would help @Sammysedo.

Sammysedo, 2020-01-17:
@fmauch I was unable to fix the issue for now. Currently using a proximity switch as a workaround which works fine for now.

It also turns out I was being misleading on my last comment. As far as I know, I am not using Ethernet/IP, but a standard Ethernet link with static ip addresses defined on the Polyscope and the computer I am using. Do I need to switch to PROFINET or Etherner/IP for I/O communication?

fmauch, 2020-01-17:
>Do I need to switch to PROFINET or Etherner/IP for I/O communication?

No, absolutely not. 

It should be working out of the box.

Out of curiosity: Does the following command work for you:

```
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(2,True)'" 
```

This will stop your running program, this is just out of interest.

Sammysedo, 2020-01-18:
> Out of curiosity: Does the following command work for you:
> 
> ```
> rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(2,True)'" 
> ```
> 
> This will stop your running program, this is just out of interest.

This seems to work. I could probably write a simple publisher into my script, but I guess it wouldn't be much use since it stops the program like you said.

gavanderhoorn, 2020-01-18:
@Sammysedo: the goal here is not to come up with alternative ways to control IO instead of using the driver.

@fmauch suggested that command to see whether the underlying infrastructure works at all.

It's just a diagnostic step.

Sammysedo, 2020-01-18:
@gavanderhoorn I realize it was a diagnostic step, I was just wondering.

Karthik-Ranga, 2020-02-06:
Has anyone got this working? Even I am facing the same problem.
But I got an Import error. There was no ur_msgs.msg package. I replaced ur_msgs.msg with ur_msgs.srv and the python script executed. But got below error. I tested for the tool though.

```
from ur_msgs.srv import SetIO
```

````
rospy.wait_for_service('/ur_hardware_interface/set_io')
set_io = rospy.ServiceProxy('/ur_hardware_interface/set_io',SetIO)
set_io(fun = 4,state = 24)
````

> [ INFO] [1580997893.269454074]: Robot requested program
>[ INFO] [1580997893.269606509]: Sent program to robot
>[ INFO] [1580997893.604252850]: Robot ready to receive control commands.
>[ERROR] [1580997917.331150139]: Cannot execute function 4. This is not (yet) supported.

I don't have anything connected to the I/O pins for now. So I tested with the tool. Is there a way to activate the gripper/open/close using the above method? or is it still experimental? 
I will soon test with the some devices connected to I/O pins and see if this works.

Karthik-Ranga, 2020-02-06:
Okay I tested with the I/O pins with no devices connected. I just monitored on the I/O tab on the teach pendant.
```
rospy.wait_for_service('/ur_hardware_interface/set_io')
set_io = rospy.ServiceProxy('/ur_hardware_interface/set_io',SetIO)
set_io(fun = 1,pin = 3,state = 0)
rospy.sleep(3)
set_io(fun = 1,pin = 3,state = 1)
```
I was able to toggle the states. 
I need to know if there is a way to do the same for the tool states.

fmauch, 2020-02-07:
> Okay I tested with the I/O pins with no devices connected. I just monitored on the I/O tab on the teach pendant.
>
> ```
> rospy.wait_for_service('/ur_hardware_interface/set_io')
> set_io = [...]

For the service to set outputs, the same notation as on the topic `/ur_hardware_interface/io_states` is used. In this, tool digital outputs are pins 16 and 17. I am aware that this is not the most elegant solution, though...

fmauch, 2020-02-07:
> This seems to work. I could probably write a simple publisher into my script, but I guess it wouldn't be much use since it stops the program like you said.

Sorry for coming back on this that late. If the direct script command works it means that the error seems to be inside the RTDE communication. To debug this further, it would be very beneficial to have #24 implemented... I'll try to give this some more priority.

Karthik-Ranga, 2020-03-04:
> No, absolutely not.
>
> It should be working out of the box.
>
> Out of curiosity: Does the following command work for you:
>
> ```
> rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: [...]

I tried this but the program did not stop. Nothing happened. Is there a solution to this? 
is there a way to send direct script command?

Circle-Kai, 2020-04-30:
**service is not responding**
I received this message on the terminal.
```
[ INFO] [1588237985.495034925]: Robot requested program
[ INFO] [1588237985.495100398]: Sent program to robot
[ INFO] [1588237985.760502715]: Robot ready to receive control commands.
[ERROR] [1588237989.833273575]: Could not get fresh data package from robot
```
This is my command line. 
```
$ rosservice call /ur_hardware_interface/set_io 1 2 1
success: True
```

jorgenfb, 2020-12-15:
I'm experiencing this issue on my machine. I have been testing on version 3.9 and 3.14 using the a simulator inside Virtualbox and on a real UR10 (3.9). ROS version kinetic.

I've started the driver with:
```
roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=192.168.56.102 headless_mode:=false
```

Calling the service to set an digital out pin returns success, but there the IO is not set.

This is the command I'm trying to run and its response:
```
> rosservice call /ur_hardware_interface/set_io "{fun: 1, pin: 2, state: 1.0}"
success: True
```

Setting IO using a script command works:
```
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(2,True)'"
``

Is there anything I can do to help find the issue?

fmauch, 2020-12-23:
As I said [earlier](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/79#issuecomment-583417360) with the RTDE handshake validation that we have implemented by now I would assume that if the driver doesn't complain about an RTDE field already being taken by another process, there shouldn't be a problem with that.

The next thing to debug this would be to check whether the data package sent to the robot actually contains the correct values. This could be quickly debugged by adding the following after [this line](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/rtde/rtde_writer.cpp#L56):

```c++
std::cout << *package << std::endl;
```

One step deeper would be to inspect the actual network package.

jorgenfb, 2021-02-05:
I can report back that the issue I was having was totally unrelated. An error in our fork made the RTDE writer never being started. Thanks for helping us debug it fmauch. Your tips pointed me in the correct direction.

fmauch, 2021-02-05:
@jorgenfb That's nice to hear!

closed 2023-06-20 by github-actions[bot]