UniversalRobots/Universal_Robots_ROS_Driver #81: Could not simultaneouosly control multiple robots

xiaohuits, 2020-01-17:
Hi, I am testing this driver with a dual-arm UR5e setup in our lab. I can correctly start the driver for one arm at a time but couldn't find a way to control the two arms at the same time (using a single computer). 

I have two launch files (`left_arm_bringup.launch` and `right_arm_bringup.launch`) that can correctly start the driver for the left/right arm. These launch files are just copies of `ur5e_bringup.launch` with IP and calibration file provided. 

Here is what I have tried to launch driver for both arms at the same time:
```
file: both_arm_bringup.launch
<?xml version="1.0"?>
<launch>
  <include file="$(find macs_ur_launch)/launch/left_arm_bringup.launch" ns="left_arm"/>
  <include file="$(find macs_ur_launch)/launch/right_arm_bringup.launch" ns="right_arm"/>
</launch>
```
I wanted to prefix the services/topics with different namespaces for each arm, so the message sent will not conflict. However, it seems that this is not going to work. It will produce a runtime error says "Connection setup failed for :50002". See the log here: [log.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4075170/log.txt)

It seems that the underlying connections can't be launched multiple times? Is there any other way that I can built the connections for both robots at the same time?

fmauch, 2020-01-17:
First of all: I don't think, it is a very good idea to control two arms from one machine, as they will share resources (computation wise and (depending on your network setup) communication wise).

However, if you want to go ahead and do this, you'll have to adapt the reverse_port 50001 and the script_sender_port 50002 for one of the drivers (They are launch file parameters).

See #57 where this got implemented.

Please try this out and let us know whether this works in your situation.

xiaohuits, 2020-01-18:
Thanks for the quick comments! The two arms will collaborate together for a common task so it will be nature for me to putting everything in the same machine first and plan together. Thanks for the head-up information through. 

I changed the reverse_port and script_sender_port of the second arm to a different number and now the both drivers can start without error.  

However, the ```rqt_joint_trajectory_controller``` will fail when I choose one of the controller manager namespace. Not sure if the driver is working correctly or not, or this is an issue with the rqt tool. I can use the rqt tool when only one driver is running.

> Error message from rqt_joint_trajectory_controller
> Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 271, in _on_cm_change
>     self._update_jtc_list()
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 243, in _update_jtc_list
>     self._robot_joint_limits = get_joint_limits()  # Lazy evaluation
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_joint_trajectory_controller/joint_limits_urdf.py", line 21, in get_joint_limits
>     description = rospy.get_param(key)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 465, in get_param
>     return _param_server[param_name] #MasterProxy does all the magic for us
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
>     raise KeyError(key)
> KeyError: 'robot_description'

I am trying to use MoveIt! interface to test out. Will update when I have more information.

fmauch, 2020-01-20:
Running everything in a namespace is quire important, as well. See the discussion in #45.

felixvd, 2020-01-25:
I am having issues with this as well. With `ur_modern_driver`, we used to have two UR robots prefixed with a unique name, by specifying a `prefix` parameter in the launch file spawning the driver node and in the URDF file calling the UR macro, so that both driver nodes publish in the root namespace and connect to the same `move_group`. However, in `ur_robot_driver`, the `tf_prefix` parameter handed into `ur_control.launch` does not seem to affect the joint names that are published to `/joint_states` - they are published without any prefix (e.g. `shoulder_pan_joint` instead of `b_bot_shoulder_pan_joint`).

Is this by design? How are multiple robots meant to be controlled together? Do we need to put them in different namespaces and create two entirely separate move_groups now? Maintaining the same `PlanningScene` seems error-prone and like a hassle.

I would appreciate any comments you have, also regarding how `roscontrol` would work with multiple robots, as I have not used it much. You can see the `moveit_config` we had in our system [here](https://github.com/o2as/ur-o2as/tree/master/catkin_ws/src/o2as_moveit_config).

gavanderhoorn, 2020-01-25:
`tf_prefix` is indeed only used to prefix the TF frames published by the driver ([here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/a926789adc0cefc616beb0fda15793cdcd65831b/ur_robot_driver/src/ros/hardware_interface.cpp#L111-L112)).

There is no prefix for the joint names specifically, as you're supposed to tell the `hardware_interface` which joints you'd like it to expose. The default joint names are defined in the `.yaml` files in the [config](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/c9b25b5a790a9efd6492a42f1706aeadb0803ca9/ur_robot_driver/config) directory or `ur_robot_driver`. For the UR5 fi:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/c9b25b5a790a9efd6492a42f1706aeadb0803ca9/ur_robot_driver/config/ur5_controllers.yaml#L6-L13

This list of joints is then used everywhere (note the anchor (`&robot_joints`), it's used later in the file to refer back to this list, using an alias (`*`)).

The `ros_control` `JointStateController` is then responsible for publishing `JointState` messages using information coming from the `hardware_interface` (it's not really a controller, but that's `ros_control` nomenclature). `JointStateController` publishes for *all* joints registered in a `hardware_interface`.

The `ur_robot_driver` `hardware_interface` only registers the names specified in the `hardware_interface` sub section of the config `.yaml` (shown above).

Taking these two things together: you don't really "prefix joint names" in so much as you specify completely new ones (the fact that it's a prefix is only apparent to you, a human). The `hardware_interface` will then register those names as names for the joints (internally it really only uses indices into arrays, names are for humans :man_shrugging:) and the `JointStateController` will start publishing `JointState` messages for all those joints *in that specific `hardware_interface`*.

So for a multi-robot setup, start two instances of the driver (each in their own namespace), make sure to create two copies of the configuration `.yaml` (one for each of the driver instances), load them in their respective namespaces and make sure to specify the correct joint names for each.

You should end up with two `ur_robot_driver` instances each publishing to a `joint_states` topic in their own namespace, which you could then combine into a single publication on `/joint_states` (or whichever topic you prefer) by running `joint_state_publisher` with a `source_list` parameter correctly configured.

---

PS: you appear to have an entire Catkin workspace in a git repository. Not very elegant .. ;)

xiaohuits, 2020-01-28:
Update: Thanks everyone! I can simultaneously run two drivers to control two UR5e robots with moveit now. Here are the summary of the tricks applied:
- Each robot will need to have different reverse_port and the script_sender_port value. 
- Use a namespace for each group of launch files that are used for launch the driver, moveit package and rviz for a single robot.

Details of my launch files can be found [here](https://github.com/macs-lab/macs_dual_arm)

felixvd, 2020-01-28:
Thanks for the detailed response! I will make separate versions of `ur5_controllers.yaml` with the joint names changed to include the prefix. I am not sure of some points:

- if they will need to be in their own namespace or changing the `ur_robot_driver` node's name will suffice
- in case they do need their own namespace, which other nodes from `ur_common.launch` need to be in it (e.g. `ur_robot_state_helper`, `controller_manager`, `ur_tool_communication_bridge`)
- which of those nodes could be shared between the `ur_robot_driver` nodes
- how to configure the `source_list` parameter of the `joint_state_publisher`. I would have simply tried to remap the `/joint_states` topic from the driver's namespace to the global one.

I am mostly noting this for documentation. I assume I would know if I was more familiar with the way `roscontrol`'s `controller_manager` works. I will update when I get to it. 

> PS: you appear to have an entire Catkin workspace in a git repository. Not very elegant .. ;)

Actually it's even more: a catkin workspace, the environment Dockerfile and the scripts to install and run it locally. It's a little custom, but fully documented in the wiki, and from experience it helps onboard new users and streamline their installs immensely.

gavanderhoorn, 2020-01-28:
> Thanks for the detailed response! I will make separate versions of `ur5_controllers.yaml` with the joint names changed to include the prefix. I am not sure of some points:
> 
> * if they will need to be in their own namespace or changing the `ur_robot_driver` node's name will suffice

both driver instances need their own namespace.

Reason is that it all comes down to parameters, and those are looked up in the namespace the driver node is in. As there's no way to have multiple sets of parameters in the same namespace with the same name, you'll need namespaces.

> * in case they do need their own namespace, which other nodes from `ur_common.launch` need to be in it (e.g. `ur_robot_state_helper`, `controller_manager`, `ur_tool_communication_bridge`)

All of them.

As @xiaohuits wrote: just namespace the launch files. The rest should work automatically.

> * which of those nodes could be shared between the `ur_robot_driver` nodes

None (or at least: I would not do that if you're just starting out).

> * how to configure the `source_list` parameter of the `joint_state_publisher`. I would have simply tried to remap the `/joint_states` topic from the driver's namespace to the global one.

Remapping can work, but that will mean that `/joint_states` consumers will receive "incomplete" `JointState` messages (ie: first of robot1, then robot2, etc).

If you use the JSP with `source_list` set to something like `['/ns_of_robot1/joint_states', '/ns_of_robot2/joint_states']` it will coalesce those into a single message and publish it on `/joint_states`. Subscribers to that topic will see a single message describing the whole system, which is much more robust.

> I am mostly noting this for documentation. I assume I would know if I was more familiar with the way `roscontrol`'s `controller_manager` works. I will update when I get to it.

I don't believe this is very tightly coupled to `ros_control` actually: it's just nodes which expect certain parameters to exist (buth with different values) and you want two sets of those parameters to exist. The typical pattern to follow is to place such nodes and their parameters in a namespace.

As to the joint names: I guess that could be called `ros_control` behaviour but in reality I've always thought the `prefix` parameter was a bit hacky. It's much cleaner to isolate things properly, and the way to do that is again namespaces.

> Actually it's even more: a catkin workspace, the environment Dockerfile and the scripts to install and run it locally. It's a little custom, but fully documented in the wiki, and from experience it [...]

Have you looked at `vcstool` and a couple of `.repos` files? Submodules can be rather confusing to newcomers, and setting up a ROS workspace with all dependencies takes 3 commands.

But it's not important.

fmauch, 2020-05-12:
Is there anything to add from a driver's perspective? Otherwise I would go ahead and close this.

xiaohuits, 2020-05-12:
I don't see a need to change the driver for now.

closed 2020-05-12 by fmauch

ManueU, 2021-02-13:
Hi, 
I am testing this driver with two UR5. I followed the instructions written in the issue, but I have some questions because the system doesn't work. 
I did the following things: 
- set the correct IP addresses
- change the "reverse_port" and the "script_sender_port" values for one of the two robots
- obtain two different calibration files
- create two different controllers.yaml files 

In the following, my launch file: 
```

<?xml version="1.0"?>
<launch>
<group ns="masterArm">
<include file="$(find ur_robot_driver)/launch/ur5_bringup.launch">
	<arg name="robot_ip" value="192.168.1.4"/>
	<arg name="tf_prefix" value="m_"/>
	<arg name="controllers" value="joint_state_controller"/>
	<arg name="stopped_controllers" value=""/>
	<arg name="controller_config_file" value="$(find ur_robot_driver)/config/ur5_controllers_master.yaml"/>
       <arg name="kinematics_config" value="$(find ur_description)/config/URcalib/master_calibration.yaml"/>
</include> 
</group>

<group ns="slaveArm">
<include file="$(find ur_robot_driver)/launch/ur5_bringup.launch">
	<arg name="robot_ip" value="192.168.1.2"/>
	<arg name="reverse_port" value="50003"/>
	<arg name="script_sender_port" value="50004"/>
	<arg name="tf_prefix" value="s_"/>
	<arg name="controllers" value="joint_state_controller my_cartesian_motion_controller"/>
	<arg name="stopped_controllers" value=""/>
	<arg name="controller_config_file" value="$(find ur_robot_driver)/config/ur5_controllers_slave.yaml"/>
       <arg name="kinematics_config" value="$(find ur_description)/config/URcalib/slave_calibration.yaml"/> 
</include>
</group> 
</launch>
```

The questions are: 

- **Do I have to change the joints names in both controllers.yaml and calibration.yaml files?** 
I can see the following errors changing the joints names in both files. **Why?** **In which way I can solve it?** 

[image]

- There's another error (the red one). **Do you know the way to solve it?**

[image]

- **What is the reason of this errors? Could you help me to solve it?** 

[image]

Thank you so much!

fmauch, 2021-02-14:
First of all, your `controllers.yaml` files seem to have errors. Without posting them it's hard to tell what is wrong.

Second, the `tf_prefix` argument is not passed to the description file. It would probably make sense to create your own description file where both robots are instantiated with the correct prefixes. See [this comment](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/81#issuecomment-578418515) how to configure the joint names and controllers.

gavanderhoorn, 2021-02-14:
@fmauch: there is the [`subst_value` attribute on `rosparam` elements](http://wiki.ros.org/roslaunch/XML/rosparam#Examples) which we *could* use to prefix names of joints in the `.yaml` files automatically with the value assigned to `tf_prefix`.

And the `prefix` arg of the `xacro:macro` could also be set from the `.launch` file.

That may make setting up multi-robot configurations a little easier.

(I'm not too sure about the `xacro:macro` part, it tends to confuse if you use `arg`s there too much)

ManueU, 2021-02-14:
Thank you so much for your quick responses! 

> Without posting them it's hard to tell what is wrong.

In the following, my controllers.yaml file: 
```
# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: &loop_hz 125
  

# Settings for ros_control hardware interface
ur_hardware_interface:
   joints: &robot_joints
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint

# Publish all joint states ----------------------------------
joint_state_controller:
   type:         joint_state_controller/JointStateController
   publish_rate: *loop_hz

# Publish wrench ----------------------------------
force_torque_sensor_controller:
   type:         force_torque_sensor_controller/ForceTorqueSensorController
   publish_rate: *loop_hz

# Publish speed_scaling factor
speed_scaling_state_controller:
   type:         ur_controllers/SpeedScalingStateController
   publish_rate: *loop_hz

# Joint Trajectory Controller - position based -------------------------------
# For detailed explanations of parameter see http://wiki.ros.org/joint_trajectory_controller
scaled_pos_joint_traj_controller:
   type: position_controllers/ScaledJointTrajectoryController
   joints: *robot_joints
   constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.2, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.2, goal: 0.1}
      elbow_joint: {trajectory: 0.2, goal: 0.1}
      wrist_1_joint: {trajectory: 0.2, goal: 0.1}
      wrist_2_joint: {trajectory: 0.2, goal: 0.1}
      wrist_3_joint: {trajectory: 0.2, goal: 0.1}
   stop_trajectory_duration: 0.5
   state_publish_rate: *loop_hz
   action_monitor_rate: 20

pos_joint_traj_controller:
   type: position_controllers/JointTrajectoryController
   joints: *robot_joints
   constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.2, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.2, goal: 0.1}
      elbow_joint: {trajectory: 0.2, goal: 0.1}
      wrist_1_joint: {trajectory: 0.2, goal: 0.1}
      wrist_2_joint: {trajectory: 0.2, goal: 0.1}
      wrist_3_joint: {trajectory: 0.2, goal: 0.1}
   stop_trajectory_duration: 0.5
   state_publish_rate: *loop_hz
   action_monitor_rate: 20

[... 74 lines cut ...]
# Cartesian motion controller   
my_cartesian_motion_controller:
    type: "position_controllers/CartesianMotionController"
    end_effector_link: "tool0"
    robot_base_link: "base"
    target_frame_topic: "target_frame"
    joints: *robot_joints

    pd_gains:
        trans_x: {p: 15.0, d: 3.0}
        trans_y: {p: 15.0, d: 3.0}
        trans_z: {p: 15.0, d: 3.0}
        rot_x: {p: 15.0, d: 3.0} 
        rot_y: {p: 15.0, d: 3.0}
        rot_z: {p: 15.0, d: 3.0}

```

> See this comment how to configure the joint names and controllers.

 I don't understand the procedure to set the joint names and controllers from the cited comment. Could you explain it to me in others terms? 

@gavanderhoorn  Is this can be seen as an alternative to the comment cited by @fmauch?

Thank you!

fmauch, 2021-02-15:
The suggestion by @gavanderhoorn was about automating the step of renaming the joint_names in the `controllers.config` file. This could certainly be an option, but I would leave it out of the loop of your troubles @ManueU.

Regarding your controller file: There seems to be something wrong with how the controller file gets uploaded to the parameter server. You can inspect the parameters on the parameter server using `rosparam list` and`rosparam get` in order to find out what is wrong.

ManueU, 2021-02-15:
Thank you! @fmauch 
I noticed that the presented errors depended from a distraction error: I have to define the "my_cartesian_motion_controller" in the ur5_controllers_slave.yaml and not in the ur5_controllers_master.yaml. Sorry!

But, now the error is the following: 

[image]

**Do you have any advice for me?**

Thank you!

fmauch, 2021-02-15:
As I've said earlier, you'll have to create a description matching your robot setup. With two robots you basically have two options:

1. Create one common workcell description that contains both robots with their respective tf_prefix similar to the one [containing one arm](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_description/urdf/ur5_robot.urdf.xacro)
2. Create two separate descriptions only containing one arm and loading them into the namespaces (which you currently do), but also setting the prefix there. For this, you'll also have to create your own xacro files and set the prefix there. Simplifying this is what @gavanderhoorn meant by 
> And the prefix arg of the xacro:macro could also be set from the .launch file.

Either way, there seems to be something asking for the parameter `/robot_description` which I guess is [this line](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_base/include/cartesian_controller_base/cartesian_controller_base.hpp#L79). In my opinion this is a classical case why you should never read absolute parameter paths inside your node, but some people also argue, that there should only be ONE `robot_description` parameter, anyway (which would be option 1 above). Maybe @stefanscherzinger wants to say something to that.

To continue with your approach getting two robots up and running I suggest leaving out the cartesian_controller at first and tackling this separately once both robots are working correctly.

ManueU, 2021-02-15:
Thank you for the help! 
Now all work, but the "/robot_description error" appears when I add the my_cartesian_motion_controller to the .launch file.

stefanscherzinger, 2021-02-17:
Yes, the best approach is to change that in the cartesian_controller_base via using a [different init method][1].
What could work in the meantime is to try an ugly remap (scope-limited) to your node:
```yaml

<group>
        <remap from="/robot_description" to="/my_arm_specific_robot_description"/>
        <!-- driver node here -->
</group>
```
But this is extremely brittle due to remapping /robot_description for the complete driver node..

[1]: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/13

Chemaslane, 2021-04-12:
Hi, 

I have been following this topic and have been able to make some progress, but still isn't working.

I'm launching the drivers like this:

"roslaunch ur_robot_driver ur10e_bringup.launch"

where ur10e_bringup.launch is:

`<?xml version="1.0"?>
<launch>
  <group ns="robot_1_">

	  <arg name="debug" default="false" doc="Debug flag that will get passed on to ur_common.launch"/>
	  <arg name="robot_ip" default="192.168.50.69" doc="IP address by which the robot can be reached."/>
	  <arg name="reverse_port" default="50001" doc="Port that will be opened by the driver to allow direct communication between the driver and the robot controller."/>
	  <arg name="script_sender_port" default="50002" doc="The driver will offer an interface to receive the program's URScript on this port. If the robot cannot connect to this port, `External Control` will stop immediately."/>
	  <arg name="tf_prefix" default="" doc="tf_prefix used for the robot."/>
	  <arg name="controllers" default="joint_state_controller scaled_pos_traj_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>
	  <arg name="stopped_controllers" default="pos_traj_controller joint_group_vel_controller" doc="Controllers that are initally loaded, but not started."/>
	  <arg name="controller_config_file" default="$(find ur_robot_driver)/config/ur10e_controllers_commandia_1.yaml" doc="Config file used for defining the ROS-Control controllers."/>
	  <arg name="robot_description_file" default="$(find ur_e_description)/launch/ur10e_upload.launch" doc="Robot description launch file."/>
	  
	  <arg name="kinematics_config" default="/home/inescop/ur_ros_driver_ws/ur10-calibration.yaml" doc="Kinematics config file used for calibration correction. This will be used to verify the robot's calibration is matching the robot_description."/>
	  <arg name="use_tool_communication" default="false" doc="On e-Series robots tool communication can be enabled with this argument"/>
	  <arg name="tool_voltage" default="0" doc="Tool voltage set at the beginning of the UR program. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_parity" default="0" doc="Parity configuration used for tool communication. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_baud_rate" default="115200" doc="Baud rate used for tool communication. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_stop_bits" default="1" doc="Number of stop bits used for tool communication. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_rx_idle_chars" default="1.5" doc="Number of idle chars in RX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_tx_idle_chars" default="3.5" doc="Number of idle chars in TX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_device_name" default="/tmp/ttyUR" doc="Local device name used for tool communication. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="tool_tcp_port" default="54321" doc="Port on which the robot controller publishes the tool comm interface. Only used, when `use_tool_communication` is set to true."/>
	  <arg name="limited" default="false" doc="Use the description in limited mode (Every axis rotates from -PI to PI)"/>
	  <arg name="headless_mode" default="false" doc="Automatically send URScript to robot to execute. On e-Series this does require the robot to be in 'remote-control' mode. With this, the URCap is not needed on the robot."/>

	  
	  <arg name="priority" default="true" doc="Increase priority of node execution."/>

	  <include file="$(find ur_robot_driver)/launch/ur_common.launch">
		<arg name="debug" value="$(arg debug)"/>
		<arg name="use_tool_communication" value="$(arg use_tool_communication)"/>
		<arg name="controller_config_file" value="$(arg controller_config_file)"/>
		<arg name="robot_description_file" value="$(arg robot_description_file)"/>
		<arg name="kinematics_config" value="$(arg kinematics_config)"/>
		<arg name="robot_ip" value="$(arg robot_ip)"/>
		<arg name="reverse_port" value="$(arg reverse_port)"/>
		<arg name="script_sender_port" value="$(arg script_sender_port)"/>
		<arg name="limited" value="$(arg limited)"/>
		<arg name="tf_prefix" value="$(arg tf_prefix)"/>
		<arg name="controllers" value="$(arg controllers)"/>
		<arg name="stopped_controllers" value="$(arg stopped_controllers)"/>
		<arg name="headless_mode" value="$(arg headless_mode)"/>
		<arg name="tool_voltage" value="$(arg tool_voltage)"/>
		<arg name="tool_parity" value="$(arg tool_parity)"/>
		<arg name="tool_baud_rate" value="$(arg tool_baud_rate)"/>
		<arg name="tool_stop_bits" value="$(arg tool_stop_bits)"/>
		<arg name="tool_rx_idle_chars" value="$(arg tool_rx_idle_chars)"/>
		<arg name="tool_tx_idle_chars" value="$(arg tool_tx_idle_chars)"/>
		<arg name="tool_device_name" value="$(arg tool_device_name)"/>
		<arg name="tool_tcp_port" value="$(arg tool_tcp_port)"/>
		<arg name="priority" value="$(arg priority)"/>
	  </include>
  </group>
</launch>`

I have another file ur10e_bringup_2.launch, changing the namespace, IP and ports.

Launching both drivers at the same time, seems to works correctly.
[... 33 lines cut ...]
  <link name="world" />

  
  <joint name="world_joint" type="fixed">
    <parent link="world" />
[... cut ...]
  </joint>
  
  
  <joint name="robot_2_joint" type="fixed">
    <parent link="world" />
    <child link = "robot_2_base_link" />
    <origin xyz="0.840 -0.260 0.050" rpy="0.0 0.0 1.5707" />
  </joint>

...`

The robot that appears in the simulation doesn't have the correct angles for the axis, and if I try to move it from Rviz, I receive this error:

[image]

I have tried changing the names in the .yaml file like commented before in the post and shown in the image below, with no change

[image]

Chemaslane, 2021-04-19:
I think the problem is that in "ur10e_bringup.launch" i'm using namespaces to differentiate between robots, while in the rviz file ("ur10_joint_limited_robot.urdf.xacro") i'm using prefixes.

The problem is that, if I try to use namespaces in the  "ur10_joint_limited_robot.urdf.xacro" file, the moveit setup assistant doesn't seem to find any objet.

I'm trying something like this:
Any help would be apreciated.

    <?xml version="1.0"?>
    <robot xmlns:xacro="http://ros.org/wiki/xacro"
           name="ur10" >
    
    
    <xacro:include filename="$(find ur_description)/urdf/common.gazebo.xacro" />

    
    <xacro:include filename="$(find ur_description)/urdf/ur10.urdf.xacro" />
    
    <link name="world" /
    
    <group ns="robot_1_">

      
      
      <xacro:ur10_robot prefix="" joint_limited="true"
		     shoulder_pan_lower_limit="${-pi}" shoulder_pan_upper_limit="${pi}"
		     shoulder_lift_lower_limit="${-pi}" shoulder_lift_upper_limit="${pi}"
		     elbow_joint_lower_limit="${-pi}" elbow_joint_upper_limit="${pi}"
		     wrist_1_lower_limit="${-pi}" wrist_1_upper_limit="${pi}"
		     wrist_2_lower_limit="${-pi}" wrist_2_upper_limit="${pi}"
		     wrist_3_lower_limit="${-pi}" wrist_3_upper_limit="${pi}" />		
    
      
      <joint name="world_joint" type="fixed">
        <parent link="world" />
        <child link = "base_link" />
        <origin xyz="0.838 1.383 0.050" rpy="0.0 0.0 1.5707" />
      </joint>
    ...

    </group>
    <group ns="robot_2_">

        
        
        <xacro:ur10_robot prefix="robot_2_" joint_limited="true"
		       shoulder_pan_lower_limit="${-pi}" shoulder_pan_upper_limit="${pi}"
		       shoulder_lift_lower_limit="${-pi}" shoulder_lift_upper_limit="${pi}"
		       elbow_joint_lower_limit="${-pi}" elbow_joint_upper_limit="${pi}"
		       wrist_1_lower_limit="${-pi}" wrist_1_upper_limit="${pi}"
		       wrist_2_lower_limit="${-pi}" wrist_2_upper_limit="${pi}"
		       wrist_3_lower_limit="${-pi}" wrist_3_upper_limit="${pi}" />
		       
        
        <joint name="robot_2_joint" type="fixed">
          <parent link="world" />
          <child link = "robot_2_base_link" />
          <origin xyz="0.840 -0.260 0.050" rpy="0.0 0.0 1.5707" />
        </joint>
        ...

      </group>
    </robot>

felixvd, 2022-03-04:
Thanks for reminding me of this thread. We also released a complete example of how to control two arms with Robotiq grippers: https://github.com/o2ac/o2ac-ur

evacheung0929, 2023-03-13:
Hi, 

I am a ROS beginner, and have been testing out dual-UR5 communication with real robots following the instruction. I have a couple of questions regarding the urdf and controller file as I kept failing to initialise my controller, and joint state were received but were not found in the URDF. I suspect the tf_prefix may not be called properly, but not too sure how to fix it.
[image]

I have tried both approach
> 1. Create one common workcell description that contains both robots with their respective tf_prefix similar to the one [containing one [...]

I have two robots: 'dispense' and 'pick and place', each robot has their own urdf, which define its prefix. This urdf file is then uploaded to the server individually using load_ur5e_xxx.launch, the load launch is then called in the bringup file corresponding to the robot ip address.

### Dispense.xacro
`
<?xml version="1.0"?>
									
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="ur5e_robot">
  <xacro:include filename="$(find ur_description)/urdf/inc/ur5e_macro.xacro"/>
 
   
  <xacro:ur5e_robot prefix="dispense_" kinematics_parameters_file="$(find ur_description)/config/ur5e/kinematic_dispense.yaml"/>

   
  <link name="world" />
  <joint name="world_to_dispense" type="fixed">
    <parent link="world" />
    <child link = "dispense_base_link" />
    <origin xyz="-1.5 0 0" rpy="0.0 0.0 0.0" />
  </joint>
</robot>
`

### pp.xacro
<?xml version="1.0"?>
									
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="ur5e_robot">

  <xacro:include filename="$(find ur_description)/urdf/inc/ur5e_macro.xacro"/>

  
  <xacro:ur5e_robot prefix="pickAndPlace_" kinematics_parameters_file="$(find ur_description)/config/ur5e/kinematic_pick_and_place.yaml" />

   
  <link name="world" />

  <joint name="world_to_pick" type="fixed">
    <parent link="world" />
    <child link = "pickAndPlace_base_link" />
    <origin xyz="-0.5 0 0" rpy="0.0 0.0 0.0" />
  </joint>
</robot>

### Load_ur5e_pick_and_place.launch
`
<?xml version="1.0"?>
<launch>
  
  <arg name="joint_limit_params" default="$(find ur_description)/config/ur5e/joint_limits.yaml"/>
  <arg name="kinematics_params" default="$(find ur_description)/config/ur5e/kinematic_pick_and_place.yaml"/>
  <arg name="physical_params" default="$(find ur_description)/config/ur5e/physical_parameters.yaml"/>
  <arg name="visual_params" default="$(find ur_description)/config/ur5e/visual_parameters.yaml"/>
  
  <arg name="transmission_hw_interface" default="hardware_interface/PositionJointInterface" />
  <arg name="safety_limits" default="false" doc="If True, enable the safety limits controller"/>
  <arg name="safety_pos_margin" default="0.15" doc="The lower/upper limits in the safety controller" />
  <arg name="safety_k_position" default="20" doc="Used to set k position in the safety controller" />

  <arg name="robot_model" value="ur5e" />
  <arg name="prefix" default= "pickAndPlace_" />

  
  <param name="robot_description" command="$(find xacro)/xacro '$(find ur_description)/urdf/pp.xacro'
[... 491 lines cut ...]
/pick_and_place/vel_joint_traj_controller/gains/wrist_2_joint/i_clamp
/pick_and_place/vel_joint_traj_controller/gains/wrist_2_joint/p
/pick_and_place/vel_joint_traj_controller/gains/wrist_3_joint/d
/pick_and_place/vel_joint_traj_controller/gains/wrist_3_joint/i
/pick_and_place/vel_joint_traj_controller/gains/wrist_3_joint/i_clamp
/pick_and_place/vel_joint_traj_controller/gains/wrist_3_joint/p
/pick_and_place/vel_joint_traj_controller/joints
/pick_and_place/vel_joint_traj_controller/state_publish_rate
/pick_and_place/vel_joint_traj_controller/stop_trajectory_duration
/pick_and_place/vel_joint_traj_controller/type
/pick_and_place/vel_joint_traj_controller/velocity_ff/elbow_joint
/pick_and_place/vel_joint_traj_controller/velocity_ff/shoulder_lift_joint
/pick_and_place/vel_joint_traj_controller/velocity_ff/shoulder_pan_joint
/pick_and_place/vel_joint_traj_controller/velocity_ff/wrist_1_joint
/pick_and_place/vel_joint_traj_controller/velocity_ff/wrist_2_joint
/pick_and_place/vel_joint_traj_controller/velocity_ff/wrist_3_joint
/rosdistro
/roslaunch/uris/host_cheungy_virtualbox__46281
/rosversion
/run_id
`

### Summary

I hope what I have shared can provide a general insight of what I have done, would really appreciate any advice! Thanks!

Chemaslane, 2023-03-13:
Gracias por tu mensaje. Estaré fuera de la oficina hasta el 7 de mayo.
Thank you for your message. I will be out of office until 7 May.

Intentaré contestarle lo antes posible.
We will try to contact you as soon as possible.

Para cualquier consulta urgente puedes mandar un mail a:
For any urgent enquiries you can send an email to:

	***@***.***

¡Muchas gracias!
Thank you in advance!

Un saludo, 
Best regards,

oskarsVismanis, 2023-03-14:
> Hi,
>
> I am a ROS beginner, and have been testing out dual-UR5 communication with real robots following the instruction. I have a couple of questions regarding the urdf and controller file as I kept [...]

@evacheung0929  could you add a screen capture of `rqt_graph` with everything launched? There you might see which namespaces are not connected and so, find what you need to fix!

MDario1997, 2024-08-05:
Good afternoon, i have the same problem but running an UR3 and a MIR100 in ROS Noetic 20.04 installed in Oracle VB.
In order to run both i run their launch scripts:
MIR script: roslaunch mir_driver mir.launch

UR3 script: roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.x.x.x kinematics_config:=$(rospack find ur_calibration)/ur3_robot_calibration.yaml)

As soon as i run them both an error occurs: [WARN] Shutdown request received. [WARN] Reason given for shutdown: [[/robot_state_publisher] Reason: new node registered with same name].
To solve this problem i changed in the mir.launch file the name of the node, but then i had these warnings:
In the Mir launch terminal:
[image]
In the UR launch terminal:
[image]
I guess that the problem is that the /robot_state_publisher node has as a pkg: robot_state_publisher, that is used by both and  that remains the same even if i change the name of the node of the UR or MIR, so should i have a separate package for MIR and UR in order to work simultaneously?