UniversalRobots/Universal_Robots_ROS_Driver #82: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'

Karthik-Ranga, 2020-01-17:
# Summary
I am running UR10e using moveit config. When I execute the python code, I get the below warring.
Testing this on simulator before it is connected to real robot.

`[ WARN] [1579289783.775171923]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'`

The robot doesn't show up in the RViz and shows the same warning. 
I know this was a known problem in the previous driver. But the same problem has been occurring here too. 
Is there a probable workaround for this?

# Versions
 - ROS Driver version: Melodic
 - Affected Robot Software Version(s): -
 - Affected Robot Hardware Version(s): UR10e
 - Robot Serial Number: -
 - UR+ product(s) installed: -
 - URCaps Software version(s): -

# Impact
The robot frames are not named correctly in urdf.xacro for the moveit! and RViz to 

# Issue details
Not able to visualize the robot in the RViz as the frames are not matching in the URDF.

## Use Case and Setup
Trying to control the robot using python code with collision avoidance.

## Project status at point of discovered
While running moveit_config and RViz

## Steps to Reproduce
This warning shows up when any moveit_config is run.

## Expected Behavior
The UR robot should show up in the RViz for visualization.

## Actual Behavior
The frames should have same name for the robot to show up in RViz and also moveit to communicate.

gavanderhoorn, 2020-01-17:
This is not something the driver prints, but MoveIt/KDL.

The warning is harmless, and you can ignore it.

> The robot doesn't show up in the RViz

That would be something to diagnose, but it's not caused by whatever causes the warning you report to be printed.

Karthik-Ranga, 2020-01-17:
[image]

The robot shows up in Gazebo, but in RViz it says that there is no transform.

gavanderhoorn, 2020-01-17:
We would need to know how you've started everything.

Note that the driver is not at all used with Gazebo, so I'm not sure why you are showing it.

Karthik-Ranga, 2020-01-17:
I ran below lines in 3 seperate terminals.
1) roslaunch ur_e_gazebo ur10e.launch
2) roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch sim:=true
3) rosrun ur10_e_moveit_config ur10.py

And then I initialized Rviz through rosrun rviz rviz
added the RobotModel and then I got this error.

gavanderhoorn, 2020-01-17:
As I wrote earlier: the driver is not used with Gazebo.

So it makes sense to me RViz complains, as there will be no one publishing TF frames.

Karthik-Ranga, 2020-01-17:
I understand it is not meant for Gazebo. Ultimately I will use real robot. But Before I do, I wanted to test the collision avoidance in Gazebo.

> So it makes sense to me RViz complains, as there will be no one publishing TF frames.

what do you mean by this? Can you elaborate?

gavanderhoorn, 2020-01-17:
> I understand it is not meant for Gazebo. Ultimately I will use real robot. But Before I do, I wanted to test the collision avoidance in Gazebo.

There is no way to make the driver work with Gazebo.

I don't know what else to tell you.

Karthik-Ranga, 2020-01-17:
I dont understand. Maybe I am wrong. Whatever the motion planning I do in the python code, I am able to visualize that in gazebo. It is working.

But I have problem visualizing the robot in the Rviz. that's what I have trouble with. The robot is not showing up in the RViz.

gavanderhoorn, 2020-01-17:
I believe the issue is actually in your RViz configuration.

You have `map` configured as the *Fixed Frame*.

There is no `map` frame published by anything, so RViz will not know how to render the scene correctly.

You'll want to set something like `world` or `base_link` as the *Fixed Frame*.

Note that this would have been done automatically for you if you'd used the `moveit_rviz.launch` launch file that is provided in the `ur10_e_moveit_config` package.

Karthik-Ranga, 2020-01-17:
That worked! perfect!
Thanks a lot!

closed 2020-01-17 by Karthik-Ranga

gavanderhoorn, 2020-01-17:
Note that this wasn't an issue with the driver package(s) in this repository, but rather a configuration issue on the RViz side.