UniversalRobots/Universal_Robots_ROS_Driver #83: UR 10 'Connection to robot dropped, waiting for new connection.' 'Could not get fresh data package from robot'

tahir1069, 2020-01-20:
Robot connection drops and than reconnects simultaneously. And also on the same time an error pop up could not get the fresh data.

Versions
 - ROS Driver version: Master
 - Affected Robot Software Version(s): 5.6.0
 - Affected Robot Hardware Version(s): UR-10e
 - URCaps Software version(s): External Control 1.0.1, rs485 1.0.0

# Impact
Just after running the driver first I get error in the following order:

> [ERROR] [1579530604.841603325]: Unsuccessful in setting main thread realtime priority. Error code: 1
    
> [ERROR] [1579530749.987810966]: Main thread: Scheduling is NOT SCHED_FIFO!

After these I get these three infos and error one after another: 

> [ INFO] [1579530846.068922267]: Connection to robot dropped, waiting for new connection.

> [ INFO] [1579530846.074031528]: Robot ready to receive control commands.

> [ERROR] [1579530846.137012700]: Could not get fresh data package from robot

# Issue details

I have installed everything as described in the installation instruction on the driver GitHub. URCaps, driver realtime preemptable Linux kernel and all the network configuration. First I run the bringup.launch and than a program on teach pendant containing only ExterncalControl URCap but I don't know why the above errors shows all the time.

fmauch, 2020-01-20:
As the startup tells, you are not able to perform realtime scheduling of the main control thread. Either your realtime patched kernel is not setup correctly, or your user doesn't have the respective rights to perform realtime scheduling of threads. In the latter case see [the realtime documentation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md#setup-user-privileges-to-use-real-time-scheduling).

However, `Could not get fresh data package from robot` is not (only) explainable by this. This could also be a problem related to the network setup. Do you have a direct connection between the robot and the control PC? If it only occurs every now and then, this will not be a huge problem, but it should be tackled nevertheless.

tahir1069, 2020-01-23:
The real-time kernel I have installed from [ROS2 tutorial](https://index.ros.org/doc/ros2/Tutorials/Building-Realtime-rt_preempt-kernel-for-ROS-2/). The one explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md) didn't worked for me. It installed fine but while booting Ubuntu Desktop was gone, I tried to installed from terminal but also it didn't worked. So I followed the one tutorial on Ubuntu (just a small difference from [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md#setup-user-privileges-to-use-real-time-scheduling) that in [this](https://index.ros.org/doc/ros2/Tutorials/Building-Realtime-rt_preempt-kernel-for-ROS-2/) they have used Ubuntu config file so it works totally fine for me). Well the only thing which was not allowing to run the real-time scheduling of threads was the permissions to the right real-time group. After @fmauch comment, I changed the user [privileges](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md#setup-user-privileges-to-use-real-time-scheduling). And everything related to realtime scheduling is resolved.

> NODES
  /
    controller_stopper (controller_stopper/node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ros_control_controller_spawner (controller_manager/spawner)
    ros_control_stopped_spawner (controller_manager/spawner)
    ur_hardware_interface (ur_robot_driver/ur_robot_driver_node)
  /ur_hardware_interface/
    ur_robot_state_helper (ur_robot_driver/robot_state_helper)

> auto-starting new master
> process[master]: started with pid [3809]
> ROS_MASTER_URI=http://localhost:11311

> setting /run_id to 13933cb2-3dbe-11ea-839b-448a5b43938a
> process[rosout-1]: started with pid [3820]
> started core service [/rosout]
> process[robot_state_publisher-2]: started with pid [3823]
> process[ur_hardware_interface-3]: started with pid [3824]
> process[ros_control_controller_spawner-4]: started with pid [3826]
> process[ros_control_stopped_spawner-5]: started with pid [3830]
> process[controller_stopper-6]: started with pid [3832]
> [ INFO] [1579769718.345386834]: Main thread: SCHED_FIFO OK
> [ INFO] [1579769718.346740380]: Main thread priority is 99
> [ INFO] [1579769718.353343077]: Initializing dashboard client
> [ INFO] [1579769718.357793365]: Connected: Universal Robots Dashboard Server

> process[ur_hardware_interface/ur_robot_state_helper-7]: started with pid [3840]
> [ INFO] [1579769718.390874320]: Waiting for controller manager service to come up on /controller_manager/switch_controller
> [ INFO] [1579769718.397728228]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
> [ INFO] [1579769718.419142675]: Initializing urdriver
> [ INFO] [1579769718.420206063]: Checking if calibration data matches connected robot.
> [ INFO] [1579769718.420583978]: Producer thread: SCHED_FIFO OK
> [ INFO] [1579769718.420611049]: Thread priority is 99
> [INFO] [1579769718.707723]: Controller Spawner: Waiting for service controller_manager/load_controller
> [ INFO] [1579769718.723557373]: Calibration checked successfully.
> [INFO] [1579769718.730374]: Controller Spawner: Waiting for service controller_manager/load_controller
> [ INFO] [1579769719.426164734]: Producer thread: SCHED_FIFO OK
> [ INFO] [1579769719.426211029]: Thread priority is 99
> [ INFO] [1579769719.818139044]: Setting up RTDE communication with frequency 500.000000
> [ INFO] [1579769720.842372410]: Producer thread: SCHED_FIFO OK
> [ INFO] [1579769720.842467170]: Thread priority is 99
> [ INFO] [1579769720.842695315]: Loaded ur_robot_driver hardware_interface
> [ INFO] [1579769720.898058650]: waitForService: Service [/controller_manager/switch_controller] is now available.
> [ INFO] [1579769720.898093348]: Service available.
> [ INFO] [1579769720.898116460]: Waiting for controller list service to come up on /controller_manager/list_controllers
> [ INFO] [1579769720.905226195]: Service available.
> [ INFO] [1579769721.111906233]: Robot's safety mode is now NORMAL
> [ INFO] [1579769721.113495836]: Robot mode is now RUNNING
> [INFO] [1579769721.125981]: Controller Spawner: Waiting for service controller_manager/switch_controller
> [INFO] [1579769721.131445]: Controller Spawner: Waiting for service controller_manager/unload_controller
> [INFO] [1579769721.134406]: Loading controller: joint_state_controller
> [INFO] [1579769721.147322]: Loading controller: scaled_pos_traj_controller
> [INFO] [1579769721.147895]: Controller Spawner: Waiting for service controller_manager/switch_controller
> [INFO] [1579769721.149796]: Controller Spawner: Waiting for service controller_manager/unload_controller
> [INFO] [1579769721.152697]: Loading controller: pos_traj_controller
> [INFO] [1579769721.194871]: Loading controller: speed_scaling_state_controller
> [INFO] [1579769721.234846]: Controller Spawner: Loaded controllers: pos_traj_controller
> [INFO] [1579769721.237310]: Loading controller: force_torque_sensor_controller
> [INFO] [1579769721.247292]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
> [INFO] [1579769721.254859]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller

Initially I was communicating over the Wifi network, so now I connected both the ROS PC and UR10 directly and "*[ERROR] Could not get fresh data package from robot*" is gone.

But now again I am in a new problem that the robot is not receiving commands as "*[ INFO] Robot ready to receive control commands.*" is not shown. Can you please share what can be the problem?

Thank you so much in advance.

closed 2020-01-23 by tahir1069