UniversalRobots/Universal_Robots_ROS_Driver #84: Use upstream repo, not fmauch/universal_robot

davetcoleman, 2020-01-20:
Currently the README build instructions ask us to clone [fmauch/universal_robot](https://github.com/fmauch/universal_robot), which is a branch in a fork of a fork. I'd like to submit a fix to this repo but it would be best if I could submit that fix to the upstream https://github.com/ros-industrial/universal_robot.

Are there efforts to get these changes merged back into the original ROS-Industrial repo? @gavanderhoorn ?

Side note, the fix I'd like to make:
```
ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_davetop_30024_1507649081703232519/manipulator/kinematics_solver_attempts' from your configuration.
ros.moveit_core.kinematics_base: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
```

gavanderhoorn, 2020-01-20:
> Are there efforts to get these changes merged back into the original ROS-Industrial repo?

Yes.

fmauch, 2020-01-21:
Now that we have an open issue clearly covering this:

There is a [PR going on](https://github.com/ros-industrial/universal_robot/pull/371) changing a lot in generating the robot descriptions.

We plan to merge this in first, so the calibration stuff will be generated on top of that. The [current PR of my fork](https://github.com/ros-industrial/universal_robot/pull/414) is therefore currently on hold.

boilerbots, 2020-08-19:
I am interested in seeing this getting cleaned up. Anyway we can make progress on this?

I am also hoping we can get a proper package built and deployed for Melodic.

fmauch, 2024-06-12:
This seems like is has been resolved long ago...

closed 2024-06-12 by fmauch