UniversalRobots/Universal_Robots_ROS_Driver #85: Tried to run 'test_move.py', but only showing 'Waiting for server...'

dkfwlak, 2020-01-22:
# Summary

Tried to run test motion with 'test_move.py' but the terminal only showed a message 'Waiting for server...'.

*Introduction to the issue*

# Versions
 - ROS Driver version: Melodic with Ubuntu 18.04 LTS
 - Affected Robot Software Version(s): Over 5.1 with UR3e + OnRobot RG2
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 1.0.1

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*

# Issue details

Hello.

First, I really thank you for your comment in last time (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/80#issuecomment-575511944), so now I am able to receive TCP F/T value.

Now, I'm trying to run UR3e with 'test_move.py' from 'https://github.com/fmauch/universal_robot/tree/kinetic-devel/ur_driver'.

After starting UR3e, I tried to run 'test_move.py' with
'rosrun ur_driver test_move.py',
but the terminal only showed following message
'Waiting for server...'.

What prevents 'test_move.py' from running and how can I fix to make it running?

I post message when I terminated with 'ctrl+c' after waiting more than 5 minutes:
rosrun ur_driver test_move.py
Waiting for server...
^CTraceback (most recent call last):
  File "/home/mecha/catkin_ws/src/fmauch_universal_robot/ur_driver/test_move.py", line 106, in <module>
    if __name__ == '__main__': main()
  File "/home/mecha/catkin_ws/src/fmauch_universal_robot/ur_driver/test_move.py", line 96, in main
    client.wait_for_server()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/actionlib/simple_action_client.py", line 67, in wait_for_server
    return self.action_client.wait_for_server(timeout)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/actionlib/action_client.py", line 616, in wait_for_server
    time.sleep(0.01)
KeyboardInterrupt

I'm hoping there is a solution of more.

Thank you very much.

fmauch, 2020-01-22:
As being said in #80, please do not use this issue tracker as a support platform for ROS. You are trying to interface this driver with a script from another repository. Please try to find help on [ROS Answers](https://answers.ros.org) and/or by researching on how to write action clients. There you will see that your action client (`test_move.py`) and the action server (this driver) have to agree on the topic they are communicating on, which is not the case.

closed 2020-01-22 by fmauch

dkfwlak, 2020-01-22:
Sorry for disturbing you. I thought that because (test_move.py) is included at (ur_driver), I should ask about it here. I'm still not clear with concepts of ROS but I'll make them clear to not to disturb you.

fmauch, 2020-01-22:
> I'm still not clear with concepts of ROS but I'll make them clear to not to disturb you.

It is totally fine to ask if you're not too familiar with ROS. Just use the right platform (ROS Answers) to ask questions about how to do things. Issue trackers in repositories are meant to discuss bugs inside the package's code.

BigHopes, 2020-01-27:
Hello all, 
I have the same problem, I learned from this discussion that I shouldn't use test_move.py, 
but how to test the movement of the robot then? Is there any similar program to run? How can I write one?

dkfwlak, 2020-02-06:
@BigHopes 
I changed 'follow_joint_trajectory' to 'scaled_pos_traj_controller/follow_joint_trajectory' at line 94 in 'text_move.py' (https://github.com/fmauch/universal_robot/blob/kinetic-devel/ur_driver/test_move.py) and that made the robot to move. I can't tell you why because I'm still unfamiliar with ROS, but it seems that line is associated with action client and server @fmauch said. I hope my solution works on you too.

hariharan382, 2020-10-31:
> @BigHopes
> I changed 'follow_joint_trajectory' to 'scaled_pos_traj_controller/follow_joint_trajectory' at line 94 in 'text_move.py' [...]

this didnot solve this issue