UniversalRobots/Universal_Robots_ROS_Driver #87: Use of Tool Digital I/O in e-series?

rolling2020, 2020-01-24:
In the past we've controlled grippers through this driver (and before that, ur_modern_driver) by using the regular (non-tool) digital I/O. However, we are trying to switch to a gripper that is meant to be controlled by changing the digital tool I/O on an e-series robot. While it looks like analog tool I/O is supported, it doesn't appear that the same interfaces are capable of being used for the digital tool I/O.

I'm hoping someone can clarify if there is an interface available for using digital tool I/O, and if so, which interface it is? If not, are there plans to add such an interface?

gavanderhoorn, 2020-01-24:
> While it looks like analog tool I/O is supported, it doesn't appear that the same interfaces are capable of being used for the digital tool I/O.

It would be good if you could explain which interface you are referring to here.

rolling2020, 2020-01-26:
Hey, thanks for the response. The interface we've used in the past is the "SetIO" service. The definition of which (that was also used with previous community UR drivers) has the following options for the type of output being set:
```
int8 FUN_SET_DIGITAL_OUT = 1
int8 FUN_SET_FLAG = 2
int8 FUN_SET_ANALOG_OUT = 3
int8 FUN_SET_TOOL_VOLTAGE = 4
```
However, I've just looked at the source and it looks like the server callback for this service is actually deciding which type of "digital out" to use based on the pin called in the server message.

```
  if (req.fun == req.FUN_SET_DIGITAL_OUT && ur_driver_ != nullptr)
  {
    if (req.pin <= 7)
    {
      res.success = ur_driver_->getRTDEWriter().sendStandardDigitalOutput(req.pin, req.state);
    }
    else if (req.pin <= 15)
    {
      res.success = ur_driver_->getRTDEWriter().sendConfigurableDigitalOutput(req.pin - 8, req.state);
    }
    else
    {
      res.success = ur_driver_->getRTDEWriter().sendToolDigitalOutput(req.pin - 16, req.state);
    }
  }
```
So that answers one part of my question: pin values of 16 and 17 can be used for setting the digital tool output values. 
The digital tool input values are also presumably broadcast on the IOStates msg type with the same indices, but I have not checked that yet.

fmauch, 2020-01-27:
> So that answers one part of my question: pin values of 16 and 17 can be used for setting the digital tool output values.
The digital tool input values are also presumably broadcast on the IOStates msg type with the same indices, but I have not checked that yet.

Yes to both. Which part of your question is not answered, yet?

rolling2020, 2020-01-27:
> Yes to both. Which part of your question is not answered, yet?
You've just confirmed for me that the Input is also received on places 16 and 17 in the digital input array part of io_states, so that answers both parts. Thanks!
I don't believe that is documented anywhere, so thank you for confirming that.

closed 2020-01-27 by rolling2020