UniversalRobots/Universal_Robots_ROS_Driver #91: Filehandles of TCP connection not close when "connection to robot dropped"

ipa-lth, 2020-01-27:
# Summary

Everytime the external_control is activate in the ur programm the ur_robot_driver opens a new socket (and seems to not close it when the connection is dropped).
This makes eventually leads to a "too many files open" exception and the crash of the driver.
The ur program continues to run and seems to skip over (??) the external_control urcap 

# Versions
 - ROS Driver version: [master](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/f2db8f3d000c24b77e76a2a2ae89ac2feaf7dc1c)
 - Tested with Robot Software Version(s): ursim 3.7

# Impact
With this issue this can not be used in an industrial (production line) setup. It will eventually stop and the driver needs to restart.

## Steps to Reproduce
* start the ros_controller (ur_robot_driver)
* start another node, that waits for "true" on rostopic "/ur_hardware_interface/robot_program_running"; if so sleep for 0.5s; then call rosservice "/ur_hardware_interface/hand_back_control"

* start a ur program with an infinite loop with "ExternalControl"

## Expected Behavior
Resources of the program should be stable and not grow over time.

## Actual Behavior
watch -n 0.5 cat /proc/_{PID of ur_robot_driver}_/net/sockstat

sockets: used 3082 **[continuesly increasing]**
TCP: inuse 1056  **[continuesly increasing]** orphan 0 tw 295 alloc 1057 mem 211
UDP: inuse 18 mem 27
UDPLITE: inuse 0
RAW: inuse 0
FRAG: inuse 0 memory 0

fmauch, 2020-01-27:
Thanks for reporting. This should definitely not be that way.

gavanderhoorn, 2020-01-27:
I seem to remember `ur_modern_driver` having this problem at one point as well.

@simonschmeisser, I believe it was you or your colleague who submitted a fix?

simonschmeisser, 2020-01-27:
I think it was @axelschroth who fixed it back then and he's currently working on this driver so he might know more

fmauch, 2020-01-27:
There's definitely something fishy going on. I just stumbled upon these lines:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/f2db8f3d000c24b77e76a2a2ae89ac2feaf7dc1c/ur_robot_driver/src/ur/ur_driver.cpp#L207-L208

This looks odd to me and removing the explicit destructor call leads to a crash due to the socket not being able to bind. I'll talk to @t-schnell tomorrow, but I don't assume, he had anything special in mind with this.

axelschroth, 2020-01-28:
Yep, the original issue in ur_modern_driver seems not to be the reason here.

Maybe this is the problem:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/f2db8f3d000c24b77e76a2a2ae89ac2feaf7dc1c/ur_robot_driver/src/comm/tcp_socket.cpp#L119-L122

If SocketState is Disconnected, the socket won't get closed at all..
For example in the read() method - state changes to Disconnected if no data has been read:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/f2db8f3d000c24b77e76a2a2ae89ac2feaf7dc1c/ur_robot_driver/src/comm/tcp_socket.cpp#L163-L165

fmauch, 2020-01-28:
> Yep, the original issue in ur_modern_driver seems not to be the reason here.
>
> Maybe this is the problem: [...]

I was just about to run a test with exactly that modified when I saw your answer :)

fmauch, 2020-01-28:
As we use read timeouts in `ur_robot_driver` our socket can (and in this particular case will) indeed get to the state `Disconnected` in which case, it will not get closed.

I changed the line to
```c++
  if (state_ == SocketState::Closed)
    return;
  state_ = SocketState::Closed;
  ::close(socket_fd_);
  socket_fd_ = -1;
```
 which seems to not spawn a high number of sockets anymore.

However, I think a more relevant check would be 
```c++
  if (socket_fd_ < 0)
    return;
  state_ = SocketState::Closed;
  ::close(socket_fd_);
  socket_fd_ = -1;
```

PR #93 (2020-01-30): Close all closable sockets
Sockets do not necessarily have to be in state connected when they should be
closed. Before, only connected sockets got closed leading to a "socket leak"
if a socket was disconnected before a close request was processed.

With this fix all sockets with a valid file descriptor get closed when close()
is being called.

This fixes #91
closed 2020-01-30 by fmauch