UniversalRobots/Universal_Robots_ROS_Driver #92: Building error related to SetSpeedSliderFraction.h

joaocabogon, 2020-01-27:
Hello,

I am having a problem when trying to build the master branch with catkin build. Any idea why?

When I check my ur_msgs package I have there the SetSpeedSliderFraction.srv

Using Ubuntu 16.04 and ROS Kinetic.

`Errors     << ur_robot_driver:make /home/joao/ros_ws/logs/ur_robot_driver/build.make.011.log                    
In file included from /home/joao/ros_ws/src/atrv_ur5e/universal_robots_ros_driver/ur_robot_driver/src/ros/hardware_interface.cpp:28:0:
/home/joao/ros_ws/src/atrv_ur5e/universal_robots_ros_driver/ur_robot_driver/include/ur_robot_driver/ros/hardware_interface.h:46:44: fatal error: ur_msgs/SetSpeedSliderFraction.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/ur_robot_driver_node.dir/src/ros/hardware_interface.cpp.o] Error 1`

fmauch, 2020-01-28:
Did you follow the build guide as proposed here: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building

Especially cloning the `universal_robot` repository from my fork. Unfortunately, this is still necessary...

gavanderhoorn, 2020-01-28:
> Especially cloning the `universal_robot` repository from my fork. Unfortunately, this is still necessary...

Not for `SetSpeedSliderFraction`: [ur_msgs/SetSpeedSliderFraction](http://docs.ros.org/kinetic/api/ur_msgs/html/srv/SetSpeedSliderFraction.html).

That's even been released.

fmauch, 2020-01-28:
> > Especially cloning the `universal_robot` repository from my fork. Unfortunately, this is still necessary...
> 
> Not for `SetSpeedSliderFraction`: [ur_msgs/SetSpeedSliderFraction](http://docs.ros.org/kinetic/api/ur_msgs/html/srv/SetSpeedSliderFraction.html).
> 
> That's even been released.

Right, that's one of the minor things I had merged :)
@joaocabogon: I still suggest that you use the forked version as the robot launch files will not work with the upstream version, currently.

gavanderhoorn, 2020-01-28:
> I still suggest that you use the forked version as the robot launch files will not work with the upstream version, currently.

Of course. That would still be necessary.

I'm just surprised there are build failures for this service specifically, as "even upstream" has that file, and it has been released as well.

joaocabogon, 2020-01-28:
I followed the guide you propose, but in fact I am using an universal_robot old repo version with some important changes I made for myself. So basically, I just updated the ur_msgs package. I was trying to avoid having to update the whole universal_robot repository

gavanderhoorn, 2020-01-28:
There have been quite a few changes -- and I believe for the better -- since the change to `ur_rtde_driver`.

If you have the opportunity, I would upgrade to the current version.

Note that if you're not using the `universal_robot` fork you won't be able to benefit from the kinematic calibration this driver supports.

gavanderhoorn, 2020-01-28:
Well it's software. I won't say there is "no way", but it's going to be involved, as you'd need to port all the required commits from the current driver to the `ur_rtde_driver` version.

That's not going to be trivial I believe.

fmauch, 2020-05-12:
Is there anything to add here or should I close this?

fmauch, 2020-08-14:
I'll close this as no objection was raised against my previous question.

closed 2020-08-14 by fmauch