UniversalRobots/Universal_Robots_ROS_Driver #94: Controller not running

LanderU, 2020-01-28:
# Summary

When I try to switch the controller type. This is the output:

```bash
[ERROR] [1580213554.280455627]: Robot control is currently inactive. Starting controllers that claim resources is currently not possible. Not starting controller 'scaled_pos_traj_controller'
[ERROR] [1580213554.280507040]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
```

*Introduction to the issue*

# Versions
 - ROS Driver version: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/f2db8f3d000c24b77e76a2a2ae89ac2feaf7dc1c
 - Affected Robot Software Version(s): 3.12
 - Affected Robot Hardware Version(s): CB.3.1
 - ROS version: melodic
 - Robot Serial Number: 201631450
 - UR+ product(s) installed: ?
 - URCaps Software version(s):  https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/externalcontrol-1.0.1.urcap

# Impact

* Unable to move the robot using ROS.

# Issue details

I can connect the robot over ROS but, cannot move the robot because it claims that the controller is not running. When I execute:

```bash
root@alias:~/ur_ws# rosservice call /controller_manager/list_controllers "{}" 
controller: 
  - 
    name: "joint_state_controller"
    state: "running"
    type: "joint_state_controller/JointStateController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
  - 
    name: "pos_traj_controller"
    state: "stopped"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: [elbow_joint, shoulder_lift_joint, shoulder_pan_joint, wrist_1_joint, wrist_2_joint,
  wrist_3_joint]
  - 
    name: "scaled_pos_traj_controller"
    state: "stopped"
    type: "position_controllers/ScaledJointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "ur_controllers::ScaledPositionJointInterface"
        resources: [elbow_joint, shoulder_lift_joint, shoulder_pan_joint, wrist_1_joint, wrist_2_joint,
  wrist_3_joint]
  - 
    name: "speed_scaling_state_controller"
    state: "running"
    type: "ur_controllers/SpeedScalingStateController"
    claimed_resources: 
      - 
        hardware_interface: "ur_controllers::SpeedScalingInterface"
        resources: []
  - 
    name: "force_torque_sensor_controller"
    state: "running"
    type: "force_torque_sensor_controller/ForceTorqueSensorController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::ForceTorqueSensorInterface"
        resources: []
````

When trying to switch the status using:

```bash
root@alias:~/ur_ws# rosservice call /controller_manager/switch_controller "start_controllers:
- 'scaled_pos_traj_controller'
stop_controllers:
- ''
strictness: 1" 
ok: False
```
And the driver output is:

```bash
[ERROR] [1580213554.280455627]: Robot control is currently inactive. Starting controllers that claim resources is currently not possible. Not starting controller 'scaled_pos_traj_controller'
[ERROR] [1580213554.280507040]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
```

## Steps to Reproduce

Here the the steps in video:

#### Driver:

https://asciinema.org/a/jC1zgZYCoWrbDulS5bDYngvwk

#### Other terminal:

https://asciinema.org/a/hhUIMKdGb4EvobUEZAPYXb97M

If more information is needed, let me know.

Thanks in advance.

## Expected Behavior
Change the controller from stopped to running.

## Actual Behavior
Cannot change the controller status.

fmauch, 2020-01-28:
You have to get the script code running on the robot through the `External Control` URCap as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#setting-up-a-ur-robot-for-ur_robot_driver) and [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot).

Once this is running, you will see the output `Robot ready to receive control commands.` in the shell running the driver. The default controller will be then started automatically and you can switch controllers as you like.

LanderU, 2020-01-28:
Thanks for the quick answer! I'm able now to move and change between controllers. Thanks you!

closed 2020-01-28 by LanderU