UniversalRobots/Universal_Robots_ROS_Driver #95: When will velocity control be implemented?

akihikoy, 2020-01-29:
Hi,

Could you tell me when you are going to implement a velocity control mode?

According to [Feature comparison and roadmap](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/261fac97ade85b0030d5fcfcc8eb9cc1f01654d8/ur_robot_driver/doc/features.md), it seems that implementing a velocity control mode is planned.  I would like to know when it will be released.

Many thanks!

gavanderhoorn, 2020-01-29:
Have you seen #1 ?

fmauch, 2020-01-29:
Have a look at #1. From my side this is currently under testing and reviewing before it can be merged. If you like, go ahead and test it and give feedback.

akihikoy, 2020-01-29:
Great! I will test  #1.

fmauch, 2020-02-05:
#1 got merged. Closing this.

closed 2020-02-05 by fmauch