UniversalRobots/Universal_Robots_ROS_Driver #96: Enhancement: Prefix log messages with program name

fmauch, 2020-01-29:
As @gavanderhoorn  suggested in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/1#issuecomment-579651849 it might be a good idea to prefix all log messages with "ExternalControl: " to make them more visible inside TP logs.

ipa-kut, 2020-07-07:
I will start working on this now, as part of wrid20

ipa-kut, 2020-07-07:
I am currently adding the fixed string "ExternalControl: " to all textmsg() calls in ros_control.urscript.   

> We could also take the basename of the used script file to make it a bit more flexible.

 I must admit my knowledge of this driver is quite limited, so please let me know a list of any other scripts/files that need this enhancement.

fmauch, 2020-07-07:
This is the only file that will get affected by this. It's fine to go ahead with "ExternalControl: "

fmauch, 2020-07-07:
Fixed by #222 

@gavanderhoorn Could you please move the associated card on the issue board?

closed 2020-07-07 by fmauch

gavanderhoorn, 2020-07-07:
Done.